MarlinFirmware/Marlin #2287: can't compile if i select 2 extruders

juri74, 2015-06-13:
complilatuon run well in one extruder mode,
but if i change #define EXTRUDERS from 1 to 2 i get this error:

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:665:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:665:1: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 141 [ ivtmp.171 ])) [4 MEM[base: _101, offset: 0B]+0 S4 A8])) temperature.cpp:449 99 {*movsf}
     (expr_list:REG_INC (reg:HI 141 [ ivtmp.171 ])
        (nil)))
temperature.cpp:665: confused by earlier errors, bailing out

i'm using the latest marlin firmware and arduino ide 1.5.8
boards selected is "BOARD_RAMPS_13_EEB"

AnHardt, 2015-06-13:
For Marlin 1.0.2 (the Release branch) this is a known error.

>Does not compile under Windows with more than one extruder under Arduino 1.6.x - use 1.0.6.

It looks like this is also true for 1.5.8.

The error is fixed in the Development branch.

closed 2015-06-13 by AnHardt

juri74, 2015-06-13:
thank you for your quick reply AnHardt, i switched to development version and now it compile flawlessly with 2 extruders selected... problem now is that encoder don't do absolutely nothing!
i made only these changes on configuration.h:
changed motherboard to BOARD_RAMPS_13_EEB
selected 2 extruders
uncommented  #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

i tried to uncomment some other boards unsuccessfully, is this a know issue? or i'm doing something wrong?

AnHardt, 2015-06-13:
@juri74 
Do you see a MINTEMP_ERROR on the display?
If Yes - you need to define a second thermometer.

juri74, 2015-06-14:
thank you to help me sorting this out, i'm doing a 2 extruders printer for a friend, never had 2 extruders.
yes, i see MINTEMP error, but this error was shown even with 1 extruder and encoder worked well.
i'm testing it with only mega2560 + ramps 1.4 + reprapdiscount full graphic smart controller. and nothing else connected.
should i connect sensors? or i have to "define" a second thermometer? but how?
thanks

AnHardt, 2015-06-14:
The other way to test, is to use tables 999 and 998.