MarlinFirmware/Marlin #2288: Remove BOARD_MELZI_1284 from boards.h?

silpstream, 2015-06-13:
I initially used `BOARD_MELZI_1284` on the Release branch with no issues. Upon changing to Development branch, I ran into a problem where the fan could no longer be controlled because `FAN_PIN` and `LED_PIN` were no longer being defined for this board.

Primarily because in Release we were checking using `#ifdef MELZI`, but in Development we use `#if MB(MELZI)`. We also undefined `MOTHERBOARD` specifically in `pins_MELZI_1284.h` so adding ` || MB(MELZI_1284)` wouldn't really help.

Anyhow, I was initially going to fix this in a pull request, but further inspection shows that `BOARD_MELZI_1284` isn't needed and we already handle the 644 vs 1284 processor differences with `#if defined(__AVR_ATmega1284P__)`, and I couldn't find any other instance where `MELZI_1284` (or `BOARD_SANGUINOLOLU_11` for that matter) would be called or needed.

As such wouldn't it be better to just remove the option for `BOARD_MELZI_1284` entirely from `boards.h`?

Wackerbarth, 2015-06-13:
I believe so. This is another example of a case where an instance (BOARD_MELZI_1284) was used to describe a property. The use of __AVR_ATmega1284P__ is more general in that it is both necessary (for the compiler to generate appropriate code) and more general (non-MELZI boards might also use this processor)  We have a similar case with configurations based on the at90usb1286 processors. There, too, unification is in order to reduce the otherwise duplicated code.

markotime, 2015-06-15:
Actually, the V2 Melzi board uses a 44-pin chip, not 40, so things like FAN move from 4 to 44.
I have no idea how to make it work, and I fumble-fingeredly over-wrote the original firmware.  I can find no reference to the 44 pin version, anywhere.  anyone help? /thanks

markotime, 2015-06-15:
My point being that while the DIP 1284p conveniently maps to the 644, the 1284p quad / 44 pin one
does not.  Worth a look.

silpstream, 2015-06-16:
@markotime, I'm not aware of any Melzi board that uses the DIP package. Anyhow the TQFP and DIP packages have the same ports and this is handled in the Arduino bootloader to map to the relevant pins, which it does in a similar way for both packages.

markotime, 2015-06-16:
> No virus found in this message.
> Checked by AVG - www.avg.com <http://www.avg.com>
> Version: 2015.0.5961 / Virus Database: 4365/10025 - Release Date: 06/15/15

Thank you.
Now, how can I arrange to load Marlin onto my Melzi V2 board and have 
the pins map
correctly?

markotime, 2015-06-16:
\
Bearing in mind that my formal computer training ended in aobut 1968 
with Fortran IV <grin>

silpstream, 2015-06-16:
@markotime , I suggest that you raise/open this as a separate question on the reprap forums. You'll get more help there as this is the wrong place to be asking and this thread specifically should not be cluttered with anything not related to the original topic it was opened for.

That being said. Try:
https://reprappro.com/documentation/mendel-tricolour/maintenance/#Uploading_new_firmware

They have a good walk through, but make sure configuration.h is customised for your specific setup. Also do a google search for "marlin melzi", you'll get a lot of good hits.

This is about as much help as I can provide from within this thread. Good luck!

markotime, 2015-06-16:
> Version: 2015.0.5961 / Virus Database: 4365/10029 - Release Date: 06/16/15

Been there, done all that, nowhere is pins.h mapped properly for that 
board.  But thanks for looking!

silpstream, 2015-06-16:
@markotime , I searched out your reprap forum post and replied you there.

AnHardt, 2015-06-16:
Board 66 was introduced from @makr3d with PR #542 togetrer with the MAKRPANEL.
At this time the boards had no names. The pinning is exactly the same as for board 63 MELZI. Other details vary slightly.

>+// 66 = Melzi with ATmega1284 (MaKr3d version)

In the process if naming board 66 got the name MELZI_1284

>#define BOARD_MELZI_1284        66   // Melzi with ATmega1284 (MaKr3d version)

Maybe MELZI_MAKR3D would have been the better name.

>#define MOTHERBOARD BOARD_SANGUINOLOLU_11

is an error introduced during splitting the big pins.h apart and should read

>#define MOTHERBOARD BOARD_MELZI

as we later ask in (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/pins_SANGUINOLOLU_11.h#L32) for

>#if MB(MELZI)

The MELZI boards never had the socket (40 pin) type 644, so the pinning is for sure for the TQFP (44 pin) version.

silpstream, 2015-06-20:
That definitely sheds light on why we have MELZI_1284. I can't see how the MAKR3D Melzi board differs for standard Melzi, but the changes you made make sense, even if just to make sure each board manufacturer has some representation.

AnHardt, 2015-06-20:
Indeed the pinning is exacty the same for the `MAKR3D Melzi`, while the hardware differs a bit (stepperdrivers on sockets, ...). Removing instead of #2303 is an option. But that's a more political question __I__ don't want to answer.

thinkyhead, 2015-06-22:
Mergified!

closed 2015-06-22 by thinkyhead