MarlinFirmware/Marlin #2289: Compilation error

Boogy62, 2015-06-13:
With #define ULTRA_LCD uncomented, there is an error: 

C:\Users\Paul\AppData\Local\Temp\build2040415868027422873.tmp/Marlin_main.cpp:4360: undefined reference to `lcd_buzz(long, unsigned int)'

I think that is there : 

#if HAS_LCD_BUZZ

  /**
   * M300: Play beep sound S<frequency Hz> P<duration ms>
   */
  inline void gcode_M300() {
    uint16_t beepS = code_seen('S') ? code_value_short() : 110;
    uint32_t beepP = code_seen('P') ? code_value_long() : 1000;
    if (beepP > 5000) beepP = 5000; // limit to 5 seconds
    lcd_buzz(beepP, beepS);
  }

#endif // HAS_LCD_BUZZ

AnHardt, 2015-06-13:
My crystal ball tells me ---- uh nothing today.
Version? Board?

AnHardt, 2015-06-13:
A bit off polishing the ball tells me: Development branch, at least RAMPS.

Some when we made dependencies for ULTRA_LCD when we meant ULTIPANEL where ULTIPANEL is used more or less a placeholder for 'has a panel'.

thinkyhead, 2015-06-15:
I've never heard of an `ULTRA_LCD` by itself but apparently some 16x2 display(s) only need `ULTRA_LCD` to be set, and nothing else. I don't think that the 16x2 displays that are supported by this option have buzzers, though. Perhaps you are using a different LCD controller…?

thinkyhead, 2015-06-15:
Aha, here's something else I just found in `Marlin_main.cpp`…
```
#define HAS_LCD_BUZZ (defined(ULTRALCD) || (defined(BEEPER) && BEEPER >= 0) || defined(LCD_USE_I2C_BUZZER))
```
Note that `ULTRA_LCD` is misspelled as `ULTRALCD`. I will fix that forthwith!

thinkyhead, 2015-06-15:
https://github.com/MarlinFirmware/Marlin/commit/5dccfda37d301a12cb5b89f176499afc5c4d681c

Wackerbarth, 2015-06-23:
Can we close this?

Boogy62, 2015-06-23:
Yes !

closed 2015-06-23 by Boogy62