MarlinFirmware/Marlin #2291: Suggestion for new feature

plgarcia, 2015-06-14:
On my printer and depending on the PLA I use, when I stop the fan as the nozzle is still hot, sometimes the pla distorts as the body of the extruder warms. This leads somtimes to tap the extruder next time the printer is used.

A security could be implemented and parametrable for the fan to be on (50% for example) as long as the temperature of the nozzle is higher than a certain level (exemple 50°), and so what ever is commanded ether by control menu or by gcode.

M117 would the lead to 50% speed of the fan, until the temp of the nozzle is < 50° than stops the fan.

plgarcia, 2015-06-16:
The PLA must be hot in the nozzle, but it must be cold enough before the nozzle even in the body of the extruder.
My machine has only one fan that cools down the body and the extruded PLA. Further more the pla is not guided all along its path. Any distortion of the PLA may distub the flow of the PLA and ends on my machine to a interruption.
Without the fan the body becomes warm and this causes this kind of distortion.
This may happen during printing, but also after printing. When the nozzle cools down
 it communicate its heat to the body and this may suffice to cause problem on next use.
I solved my problem by ensuring that the fan is always on during printing and by adding at the end of the gcode :
M106    S255   ;start fan full speed
M109 S60
M104 S0                     ;extruder heater off
M107       ;stops the fan
so the fan is stoped when the temp is lower than 60°.
 And since then all extrision problems are gone.

I plan to mount a second fan, one for the body, one for the extruded pla.

But It would b nice this security is implemented id mendel.
When the temp is higher than a certain level than the fan cooling the body of the extruder is still on, and if there is no specific fan than the M117 or M106 command should not be able to stop the fan.

I Have two problems : 
How to connect the new fan ?
Having this security function implemented and parametrable.

I may help to implement this.

plgarcia, 2015-06-16:
In fact it is quite simple to do :
There is a functionnality called TEMP_STAT_LEDS that almst does the job.
The only thing to do is to add a fet to command the fan.

Wurstnase, 2015-06-16:
This feature exist in the advanced_config? Or did you mean something different?

ntoff, 2015-06-17:
This just sounds like a badly designed hot end. The print cooling fan should *only* be controlled by the slicer for cooling the print and shouldn't be relied upon to keep the extruder heatsink cold.

You need to add a second fan for the extruder body only. You can get a small 12v fan connect it up to the 12v input on the RAMPS board.

wes1007, 2015-06-17:
All Metal Hotends require a fan to keep them cool. 

Id like to add to this:

Would be nice to have this feature expanded to include ramping the fan up to speed dependent on temperature. You set your starting point and the point where the fan is at full speed.

For example:
MIn temp is set to 50c with a duty cycle of 100 and max is set to 100c with a duty cycle of 255. WHen the hotend hits 50C the fan starts up at the speed. when the hotend hits 100c or higher it will go to full power.

AnHardt, 2015-06-17:
@plgarcia 
Please search for `EXTRUDER_AUTO_FAN` in `Configuration_adv.h`.
This is pretty close to what you suggested in your first message..

plgarcia, 2015-06-17:
All this sounds great. This is exacty what I want. Sory I missed that in the code.

I agree that there is a problem in the desing of the machine. There should be 2 fans. And I design a support that will have 2 fans, ont for the pla, one for the body of the extruder.
I like the idea to control the fan by program, but there is no power output left for this on the ramp.

I plan to use SER1 D11 for example and I will add power stage (as the one for  heater and fan on the ramp). Is that a pertinant choice.  Is there any solution available for that ?

Thanks for these answers.
Good software!

AnHardt, 2015-06-17:
Things like this:
http://www.ebay.de/itm/New-4-Route-MOSFET-Button-IRF540-V2-0-MOSFET-Switch-Module-Arduino-Compatible-/261125619383?pt=LH_DefaultDomain_0&hash=item3ccc4cbeb7
http://www.ebay.de/itm/1pcs-IRF520-MOS-FET-Driver-Module-for-Arduino-/381176546414?pt=LH_DefaultDomain_0&hash=item58bfe4846e
work well for me.
Maybe 
http://www.geeetech.com/wiki/index.php/Reprap_Ramps1.4_RRD_Fan_Extender
could be even more convenient, but have a look at #2104.

plgarcia, 2015-06-17:
I checked the code. That would be very easy to implement that proportionality suggested by wes1007.
I think also that the code with the blue and read light should be movers from main to temperature.

@AnHardt : Dankeshön. I think that is exactly what I need!

wes1007, 2015-06-17:
@ AnHardt I was going to suggest making one out of vero/strip board and sourcing the MOSFETs but at that price its easier to go with that. ( I have just finished making a board with 4 FETS)

eboston, 2015-06-17:
Did you check out the requirements for that Geeetech unit? It needs Marlin
v3!  Given we are only on v1.0.3 dev, going to be a while before it can be
used.  :D

>
> Things like this:
http://www.ebay.de/itm/New-4-Route-MOSFET-Button-IRF540-V2-0-MOSFET-Switch-Module-Arduino-Compatible-/261125619383?pt=LH_DefaultDomain_0&hash=item3ccc4cbeb7
http://www.ebay.de/itm/1pcs-IRF520-MOS-FET-Driver-Module-for-Arduino-/381176546414?pt=LH_DefaultDomain_0&hash=item58bfe4846e

brainscan, 2015-06-17:
I just have my fan for the hotend connected straight to 12v from my atx and then control the power supply to switch off as soon as the nozzle has dropped below 60°c after a print. I can't control it independently but as I only want it to run at full speed while printing this has been my best solution so far.

wes1007, 2015-06-17:
Mine is the same. However my printer shuts off right after the print is done.

The idea of having the fan ramp up is that you could **maybe** get faster heat up times.

plgarcia, 2015-06-17:
The fan we are talking about is the fan of the body of the extruder right ?
If well designed (not like my machine) it should not blow on the nozzle. Changing the speed of that fan should not change much the time to heat-up the hotend.
On my machine the 40w hotend raises its temperature in 2 minutes with the fan on blowing everywhere.

plgarcia, 2015-06-18:
I had a quite exhautive look to the code.
When there is min temperature for the extruder there should be no case the priter should be stopped if this minimum is no reached.
Even with the M80 ans M81 command.

Differring the shut down to reach that temperature means that the shut down could be cancelled, if for example a M80 arrives before the min temp is reached.

I have not much time now but I think I will propose a change this summer.

Should we close the issue as I get mmy response ?

Regards
Pascal Garcia

AnHardt, 2015-06-18:
@wes1007 
The trick for waiting is to use 'M109 __R__ 60' instead of 'M109 __S__ 60'.

plgarcia, 2015-06-18:
Ok fine, I did not find this option on the documentation.
Thank you.

thinkyhead, 2015-06-22:
__M109: Set Extruder Temperature and Wait__

Set extruder heater temperature in degrees celsius and wait for this temperature to be achieved.

__Parameters__
* Snnn minimum target temperature, waits until heating
* Rnnn maximum target temperature, waits until cooling (Sprinter)
* Rnnn accurate target temperature, waits until heating and cooling (Marlin)

__Example__

    M109 S215

closed 2015-06-22 by thinkyhead