MarlinFirmware/Marlin #2293: Auto-leveling not working as expected in Robo3D R1

WheresWaldo, 2015-06-14:
Background: Robo3D R1 (R1) with auto-level and 8 mm threaded rods using stock firmware based on 1.0.0 works after a fashion. This is a report about auto-leveling in 1.0.2 release with the proper modifications for use on the R1.

Issue stems from the use of Z_PROBE_OFFSET_FROM_EXTRUDER. I will explain as best as I can. The R1 uses two switches under the x-carriage in series to determine when it has hit the endstop for the Z axis. These switches are normally closed and actuating either results in the endstop being triggered. They are affixed nuts that can travel independently from the Z axis mount. When the lead screws are turned beyond the point that the extruder head hits the bed, one or both switches activate (open). This is how the R1 knows it has hit the endstop. The distance the lead screws move between the time that the hotend hits the bed and the switch actually triggers the endstop was measured on my machine, but should apply to all R1's with auto-leveling. In my specific case the Z_PROBE_OFFSET_FROM_EXTRUDER is a negative value. The probe actually triggers below the top of the bed. In my specific case one or both switches trigger -1.19 mm from the top of the bed.

On line 402 this offset was inserted in the proper line. Among other changes in Configuration.h to make it work for the R1.

```Diff
--- D:\Documents\Arduino\Marlin-1.0.2-1\Marlin\Configuration.h 6/14/2015  4:07:30 PM
+++ D:\Documents\Arduino\Robo_1.0.2\Configuration.h 6/14/2015  4:09:02 PM
@@ -28,34 +28,34 @@
 //#define STRING_VERSION "1.0.2"
 
 #define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
-#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
+#define STRING_CONFIG_H_AUTHOR "(BH, R1 Single Extruder)" //--ROBO-BH
 
 // SERIAL_PORT selects which serial port should be used for communication with the host.
 // This allows the connection of wireless adapters (for instance) to non-default port pins.
@@ -44,54 +44,54 @@
 // The following define selects which electronics board you have.
 // Please choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_ULTIMAKER
+  #define MOTHERBOARD BOARD_RAMPS_13_EFB	//--ROBO-BH
 #endif
 
 // Define this to set a custom name for your generic Mendel,
-// #define CUSTOM_MENDEL_NAME "This Mendel"
+#define CUSTOM_MENDEL_NAME "Robo3D v1.0.2"	//--ROBO-BH
 
 // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
 // You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
@@ -104,113 +104,113 @@
 // 147 is Pt100 with 4k7 pullup
 // 110 is Pt100 with 1k pullup (non standard)
 
-#define TEMP_SENSOR_0 -1
+#define TEMP_SENSOR_0 1		//--ROBO-BH
-#define TEMP_SENSOR_1 -1
+#define TEMP_SENSOR_1 0  	//--ROBO-BH
 #define TEMP_SENSOR_2 0
-#define TEMP_SENSOR_BED 0
+#define TEMP_SENSOR_BED 1	//--ROBO-BH
 
 // This makes temp sensor 1 a redundant sensor for sensor 0. If the temperatures difference between these sensors is to high the print will be aborted.
 //#define TEMP_SENSOR_1_AS_REDUNDANT
@@ -129,138 +129,138 @@
 // When temperature exceeds max temp, your heater will be switched off.
 // This feature exists to protect your hotend from overheating accidentally, but *NOT* from thermistor short/failure!
 // You should use MINTEMP for thermistor short/failure protection.
-#define HEATER_0_MAXTEMP 275
+#define HEATER_0_MAXTEMP 330	//--ROBO-BH
 #define HEATER_1_MAXTEMP 275
 #define HEATER_2_MAXTEMP 275
-#define BED_MAXTEMP 150
+#define BED_MAXTEMP 120  		//--ROBO-BH
 
 // If your bed has low resistance e.g. .6 ohm and throws the fuse you can duty cycle it to reduce the
 // average current. The value should be an integer and the heat bed will be turned on for 1 interval of
@@ -219,225 +219,225 @@
 //if PREVENT_DANGEROUS_EXTRUDE is on, you can still disable (uncomment) very long bits of extrusion separately.
 #define PREVENT_LENGTHY_EXTRUDE
 
-#define EXTRUDE_MINTEMP 170
+#define EXTRUDE_MINTEMP 165		//--ROBO-BH
 #define EXTRUDE_MAXLENGTH (X_MAX_LENGTH+Y_MAX_LENGTH) //prevent extrusion of very large distances.
 
 /*================== Thermal Runaway Protection ==============================
@@ -248,262 +248,262 @@
 // uncomment the 2 defines below:
[... 195 lines cut ...]
 
 #define ABS_PREHEAT_HOTEND_TEMP 240
-#define ABS_PREHEAT_HPB_TEMP 100
+#define ABS_PREHEAT_HPB_TEMP 80		//--ROBO-BH
 #define ABS_PREHEAT_FAN_SPEED 255   // Insert Value between 0 and 255
 
 //LCD and SD support
@@ -559,565 +559,565 @@
 // http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
 //
 // ==> REMEMBER TO INSTALL U8glib to your ARDUINO library folder: http://code.google.com/p/u8glib/wiki/u8glib
-//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER	//--ROBO-BH
 
 // The RepRapWorld REPRAPWORLD_KEYPAD v1.1
 // http://reprapworld.com/?products_details&products_id=202&cPath=1591_1626
```
Here is the issue. After uploading and resetting the printer I perform a M503 to make sure that I get the following output <b>M206 X0.00 Y0.00 Z-1.19</b>. Once it is determined the correct offset is both in Memory and in EEPROM. I proceed with a <b>G28</b> followed by a <b>G29</b>.

The output of the G28 is noted in a serial monitor as well as on an LCD Smart Controller installed on the printer. After issuing the G28 but before issuing the G29 the printer moves appropriately to home X then home Y then center the hotend to home X. The result is that my location is listed as <b>X:120 Y:127 Z:-001.1</b> (output is truncated by the software). This is behavior that is expected. If I then issue a <b>G1 Z0</b> I end up with the hotend exactly where I expect it hovering at the top of the bed and Z0 being the actual point it should be. Here is where the problem begins. Now executing a <b>G29</b> Auto-leveling command, the printer moves to the appropriate nine locations on the print bed and performs a Z axis endstop probe. When it has reached its 9th and final location the firmware makes the assumption that it is resting at the Z0 location when in fact it is actually at the Z axis trigger point. The more egregious part is that the firmware actually believes this point is Z0 + -Z_OFFSET_FROM_EXTRUDER. So not only is it 1.19 mm too low, it is telling me that it is 1.19 mm in the air. Any command to move the Z axis is now too low by double the amount of the offset. This is repeatable with any amount of negative offset you want to put in the Configuration.h file.

To verify that it isn't simply assuming that the Z_PROBE_OFFSET_FROM_EXTRUDER
should have been entered as a positive number I edited the source with a +1.19, recompiled and performed the same experiment. This time the result was that the final resting place of the G29 command was still the trigger point but the firmware reported that I was actually at Z:+002.3. Any Z axis moves are now too low by over 3.5 mm. No matter what is placed in Z_PROBE_OFFSET_FROM_EXTRUDER, G29 appears to always assume that the last Z axis endstop trigger point is Z0 and then it adds the positive value of the offset instead of what it really is.

I am willing to provide any additional information and test various solutions to solve this issue. I would not have reported this until I understood the problem and saw no solution in any of the configuration file options.

thinkyhead, 2015-06-15:
Great description, and thanks for posting as a diff. Let me see if we're on the same page…

First, you should really be using the latest Development code. A lot has changed since 1.0.2, including adding support for a Z endstop that is also a probe, which makes sense for your machine.

Currently, `M206` is used to set the `home_offset` values. Meanwhile `M851` is used to set the offset of the nozzle from the probe, `zprobe_zoffset`. So, you might want to try using `M851 Z-1.19` in addition to setting the `home_offset` with `M206`, since your "probe" is that far from Z=0. This gets a little confusing with your type of setup, so correct me if I'm wrong about setting both or just one. You will also need to enable the new `Z_PROBE_ENDSTOP` option.

brainscan, 2015-06-15:
I made a probe like this a while ago but I had to use a positive offset from extruder as it actually activated below the level of the bed. This then caused other problems with bed levelling so I changed to a capacitance probe. Not sure if it would be the same with yours but from your description it sounds like a similar principle.

WheresWaldo, 2015-06-15:
Please forgive me for not completely understanding how GITHub works, I am not a programmer, more of a documentation writer. I am more than willing and able to put the time in to test this against the development branch, I just did it to the release since it is easier to access the source (I just downloaded the .zip). I was also under the impression that 1.0.2 release was the most stable. So if someone can point me to how to get development branch code in a neat little folder that Arduino IDE can use, it would be most welcome.

AnHardt, 2015-06-15:
@WheresWaldo 
The short way:
https://github.com/MarlinFirmware/Marlin/archive/Development.zip
The long way:
Goto: https://github.com/MarlinFirmware/Marlin
At the top left, near the green icon, select the Development branch.
Than at the bottom of the right column click DownloadZIP.

Wackerbarth, 2015-06-15:
@WheresWaldo -- I strongly recommend that you shift to using git to directly access our repository.
https://help.github.com/articles/set-up-git/  provides an excellent description of the setup.

I also recommend that you create your own (free) account on github and fork the Marlin repo.
Then you can simply push your changes to your copy and we can reference them.

Don't worry about messing up a commit in your repo. git is very forgiving and allows "do-over".

WheresWaldo, 2015-06-15:
I have an account and have forked the branch already. I also have installed Git for Windows. @Wackerbarth you already pointed out a stupid mistake I made with language_en.h. So what I will do is go to my fork, select development branch then download the zip from the link. If I understand GIT correctly since I forked and not cloned any changes you guys/gals make to the dev should show up in my fork. I will do my best to document my findings.

Wackerbarth, 2015-06-15:
If you have installed Git for Windows, rather than fetching a .zip file, just use git to make a copy on your local PC. Your reaction of a fork on Github is equivalent to making an archive. It contains only what you put there (initially a copy of the repository that you forked) It does not update automatically.

The easiest way to handle updates is to have two remotes for the local copy of the repository on your PC. Set one (usually called "origin") to point to your github repository. Set the other to point to our MarlinFirmware repository (I call it "upstream" in my setup)

Then, you can "git fetch upstream" and bring the latest changes to your machine. You use "git checkout" to extract a particular version of the code that you wish to use at that time.

I strongly recommend that you learn to use SCM workflows (such as git). It makes life much more manageable and will give you skills that you will continue to use in all of your work.

WheresWaldo, 2015-06-15:
Tested against current Development branch and I must have missed something in Configuration.h here is a diff:
```diff
--- D:\Documents\Arduino\Marlin upstream\Configuration.h 6/15/2015  8:21:44 AM
+++ D:\Documents\Arduino\Robo_Dev\Configuration.h 6/15/2015  11:48:49 AM
@@ -46,52 +46,52 @@
 // User-specified version info of this build to display in [Pronterface, etc] terminal window during
 // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
 // build by the user have been successfully uploaded into firmware.
-#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
+#define STRING_CONFIG_H_AUTHOR "(BH, R1 Single Extruder)" // Who made the changes.	--ROBO-BH
 #define STRING_SPLASH_LINE1 BUILD_VERSION // will be shown during bootup in line 1
 //#define STRING_SPLASH_LINE2 STRING_DISTRIBUTION_DATE // will be shown during bootup in line 2
 
@@ -73,79 +73,79 @@
 
 // Optional custom name for your RepStrap or other custom machine
 // Displayed in the LCD "Ready" message
-// #define CUSTOM_MACHINE_NAME "3D Printer"
+#define CUSTOM_MACHINE_NAME "ROBO3D Dev"	//--ROBO-BH
 
 // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
 // You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
@@ -147,153 +147,153 @@
 #define TEMP_SENSOR_1 0
 #define TEMP_SENSOR_2 0
 #define TEMP_SENSOR_3 0
-#define TEMP_SENSOR_BED 0
+#define TEMP_SENSOR_BED 1	//--ROBO-BH
 
 // This makes temp sensor 1 a redundant sensor for sensor 0. If the temperatures difference between these sensors is to high the print will be aborted.
 //#define TEMP_SENSOR_1_AS_REDUNDANT
@@ -170,180 +170,180 @@
 // When temperature exceeds max temp, your heater will be switched off.
 // This feature exists to protect your hotend from overheating accidentally, but *NOT* from thermistor short/failure!
 // You should use MINTEMP for thermistor short/failure protection.
-#define HEATER_0_MAXTEMP 275
+#define HEATER_0_MAXTEMP 330	//--ROBO-BH
 #define HEATER_1_MAXTEMP 275
 #define HEATER_2_MAXTEMP 275
 #define HEATER_3_MAXTEMP 275
-#define BED_MAXTEMP 150
+#define BED_MAXTEMP 120		//--ROBO-BH
 
 // If your bed has low resistance e.g. .6 ohm and throws the fuse you can duty cycle it to reduce the
 // average current. The value should be an integer and the heat bed will be turned on for 1 interval of
@@ -314,337 +314,337 @@
 
 #ifndef ENDSTOPPULLUPS
   // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
-  // #define ENDSTOPPULLUP_XMAX
+  #define ENDSTOPPULLUP_XMAX	//--ROBO-BH
-  // #define ENDSTOPPULLUP_YMAX
+  #define ENDSTOPPULLUP_YMAX	//--ROBO-BH
-  // #define ENDSTOPPULLUP_ZMAX
+  #define ENDSTOPPULLUP_ZMAX	//--ROBO-BH
-  // #define ENDSTOPPULLUP_XMIN
+  #define ENDSTOPPULLUP_XMIN	//--ROBO-BH
-  // #define ENDSTOPPULLUP_YMIN
+  #define ENDSTOPPULLUP_YMIN	//--ROBO-BH
-  // #define ENDSTOPPULLUP_ZMIN
+  #define ENDSTOPPULLUP_ZMIN	//--ROBO-BH
   // #define ENDSTOPPULLUP_ZPROBE
 #endif
 
 // Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
 const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
 const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
-const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
+const bool Z_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.	--ROBO-BH
 const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
[... 170 lines cut ...]
 #define PLA_PREHEAT_HPB_TEMP 70
-#define PLA_PREHEAT_FAN_SPEED 0   // Insert Value between 0 and 255
+#define PLA_PREHEAT_FAN_SPEED 255   // Insert Value between 0 and 255	--ROBO-BH
 
 #define ABS_PREHEAT_HOTEND_TEMP 240
-#define ABS_PREHEAT_HPB_TEMP 110
+#define ABS_PREHEAT_HPB_TEMP 80		//--ROBO-BH
 #define ABS_PREHEAT_FAN_SPEED 0   // Insert Value between 0 and 255
 
 //==============================LCD and SD support=============================
@@ -684,690 +684,690 @@
 // http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
 //
 // ==> REMEMBER TO INSTALL U8glib to your ARDUINO library folder: http://code.google.com/p/u8glib/wiki/u8glib
-//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER	//test machine specific will be removed before any pull request --ROBO-BH
 
 // The RepRapWorld REPRAPWORLD_KEYPAD v1.1
 // http://reprapworld.com/?products_details&products_id=202&cPath=1591_1626```

Uploaded code, connected to printer, reset printer. Made sure that <b>M206</b> had a value of  <b>M206 X0.00 Y0.00 Z-1.20</b> and <b>M851</b> had <b>Z-Probe Offset (mm):  M851 Z-1.20</b>.

Issued a <b>G28</b> which should home all. As I expected, it homed X by doing its little dance at the endstop, then homed Y with the same dance. The printer then raised the extruder 5 mm as per line 496, then nothing. It stops before actually probing Z. I am sure that I missed something in Configuration.h, but I just don't see it. Needless to say, if you issue a G29 after that it also does not do any of the z probing, just zips along to its nine predesignated locations.

I apologize for not being a coder.

thinkyhead, 2015-06-15:
> under the impression that 1.0.2 release was the most stable

@WheresWaldo Well, 1.0.2 (1.0.2-1, etc.) is stable in the sense that it will only get bug fixes, but it isn't entirely bug-free. If you peek at the [Change Log for 1.1](http://www.marlinfirmware.org/index.php/Change_Log) (aka "1.0.3") most of the problems listed as fixed still exist in 1.0.2 and 1.0.2-1.

> forgive me for not completely understanding how GITHub works

When you start new work, always create a new branch first. Don't just modify your "Development" branch. The easiest way to create a new branch is to use the Github client or the github site. If you use the site, you have to do a `git fetch` to grab the new branch. If you use a command-line, you can create a new branch from "Development" with the command `git checkout Development -b my_new_feature`. You will periodically need to update your "Development" branch copy. You can do that with `git fetch upstream`, `git checkout Development`, `git merge upstream/Development`.

WheresWaldo, 2015-06-15:
More testing, but now one issue has cropped up. First found the line that I missed in Configuration.h Line 333 needed to be true. Now that it is working I booted up verified that <b>M206</b> and <b>M851</b> were populated. Issued a <b>G28</b>, dance done but I have no display of Z on Full Graphics LCD Smart Controller. I do know it is resting on the bed at the Z Probe Trigger height. Which previous firmwares would display <b>Z:-001.1</b>, now it displays <b>Z:---.--</b>. Then issuing a <b>G29</b> command I get back the following results after completion of the nine point dance:

```
G29 Auto Bed Leveling
Eqn coefficients: a: 0.00035312 b: 0.00061257 d: -0.06589894
planeNormal x: -0.000353 y: -0.000613 z: 1.000000

Bed Level Correction Matrix:
+1.000000 +0.000000 +0.000353
-0.000000 +1.000000 +0.000613
-0.000353 -0.000613 +1.000000
```

Problem is that I am sitting below the bed again, physically observable that the Z probe is activated. LCD still shows <b>Z:---.--</>. Now if the printer knew that the extruder was below the bed I can issue a <b>G1 Z0</b> and raise it will raise the extruder 1.18 mm, but when I do that instead of moving up 1.18 mm the printer already believes it is at Z0 and does nothing. So other than the display issue (which I am ignoring for now), the development branch appears to work exactly the same as the 1.0.2 release branch with respect to this configuration.

thinkyhead, 2015-06-15:
@WheresWaldo Thanks for the update, it is very helpful! So is it the case that your only Z endstop(s) are the switches that are disengaged when the nozzle is resting on the bed? If so, I might be able to try simulating that on my machine by pretending that the Z endstop/probe is like yours. By the way, which X ends are you using for your leveling system? Can you provide a link to them on Youmagine or Thingiverse?

WheresWaldo, 2015-06-16:
@thinkyhead, This is a standard current version Robo3D R1, the endstops / Z probes are two switches that travel along the two 8 mm threaded rods, the setup pushes the bottom of  the X Idler from the bottom, There are no maximum endstops. 
[image]
Just below the idler on both sides there is a switch that is physically attached to a nut that rides up and down the 8 mm threaded rods. When the head hits the build plate, the rods continue moving until the small micro switches are disengaged.
[image]
The X and Y have similar micro switch endstops, the X_MIN is mounted on the front right side of the X Carriage and hits a plastic stop located on the left X Idler. The Y is located under the bed and is triggered by the frame that holds the glass build plate. The only Thingiverse model available is this on here: http://www.thingiverse.com/thing:729956. The Y and Z axis are correctly modeled but the X is an 'improved' version that moved the micro switch to where the screw stop appears in this collection. One note about the Z Axis endstop switches, they are wired in the NC position and are normally open as long as the extruder is not touching the bed, These switches are located under the left and right X Idlers and are wired in parallel. If any switch disengages it closes the circuit to alert the RAMPS board the the Z Minimum endstop has been hit. This is exactly what happens when the printer tries to lower the extruder head below the top surface of the build plate. It is a funky but workable solution.

WheresWaldo, 2015-06-16:
I should add as a side note, if there is any debug code I can enable or temporarily add to my local copy of the development code I compile and upload to my machine I would be happy to do so and report the findings.

paulusjacobus, 2015-06-27:
Hi,

I believe this nasty bug (i.e. extruder is writing in the air after auto leveling) is a logical error in the routine Marlin_main.cpp. I have created a new branch with my proposed fix. It does now work great on my reprap printer and CURA. For those interested in the bug see https://github.com/MarlinFirmware/Marlin/compare/Release...paulusjacobus:patch-1
my CURA start up routine is
G21        ;metric values
M82        ;set extruder to absolute mode
M107       ;start with the fan offG28 X0 Y0  ;move X/Y to min endstops
G28 X0 Y0 Z0     ;move Z to min endstops
G92 X0 Y0 Z0;
G28 X0 Y0
G29			; ABL
G1 X 0 F(travel_speed)
G1 y 0 F(travel_speed)
G1 Z 15.5 F{travel_speed} ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Printing...

thinkyhead, 2015-06-28:
> a logical error in the routine `Marlin_main.cpp`

We've narrowed it down even more than that! It seems likely that it is one of the changes that shown in this commit: https://github.com/thinkyhead/Marlin/commit/3f0a990360d04a45c11ba6ebdfeff2b16d26bf34

We'd prefer to have this bugfix against the current Development branch, but I will have a look at your patched version of 1.0.2 and see what you found there.

thinkyhead, 2015-06-28:
@Roxy-3DPrintBoard What do you think of this change? https://github.com/MarlinFirmware/Marlin/compare/Release...paulusjacobus:patch-1

thinkyhead, 2015-06-28:
> wouldn't `Z_PROBE_OFFSET_FROM_EXTRUDER` and the offset used by M851 be the same in all cases?

No. `Z_PROBE_OFFSET_FROM_EXTRUDER` is just the default value. The `zprobe_zoffset` variable contains the actual current value, whether read from EEPROM or changed with `M851`.

thinkyhead, 2015-06-28:
I will be re-building my debug branch against the current code shortly. I have to fix up my fork since the Development branch was scrubbed (to my horror) recently in order to remove all merge commits.

WheresWaldo, 2015-06-28:
As soon as you give me an indication that the commit is ready to go I will test this against the <b>dev</b> branch, or should it be the <b>Development</b> branch?

WheresWaldo, 2015-07-03:
I just tested this against the <b>dev</b> branch with <a href="https://github.com/MarlinFirmware/Marlin/commit/23c8c5348f5ab131bd43540fe8d13fa2e20709d6">PR#2351</a> and it appears to be working correctly. I need to do a couple of test prints to make sure. Now another issue has cropped up making this a 100% visual verification, the Full Graphics LCD no longer displays Z position. I will open a new issue for that.

thinkyhead, 2015-07-07:
#2351 has been merged, so the current Development branch no longer has that particular issue. One of the last things remaining is to ensure that the Z position is determined correctly when leveling is completed. A recent code patch was applied to properly (we hope!) account for the distance from the nozzle to the probe, but it still needs thorough testing.

I have yet to fix up my debug branch for the latest Development code, as it's probably going to be an hour of rebasing, but I will post it as soon as I can.

WheresWaldo, 2015-07-07:
I have been using the Development branch since the merge and haven't seen any issues. As soon as you get the debug stuff up I will be more than happy to test it.

WheresWaldo, 2015-07-20:
I am not sure what has changed in the past week, but the current <i>Development</i> (July 20) is now completely borked when issuing <b>G28</b> or <b>G29</b>. Here is what is happening:

When you issue a <b>G28</b> it moves the extruder left and does the slow touch after it hits the endstop. This is expected behavior. The it moves to the front and does the same touch as expected. The LCD then shows the X/Y position as <b>X:120.0</b> and <b>Y:127.0</b> but the head is actually at <b>X:0 Y:0</b>. It doesn't move to the center of the bed. It then moves Z up 5 mm per the information provided by <i>Configuration.h</i>, but the actual Z endstop is never hit or set properly.

If you then issue a <b>G29</b> command it moves to each of the nine test positions. At each position instead of actually testing the bed plane, it simply moves the extruder up 5 mm and then moves onto the next grid point.

After it is all said an done, the extruder head is at the 9th test point but it is now 45 mm higher than at the start, without actually testing anything. As a point of reference, if you roll back <i>Development</i> to around July 7 and compile, upload and test this version the <b>G28</b> and <b>G29</b> commands behave as expected.

Just in case it is a missing <i>Configuration.h</i> item here is the diff between the Robo modified version and the default Marlin version.

```diff
--- D:\Documents\Arduino\Marlin_Development\Configuration.h 7/20/2015  11:48:11 AM
+++ D:\Documents\Arduino\Robo_Development\Configuration.h 7/20/2015  11:50:03 AM
@@ -73,79 +73,79 @@
 
 // Optional custom name for your RepStrap or other custom machine
 // Displayed in the LCD "Ready" message
-// #define CUSTOM_MACHINE_NAME "3D Printer"
+#define CUSTOM_MACHINE_NAME "RoBo3D 42203"  //--ROBO-BH
 
 // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
 // You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
@@ -143,153 +143,153 @@
 //     #define DUMMY_THERMISTOR_998_VALUE 25
 //     #define DUMMY_THERMISTOR_999_VALUE 100
 // :{ '0': "Not used", '4': "10k !! do not use for a hotend. Bad resolution at high temp. !!", '1': "100k / 4.7k - EPCOS", '51': "100k / 1k - EPCOS", '6': "100k / 4.7k EPCOS - Not as accurate as Table 1", '5': "100K / 4.7k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '7': "100k / 4.7k Honeywell 135-104LAG-J01", '71': "100k / 4.7k Honeywell 135-104LAF-J01", '8': "100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT", '9': "100k / 4.7k GE Sensing AL03006-58.2K-97-G1", '10': "100k / 4.7k RS 198-961", '11': "100k / 4.7k beta 3950 1%", '12': "100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT (calibrated for Makibox hot bed)", '13': "100k Hisens 3950  1% up to 300°C for hotend 'Simple ONE ' & hotend 'All In ONE'", '60': "100k Maker's Tool Works Kapton Bed Thermistor beta=3950", '55': "100k / 1k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '2': "200k / 4.7k - ATC Semitec 204GT-2", '52': "200k / 1k - ATC Semitec 204GT-2", '-2': "Thermocouple + MAX6675 (only for sensor 0)", '-1': "Thermocouple + AD595", '3': "Mendel-parts / 4.7k", '1047': "Pt1000 / 4.7k", '1010': "Pt1000 / 1k (non standard)", '20': "PT100 (Ultimainboard V2.x)", '147': "Pt100 / 4.7k", '110': "Pt100 / 1k (non-standard)", '998': "Dummy 1", '999': "Dummy 2" }
-#define TEMP_SENSOR_0 1
+#define TEMP_SENSOR_0 1		//--ROBO-BH for E3Dv6 = 5 for Hexagon = 1
 #define TEMP_SENSOR_1 0
 #define TEMP_SENSOR_2 0
 #define TEMP_SENSOR_3 0
-#define TEMP_SENSOR_BED 0
+#define TEMP_SENSOR_BED 1	//--ROBO-BH
 
 // This makes temp sensor 1 a redundant sensor for sensor 0. If the temperatures difference between these sensors is to high the print will be aborted.
 //#define TEMP_SENSOR_1_AS_REDUNDANT
@@ -170,176 +170,176 @@
 // When temperature exceeds max temp, your heater will be switched off.
 // This feature exists to protect your hotend from overheating accidentally, but *NOT* from thermistor short/failure!
 // You should use MINTEMP for thermistor short/failure protection.
-#define HEATER_0_MAXTEMP 275
+#define HEATER_0_MAXTEMP 320	//--ROBO-BH
 #define HEATER_1_MAXTEMP 275
 #define HEATER_2_MAXTEMP 275
 #define HEATER_3_MAXTEMP 275
@@ -205,225 +205,216 @@
   #define PID_INTEGRAL_DRIVE_MAX PID_MAX  //limit for the integral term
   #define K1 0.95 //smoothing factor within the PID
 
-  // If you are using a pre-configured hotend then you can use one of the value sets by uncommenting it
+//	RoBo3D R1  
-  // Ultimaker
+    #define  DEFAULT_Kp 28.67	//--ROBO-BH
-  #define  DEFAULT_Kp 22.2
+    #define  DEFAULT_Ki 2.39	//--ROBO-BH
-  #define  DEFAULT_Ki 1.08
+    #define  DEFAULT_Kd 86.01	//--ROBO-BH
-  #define  DEFAULT_Kd 114
+
-
+// FIND YOUR OWN: "M303 E0 C8 S200" to run autotune on the hotend at 200 degreesC for 8 cycles.	//--ROBO-BH
-  // MakerGear
-  //#define  DEFAULT_Kp 7.0
-  //#define  DEFAULT_Ki 0.1
-  //#define  DEFAULT_Kd 12
-
-  // Mendel Parts V9 on 12V
-  //#define  DEFAULT_Kp 63.0
-  //#define  DEFAULT_Ki 2.25
-  //#define  DEFAULT_Kd 440
[... 204 lines cut ...]
   // To disable EEPROM Serial responses and decrease program space by ~1700 byte: comment this out:
@@ -614,622 +599,607 @@
 // @section temperature
 
 // Preheat Constants
-#define PLA_PREHEAT_HOTEND_TEMP 180
+#define PLA_PREHEAT_HOTEND_TEMP 210	//--ROBO-BH
 #define PLA_PREHEAT_HPB_TEMP 70
-#define PLA_PREHEAT_FAN_SPEED 0   // Insert Value between 0 and 255
[... cut ...]
@@ -681,687 +666,672 @@
 // http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
 //
 // ==> REMEMBER TO INSTALL U8glib to your ARDUINO library folder: http://code.google.com/p/u8glib/wiki/u8glib
-//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
 
 // The RepRapWorld REPRAPWORLD_KEYPAD v1.1
 // http://reprapworld.com/?products_details&products_id=202&cPath=1591_1626
```

Please advise what you need me to test to investigate this issue.

WheresWaldo, 2015-07-20:
Resetting <b>Z_PROBE_OFFSET_FROM_EXTRUDER</b> to 0 does not change the behavior listing in the comment above. I should mention that <b>M851</b> was set to -1.0 mm during the above two tests. Changing <b>M851</b> to 0 and storing that value changes the results of the test. Now when a <b>G28</b> command is issued the final resting place is listed as <b>X:120.0 Y:127.0 Z:0</b>. But this is also not correct as the 0 location along the Z axis is still 0.1 to 0.2 mm below the surface of the print bed. It is behaving as if the <b>Z_PROBE_OFFSET_FROM_EXTRUDER</b> is totally ignored as what was expected is that after the G28 command, all axes are homed but the probe should be up 1.2 mm. This is evident by the results of the <b>M119</b> command issued immediately after the <b>G28</b> command:
```
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
z_probe: open
```
Following this command with a <b>G29</b> command results in the final Z position of 0 which Marlin thinks is at the top of the bed, when in fact it is at the Z probe endstop trigger position. Regardless of what <b>Z_PROBE_OFFSET_FROM_EXTRUDER</b> or <b>M851</b> (zprobe_offset) are set to, both those appear to be completely ignored for anything except display purposes, as the probe always ends up at the z probe trigger location thinking it is at <b>Z:0</b> or <b>Z:(0 + zprobe_offset)</b> rather than the top of the bed at the <b>Z:0</b> location.

WheresWaldo, 2015-07-20:
Is this an acceptable workaround. Since I know that we end up at the Z probe endstop trigger point and that is only 0.1 or 0.2 below the non trigger point, if I place a positive number in <b>M851<b> such as <b>M851 Z0.2</b> and then do both the <b>G28</b> and <b>G29</b> commands. the result would be that the final Z position would be -0.2 mm, then a subsequent <b>G1 Z0</b> command would actually place me at the top of the print bed.

So I guess my basic questions are: has the use of <b>M851<b> changed to require a positive number for negative offset? Has the use of <b>Z_PROBE_OFFSET_FROM_EXTRUDER</b> been superseded by the <b>M851</b> command?

paulusjacobus, 2015-07-21:
@whereswaldo back in June I figured out a part of the bugs surrounding the Autobed levelling issue (mainly air printing). The code has further evolved from that first fix and somewhere various issues have been introduced or discovered (constants and add/subtract sign issues) plus various machine variants. Your questions are very valid but I don't think we can answer them at this stage as many of us still try to understand the code. If we could answer this question then the bugs can be fixed. In my opinion Z_PROBE_OFFSET_FROM_EXTRUDER is the first setting that the firmware uses. M851 is the fine tuning to handle various filaments (some don't stick well and need to be printed with a very narrow hotend/bed gap). 
My experiences are that version 1.0.2 with a fix is working very reliable but I still need to do minor adjustments with M851 depending on the filament. I included that in my G start code sequence (i.e. G28 X0 Y0 Z0 followed by M851 Z-6.70 and G29. My Z_PROBE_OFFSET_FROM_EXTRUDER is 6.90). 
Also the ABL feature requires proper homing which a lot of people don't understand and they post this as an issue or ask it to be part of the ABL feature. (G codes in nature are single tasks)

The main question which was posted before June was should we scrap the code and start over or try to fix it. I think to iron out the bugs is still preferable now we have a workable ABL feature although it requires a lot of time to debug it. Maybe we need a small task force to get this sorted out.

WheresWaldo, 2015-07-21:
@paulusjacobus I do understand the use of M851 in gcode, but my thinking is slightly different on this matter, ABL and offsets should be to get the printer to a known mechanically sound and accurate state. I know a lot of people use M851 and M565 previous to this to adjust for bed adhesion, not sure that was ever intended to be the best way to do things.

Example, If I have an object model that depends on a layer height that must be precise, using M851 to "adjust for adhesion" can result in a part, that might print perfect and stick just well enough, but be perfectly unusable. So my understanding is that these adjustments are made so we can gain accuracy with our prints. Whether I print a part in PLA, ABS, Nylon, PolyCarb, PET or anything that comes down the line, if I need 20 mm in height I expect 20 mm in height regardless. Otherwise we are relegating 3D printing to occasional trinkets or toys and leaving all the accurate work to the 3D print services. That is not how I want to use my printer, and I feel I am not on the fringes. If 3D printing is to ever become mainstream it has to work on a consumer level, which means it not only has to be a precise machine, it needs accuracy also as well as all the other ease of use requirements that consumer products bring. In my opionion, your slicer is the place to adjust first layer for adhesion, not offsets to extruder height. To that end I am trying to document every thing I find out about making this as accurate as it is precise. As you may have noticed I am trying to give you as much information as possible when testing various sections of code. It not like I am saying "it's broken, fix it!" without any background information.

Now if this is not the development intent of measurements and offsets, to add accuracy to the already built in precision, then I will work around these issues. You can continue to use your printer the way you see fit and I will try to work around the limitations inherent in all projects as ambitious as this. I know you are probably tired reading this long-winded response, but if I might add, it seems to me that the bottom line is that there are no clear objectives set out or documented in how or what these offsets are used for. This leads to different contributors making changes based on their ideas of how it should work. Sorry if this is a little frank, please note that I hold all the contributors here in the highest regard. I don't know if you are from a software background or a mechanical background. Mine is mechanical with a technical understanding of the challenges faced. With this background I tend to look at these as more machine than computing device. I see manufacturers striving to use the best parts for a given price point, these all come with a level of mechanical precision, now my intention as well as other non-programmer types, is to use the adjustments given, whether they be dials on a panel or #defines in a configuration file and make it as accurate as possible.

Does that mean scrapping the current code and starting over, or trying to fix what is there? I will never be the person to answer that. That decision is for the community of code contributors. All I can give is my expectations, opinion and the results of my testing.

paulusjacobus, 2015-07-21:
@WheresWaldo

 "I know a lot of people use M851 and M565 previous to this to adjust for bed adhesion, not sure that was ever intended to be the best way to do things."

That's the point we don't know what the developer intended, the code is not well documented, that's why it is so difficult to fix this.
Sure we all strive to perfect this technology but it is open source and dependent on willingness of complete strangers to invest their time to develop this.

thinkyhead, 2015-07-29:
Looks like right now we have a minor regression with `G28` not deploying the Z probe – probably due to a bad precompiler condition surrounding `deploy_z_probe` in `homeaxis`. Hopefully this will be worked out very soon, but in the meantime be careful if using a Z probe to home and testing the latest code.

Otherwise, the probe final position appears to be much better in the latest Development code.

WheresWaldo, 2015-07-29:
I was going to test the Development branch tomorrow and report my findings. I have been waiting between testing. I have been currently running Development code from July 20. Other than some irregularities with ABL, the rest of the code seems solid.

thinkyhead, 2015-07-31:
Turns out that you must have `DEACTIVATE_SERVOS_AFTER_MOVE` enabled or the probe won't deploy / stow in certain instances. I'm asking around as to whether that makes any sense. The number of servo+endstop+probe settings is starting to get confusing…

thinkyhead, 2015-07-31:
@WheresWaldo Try my [debug_G29](https://github.com/thinkyhead/Marlin/tree/debug_G29) branch and post the log output. It should reveal why your homing and leveling are breaking down.

WheresWaldo, 2015-07-31:
Will try both today and report back.

WheresWaldo, 2015-07-31:
Found one more line I had overlooked before, will retest shortly.

thinkyhead, 2015-08-12:
I've updated [my debug branch](https://github.com/thinkyhead/MarlinDev/tree/debug_G29) with the latest code (and moved it under MarlinDev), which has received some adjustments to leveling in recent days.

And, if you wait another few days before testing, there are even more interesting adjustments coming that may affect this. Keep an eye on https://github.com/MarlinFirmware/MarlinDev/pulls for relevant updates.

paulusjacobus, 2015-08-12:
Thanks @thinkyhead will be looking forward to test the new code.

WheresWaldo, 2015-08-12:
I too will test this. I anxiously await the additional PRs. So far all is working with the exception that M851 mysteriously changed from negative (expected) to positive. I will run any tests you want and post the results from the debug code. I am currently running dev code from 10-AUG-2015 (from the correct GIT).

thinkyhead, 2015-08-13:
> M851 mysteriously changed from negative (expected) to positive

@WheresWaldo It should remain negative after `M851 Z-0.72`. Does it change spontaneously?

WheresWaldo, 2015-08-13:
What I meant was that using M565 the value needed to be entered as a
negative offset, which makes total sense since the endstop is triggered
after the extruder touches the bed. The extruder IS the probe. M851 at
first was the same way. Somewhere about two weeks ago or so the number had
to be input as positive in M851 or it didn't work, totally out of the blue
and unexpected. This printer does not have a probe and trigger that
activates separately. It is a solution to installing extruders of different
lengths, the offset will remain the same regardless the length of the
extruder, since the extruder is the actual probing instrument.

thinkyhead, 2015-08-14:
@WheresWaldo That's interesting. Marlin doesn't actually implement `M565` (set probe offset X, Y, and Z) but only `M851`. Only Smoothie understands `M565` at this time.

With your probe setup, using `M851` with a positive value is correct, and should give you the correct Z coordinate if you use that probe to home the Z axis with `G28`. Then, for `G29` you just need to have large enough `Z_RAISE_*` values to make sure the probe is disengaged between probes.

WheresWaldo, 2015-08-14:
@thinkyhead Interesting, the Robo implementation of 1.0.0 uses `M565` so that is my frame of reference. In 1.0.0 the number entered was negative. When I tried 1.0.2 Release, it too was entered as a negative number. Moving to 1.0.3 Dev and it became `M851` and was still a negative entry. Sometime in the last month it became a positive entry. But if you say it should be positive to effect a final negative offset, then I am okay with that, as long as it is documented and stays that way.

BTW, I can finally see some clarity in all the improvements all the developers are making and I really appreciate the work being done. I am awaiting your latest debug build to ferret out the remaining issues (if any) in the ABL code.

thinkyhead, 2015-08-14:
> it should be positive to effect a final negative offset

No, `M851` sets `zprobe_zoffset` directly. Thus, it should be positive to give a positive offset. A positive offset tells Marlin that the nozzle bed contact Z is higher than the probe trigger point Z. Most probes have a negative offset, meaning that they are lower than the nozzle when deployed, or, conversely, that their trigger point Z will be a higher number than the nozzle's Z when it's touching the bed.

To give some examples:
- For a probe that is 5mm lower than the nozzle you will have a `zprobe_zoffset` of -5.0 (as set by `Z_PROBE_OFFSET_FROM_EXTRUDER` or `M851`). If this Z probe is used for `G28 Z` homing, when it gets triggered, the `current_position[Z_AXIS]` will be set to `-zprobe_zoffset`, which is +5.0. The nozzle will be 5mm from the bed.
- For a probe that is 2mm higher than the nozzle (such as the "floating Z axis" switch style) you will have a `zprobe_zoffset` of +2.0. When this Z probe is used for `G28 Z` homing, when it gets triggered, the `current_position[Z_AXIS]` will be set to `-zprobe_zoffset`, which is -2.0. The nozzle will be 2mm "below" the bed (or the floating Z axis will have been lifted up by 2mm, disengaging the contact switch).

Wackerbarth, 2015-08-15:
@thinkyhead -- Although your description is correct, I feel that the design is confusing and should be changed to take the position of the nozzle relative to probe and yield a positive value, which is the usual case, "The position of the nozzle when the probe contacts the bed"

thinkyhead, 2015-08-15:
@Wackerbarth The user-configurable options and accompanying GCodes have always specified the __probe position__ relative to the __nozzle position__, and I think this is the correct approach because all machines have nozzles, but not all machines have probes. If the probe is _10mm left_ of the nozzle, its X offset is specified as `#define X_PROBE_OFFSET_FROM_EXTRUDER -10`. If the probe is _5mm below_ its Z offset is specified as `#define X_PROBE_OFFSET_FROM_EXTRUDER -5`. I don't see how this is confusing from a user configuration point-of-view, unless there is a pre-existing confusion. These are not intended to indicate "the position of the nozzle" – that concept is some-steps-removed from the relative position of the probe.

I'm not fond of the idea of now reversing the sign of `X_PROBE_OFFSET_FROM_EXTRUDER`, `Y_PROBE_OFFSET_FROM_EXTRUDER`, and `Z_PROBE_OFFSET_FROM_EXTRUDER` (and `zprobe_zoffset` – or the 3-element array that will replace it when we implement `M565` which also takes nozzle-relative values), nor do I feel that we should reverse the sign given in `M851` since it maintains parity with the nozzle-relative configuration options.

Wackerbarth, 2015-08-15:
I have no problem using X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER as we do now.
In reality, it can be shown mathematically that these values are significant only for determining the area in which we probe. In the calculation of the bed tilt, the values of these terms gets canceled and they do not affect the result.

The issue is with the Z component. Although I understand the consistency in describing the probing in terms of the probe location in terms of the Nozzle coordinate system, that is not the only way to describe it. It is perfectly valid, and more intuitive, to describe thinks in terms of the BED. In particular, we describe the position of the nozzle in terms relation to the registration indicators (limit switches).

In this description frame, we might say that the nozzle is at [-50mm, ..., ...] when the carriage is against the X-min-limit switch.

For the Z component, we would say that the nozzle is [..., ..., Z_PROBE_HEIGHT] ABOVE the bed when the probe triggers by contact with the bed. Except for an unusual case, such as using the nozzle to depress the bed as a probing mechanism, This is a positive number. Most humans think in terms of (unsigned) distances rather than coordinate systems.

I recognize that choosing a particular way to describe a situation is somewhat arbitrary since there are mathematically simple transformation operations which convert from one scheme to the other (in either direction). Choosing a particular scheme INTERNALLY may simplify some of the logic. However, when the user is describing his configuration, we need to be expressing it in the terms which are "user friendly".

I contend that MOST users do not think in "nozzle coordinates". and would find a POSITIVE distance easier to understand.

Your choice to describe the probe position in nozzle coordinates is an ARBITRARY method of doing so. That CHOICE has already created quite a bit of confusion. It reversed the sign of terms previously used. I feel that that choice should be changed and we should revert to a description space which restores the intuitive description.

WheresWaldo, 2015-08-15:
I just want to verify that everyone understands, with the Robo printers the nozzle is the probe so a final probe position is also the final nozzle position. The end stop is triggered after the probe/nozzle hits the bed.

I just need to make sure I understand, that the probe and the end stop are actually two different things and they are not being lumped together. My understanding and in my particular case, the probe/nozzle hits the bed but the end stop is not triggered until the z axis is lowered by a specified amount. Since the two are tied together, the result is that the location specified is where the end stop is triggered, hence the need for the offset. My trigger is below the probe/nozzle so my offset used to be negative. As I mentioned sometime in the last month it turned positive. If there is no offset the end stop trigger position becomes the 0 position in the Z axis. Once applied the end stop trigger position now shows a Z axis height below the level of the bed.

@thinkyhead `M206 Z0` is always set correctly.

thinkyhead, 2015-08-15:
@WheresWaldo Since your min endstop only triggers after the nozzle hits the bed and lowers a little more, then you could actually use `M206` to specify that amount – and it should be a negative value in that case. Or, you should enable `MANUAL_HOME_POSITIONS` and set `MANUAL_Z_HOME_POS` to that same negative value and leave `M206 Z` (`home_offset[Z_AXIS]`) set to 0.

Which value is it that you are currently setting to a positive value, which used to be negative?

WheresWaldo, 2015-08-16:
I am only using `M851`, all other settings are 0.

thinkyhead, 2015-08-16:
@WheresWaldo I see. Well actually that value _is_ used in `set_axis_is_at_home` which follows any homing operation. It's subtracted from the current Z position, like so:

    if (axis == Z_AXIS) current_position[Z_AXIS] -= zprobe_zoffset;

Thus, after homing, your current Z position will be (for example) -0.5mm if you used `M851 Z0.5`. That positive value is correct, as it specifies the distance from the nozzle to the probe (in this case it's logically "above" the nozzle).

WheresWaldo, 2015-08-16:
As long as you guys figure it out I will go along with whatever the decision is. I am using 1.1.0 Dev as a daily driver and it seems to be working well. I can live with any number regardless of sign.

thinkyhead, 2015-08-17:
I'm happy with how it works now. All offsets are as they are intended to be. So if the sign changed someplace recently (and I'm not convinced it has) then that would have been a bug correction.

closed 2015-08-17 by thinkyhead