MarlinFirmware/Marlin #2299: Servo will not pull up during homing / bed leveling

LazyPrinter, 2015-06-15:
I've just installed my custom design E3D V6 mount onto my CTC creator dual (makerbot clone). The problem I'm facing is, when I power the machine on, the servo will go to its default position (Which is up). When I start G28, the probe will go to the center of the bed and bring the probe down. After finishing, the servo will not pull the probe back up, leaving it down. Autobed leveling is the same. 

I've noticed that the screen is dimming when the servo is supposed to move. Funny thing is, if I issue an M280 P0 180 /50, the servo will move freely along with the probe.

I've ran the servo from an external power supply with no difference.

Any ideas?

thinkyhead, 2015-06-15:
Are you using the latest Development code? It's easiest for us to debug starting there (rather than from 1.0.2-1, for example).

So, are you able to move the probe with `M280`? What value are you using for `PROBE_SERVO_DEACTIVATION_DELAY`? Please post your configuration file here (or on Pastebin, or as a Github gist) so we can make sure everything is copacetic.

LazyPrinter, 2015-06-15:
Thinking of it.. Since I got two issues up at once, I'll get the dev code running on my printer and make progress on both. Will repeat the same steps with the latest dev version.

LazyPrinter, 2015-06-15:
Fixed.

When I designed and assembled my hotend mount, I forgot that the rack would be on the opposite of the servo pinion gear. Old code:

#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 50, 0} // X,Y,Z Axis Extend and Retract angles

This would explain the LCD dimming. New code

#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 10,180} // X,Y,Z Axis Extend and Retract angles

closed 2015-06-15 by LazyPrinter