MarlinFirmware/Marlin #2305: Remapping heater & heated bed pins

wes1007, 2015-06-17:
The mosfets on my RAMPS 1.4 blew and i have made an external unit. I would like to run this unit of D63 for H1 D40 for H2 and D42 for HBP (EEB config)

I have verified that everything on my end works but I just can seem to get marlin configured for this.

My Pins.h file
http://pastebin.com/emkK1cWv

Made changes at line 683, 689 and 726 for the above mentioned pins.

There isn't a lot of information around about modifying pins.

Nprod, 2015-06-17:
Maybe a silly question, but why not just replace the mosfets?

wes1007, 2015-06-17:
I wanted them on a separate board so all my connections are in line on the enclosure I am using.
I wanted to add some decent heat sinks which isnt possible on the RAMPs
Wanted more than 3 FETs (currently running 4) http://imgur.com/vTugDDf

wes1007, 2015-06-17:
I get the feeling this isnt really possible? :/

wes1007, 2015-06-17:
So ive done a whole lot of testing. Firstly it does work the way it should.

I have been running my ramps board with no Thermistors on it and it was obviously shutting off because of that.

Secondly i now have this weird issue where the pin set has the heated bed pin stays on continuously. was set to pin 8 (default) and it would register 5V no matter if the bed was turned on or off in pronterface. I then changed the heated bed pin to pin 4 (on the servo pins) and had exactly the same problem, however some magical unicorns have now turned pin 8 off. Them magical unicorns have just shown that both my ramps board and the Arduino are fine.

wes1007, 2015-06-17:
Note to self: Order more thermistors for bench testing...

If no thermistor is connected weird things happen. It seems if the temp is below 0 that the pin for the heater that temperature sensor is assigned to will turn on. 

Another thing. Setting the thermistor to option 0 (IE none) does not let you test and gives the same problems as above..

eboston, 2015-06-17:
@wes1007 Get yourself some 100K potentiometer.  That way you can vary the resistance to simulate heating and cooling.

And to test without a thermistor, use type 998 and 999, if I remember correctly.  They put the temp at a fixed value.

wes1007, 2015-06-17:
/facepalm... Pots... Why didn't I think of that. I'll do some tests in the morning (currently 11:30pm) 

I'll have a look through the code for those two types

AnHardt, 2015-06-17:
@wes1007 
You are still using the huge unsplited pins.h file. That means, you are using an relatively old version of Marlin. It would be much easier for us to help you if you could use one of our tagged versions or one of our actual branches.

wes1007, 2015-06-17:
As far as I know that's v1.02 :/ I didn't know there was anything newer....

ntoff, 2015-06-18:
The old pins.h is kind of... broken. 

I ended up splitting it up myself and just deleted everything that wasn't RAMPS related. After that, redefining pins was as simple as changing their numbers.

wes1007, 2015-06-20:
It really is a pain to edit anything in there. It would be nice to have each board as a separate file but that means more work and includes...

ntoff, 2015-06-20:
The dev version does exactly that.

wes1007, 2015-06-20:
Then I'll be loading that up and trying it out. Where exactly is this dev version you speak of?

AnHardt, 2015-06-20:
How to get the `Development` branch:
https://github.com/MarlinFirmware/Marlin/issues/2293#issuecomment-112041478
and following essages

ntoff, 2015-06-20:
also you can bookmark https://github.com/MarlinFirmware/Marlin/tree/Development to check on updates

closed 2015-09-25 by AnHardt