MarlinFirmware/Marlin #2308: Z probe out. bed, WHY?

Hat3, 2015-06-17:
Hi

i'm using a 3Drag 1.2, with last release of firmware

i followed this guide http://www.marlinfirmware.org/index.php/Bed_Auto_Leveling

But after set X and Y to 0, i receive "echo:Z probe out. bed", so i cant set home position with G92 X0 Y0 Z0

This is my conf

#define ENABLE_AUTO_BED_LEVELING // Delete the comment to enable (remove // at the start of the line)
#define Z_PROBE_REPEATABILITY_TEST  // If not commented out, Z-Probe Repeatability test will be included if Auto Bed Leveling is Enabled.

#ifdef ENABLE_AUTO_BED_LEVELING

// There are 2 different ways to pick the X and Y locations to probe:

//  - "grid" mode
//    Probe every point in a rectangular grid
//    You must specify the rectangle, and the density of sample points
//    This mode is preferred because there are more measurements.
//    It used to be called ACCURATE_BED_LEVELING but "grid" is more descriptive

//  - "3-point" mode
//    Probe 3 arbitrary points on the bed (that aren't colinear)
//    You must specify the X & Y coordinates of all 3 points

  #define AUTO_BED_LEVELING_GRID
  // with AUTO_BED_LEVELING_GRID, the bed is sampled in a
  // AUTO_BED_LEVELING_GRID_POINTSxAUTO_BED_LEVELING_GRID_POINTS grid
  // and least squares solution is calculated
  // Note: this feature occupies 10'206 byte
  #ifdef AUTO_BED_LEVELING_GRID

    // set the rectangle in which to probe
    #define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 130
    #define BACK_PROBE_BED_POSITION 170
    #define FRONT_PROBE_BED_POSITION 20

     // set the number of grid points per dimension
     // I wouldn't see a reason to go above 3 (=9 probing points on the bed)
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // not AUTO_BED_LEVELING_GRID
    // with no grid, just probe 3 arbitrary points.  A simple cross-product
    // is used to esimate the plane of the print bed

      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 180
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 170
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID

  // these are the offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER -25
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -29
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12

  #define Z_RAISE_BEFORE_HOMING 2       // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case

  #define XY_TRAVEL_SPEED 6000         // X and Y axis travel speed between probes, in mm/min

  #define Z_RAISE_BEFORE_PROBING 10    //How much the extruder will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 10  //How much the extruder will be raised when traveling from between next probing points

  //#define Z_PROBE_SLED // turn on if you have a z-probe mounted on a sled like those designed by Charles Bell
  //#define SLED_DOCKING_OFFSET 5 // the extra distance the X axis must travel to pickup the sled. 0 should be fine but you can push it further if you'd like.

  //If defined, the Probe servo will be turned on only during movement and then turned off to avoid jerk
  //The value is the delay to turn the servo off after powered on - depends on the servo speed; 300ms is good value, but you can try lower it.
  // You MUST HAVE the SERVO_ENDSTOPS defined to use here a value higher than zero otherwise your code will not compile.

//  #define PROBE_SERVO_DEACTIVATION_DELAY 300

//If you have enabled the Bed Auto Leveling and are using the same Z Probe for Z Homing,
//it is highly recommended you let this Z_SAFE_HOMING enabled!!!

  #define Z_SAFE_HOMING   // This feature is meant to avoid Z homing with probe outside the bed area.
                          // When defined, it will:
                          // - Allow Z homing only after X and Y homing AND stepper drivers still enabled
                          // - If stepper drivers timeout, it will need X and Y homing again before Z homing
                          // - Position the probe in a defined XY point before Z Homing when homing all axis (G28)
                          // - Block Z homing only when the probe is outside bed area.

  #ifdef Z_SAFE_HOMING

    #define Z_SAFE_HOMING_X_POINT (X_MAX_LENGTH/2)    // X point for Z homing when homing all axis (G28)
    #define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/2)    // Y point for Z homing when homing all axis (G28)

  #endif

  #ifdef AUTO_BED_LEVELING_GRID	// Check if Probe_Offset * Grid Points is greater than Probing Range
    #if X_PROBE_OFFSET_FROM_EXTRUDER < 0
      #if (-(X_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (RIGHT_PROBE_BED_POSITION - LEFT_PROBE_BED_POSITION))
	     #error "The X axis probing range is not enough to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS"
	  #endif
	#else
      #if ((X_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (RIGHT_PROBE_BED_POSITION - LEFT_PROBE_BED_POSITION))
	     #error "The X axis probing range is not enough to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS"
	  #endif
	#endif
    #if Y_PROBE_OFFSET_FROM_EXTRUDER < 0
      #if (-(Y_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (BACK_PROBE_BED_POSITION - FRONT_PROBE_BED_POSITION))
	     #error "The Y axis probing range is not enough to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS"
	  #endif
	#else
      #if ((Y_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (BACK_PROBE_BED_POSITION - FRONT_PROBE_BED_POSITION))
	     #error "The Y axis probing range is not enough to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS"
	  #endif
	#endif

	
  #endif
  
#endif // ENABLE_AUTO_BED_LEVELING

brainscan, 2015-06-17:
Are you trying to go through the set up guide or have you completed all the steps? 
I'm trying to work out which stage you're at, and what you mean by setting X and Y to 0? Do you mean you homed X and Y or set their position with G92? 
Is the probe over the bed when you get this error?
Sorry for all the questions.

Hat3, 2015-06-17:
Using Repetier i homed X and Y, when i try to homed Z i reveive the message "echo:Z probe out. bed".
i cant set G92 X0 Y0 Z0

AnHardt, 2015-06-17:
@Hat3 
Usually the error describes, that the probe can not reach the coordinates you told to go.
A common error is to believe that you have to put in the coordinates of the nozzle. You actually have to put in the coordinates for the PROBE.
To check that manually we additionally need to know your `// Travel limits after homing`.
Together with the *_PROBE_OFFSET_FROM_EXTRUDER values we can than determine the probable area.

Hat3, 2015-06-18:
@AnHardt 

  // Travel limits after homing
  #define X_MAX_POS 200
  #define X_MIN_POS 0
  #define Y_MAX_POS 200
  #define Y_MIN_POS 0
  #define Z_MAX_POS 210
  #define Z_MIN_POS 0

i'll try to change
  #define X_PROBE_OFFSET_FROM_EXTRUDER  0  //-25
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -34 //-29
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -6  //-12

Nothing, same message 
08:35:46.263 : echo:Z probe out. bed

Hat3, 2015-06-18:
ok i understand..

  #define X_PROBE_OFFSET_FROM_EXTRUDER 47  //-25
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 0 //-29
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -6  //-12

negative value is used for probe and servo on the left of extruder, i have a probe on the right

closed 2015-06-18 by Hat3

thinkyhead, 2015-06-22:
I'm cleaning up that instructions page. Please have another look and let us know if there are any errors in the updated version.

nkuck, 2016-07-13:
I'm having the same Z probe out. bed message after installing Marlin fw. Was ok before. Help? Thanks.

thinkyhead, 2016-07-14:
@nkuck Same idea as above. Make sure that your probe can actually reach the given points. For example, if your bed goes from 0 to 200 in the X and your probe is 10mm to the right of the nozzle, you can probe all the way to 200 on the right, but only as far as 10 on the left. The same applies to Y.

nkuck, 2016-07-14:
Hmmm, will take a look at it. Thanks. 

Live simply!
You can only take with you what you have given.

nkuck, 2016-07-14:
Re-reading your message, I don't have a probe (for auto levelling) installed yet, if that's what you are referring to. So does that mean my settings suggest that I do?

Live simply!
You can only take with you what you have given.

thinkyhead, 2016-07-14:
@nkuck You won't see that message unless you have `AUTO_BED_LEVELING_FEATURE` enabled.

nkuck, 2016-07-14:
Yes, I did enable auto bed leveling, but no probe installed yet. Are you thinking that if I install the probe that the error will go away? Sorry, but I haven’t had the opportunity to configure this feature yet.

thinkyhead, 2016-07-14:
@nkuck Once you configure the settings properly the error will go away. Change the probing limits to something that fits your nonexistent probe. Increase the min and decrease the max, towards the center, until the errors go away.

EWGRage, 2019-01-10:
Hi every one. I am new to the 3d printing scene and am having the z probe error and have no clue what to do to even start. I have tried different software with a gcode tool but none seem to work for my anet a8 only cura which does not seem to have a place for me to send gcode to. can any one give me pointers on where and or how to handle these issues. Thank you all.