MarlinFirmware/Marlin #2313: Problem with M666 Gcode

Boogy62, 2015-06-18:
Hi, I've a problem, when I type M666 in Pronterface I haven't any answer. So when I want to change endstop offsets, it does nothing. Thanks if you have an idea

AnHardt, 2015-06-19:
Do you have a delta, or a machines with 2 z-end-stops?
For a delta no output is expected.

Boogy62, 2015-06-19:
I have a delta, and in some videos there is an output, like here https://www.youtube.com/watch?v=tDLbqLve128
I try for instance M666 X+1.8 but it does nothing. Must I change manually the position of the endstop ?

thinkyhead, 2015-06-22:
This would be your problem:

    if (endstop_adj[axis] * axis_home_dir < 0) { . . . }

You have to set it in the opposite direction of the endstop. On a delta that means you can only set a negative value (moving away from the endstop).

thinkyhead, 2015-06-22:
@AnHardt Can you have a glance at the way `endstop_adj` is being used in the code and see if you concur with my opinion that it's not exactly doing the right thing? Specifically, it seems to be applying to the cartesian (pre-kinematic) XYZ and not to the ABC towers.

AnHardt, 2015-06-27:
@thinkyhead 
__I__ can't __find__ an error here. (but neither i'm an expert for the motion code, nor i can test deltas)

HedronUser, 2016-02-04:
Make sure your eeprom chitchat and eeprom settings are enabled

closed 2016-02-19 by thinkyhead