MarlinFirmware/Marlin #2314: Feature request : Auto home

plgarcia, 2015-06-18:
Auto home by menu moves first X and Y axis and when done moves Z axis.

But if the nozzle touch the bed, the nozzle is rubbing al the way long to the X0/Y0 position, what may damage the nozzle, and the film on the bed.

It woud be great to move the nozzle up for a couple of milimeters before homing.

Nandox7, 2015-06-19:
Or like it happens to me a lot it goes near the edge of the bed taking with it all that it finds like the clamps I have to hold the glass.

atunguyd, 2015-06-19:
This should not happen if your printer is setup with Z axis endstops. The endstops should have already mad sure that the Z axis is never below 0 height.

plgarcia, 2015-06-19:
In a perfect world you are right.
The fact is that, before I add alumium profiles on the acrylic the support of the bed was going up and down depending on the température. And I scratch the film.

The fact is that during the procedure the Z axis is raised, when the endstop is on. It shoud be moved earlyer.

Futher more it may happen that pla leaks after the last print. If the hotend id cold there may by a gloubibouga on the nozzle.

plgarcia, 2015-06-19:
But I aggree that the nozzle should not go outside the bed with the marlin if the endstop are properly positionned.

In fact I thins the software shoud accept that.

Why the endstops could not be outside the bed, if the relative position of the bed is known.
On machines that have many hotends that coud be of interrest the will is to print on the entire surface of the bed with both the hotends. At some pour one or the other has to go outside the bed.
One of the problem id that on many machines the heads of the screw holdings the bed are abode the bed, ans therefore that may cause problems.

thinkyhead, 2015-06-22:
We already have a feature request for extending `SAFE_HOMING` to all printers, and not just those with probes. If anyone can find the Issue number, please post it here and close this.

plgarcia, 2015-06-24:
That may happe, here the proof.
I damaged the bed as I was trying to set the Z endstop.
And the first test was to low so the nozzle was touching the bed.
I moved the endstop a bit higher, and made auto home again.
The nozzle moved x direction and y direction just for releasing the endstops and then damaged the new protection of the bed, and finaly moved the Z axis. :(

scotty1024, 2015-06-24:
Suggestion: raise head 10mm *then* adjust Z end stop and then auto home.

Your Z end stop could otherwise still be triggered after adjustment.

AnHardt, 2015-06-24:
@thinkyhead 
As far i remember we stumbled across this problem (save-homing only with ABL) during a debug-session (hangout) on boelle's printer.

Here unconditional rising is no solution for a problem because of new problems when close to the top boundary.
How about first clearing all the endstops?

plgarcia, 2015-06-25:
I am installing a sensor to implement auto leveling, and during one test the probe hang the screw of the bed. So I moved the end stop to avoid that but by the same I reduce the suface on witch I  can print.
Clearing the endstops  is the minimum to do but starting with Z. X and Y should not move as long as we are not certain the nozzle does not touch the bed.
But that is not enough. On auto levelling there is a Z_RAISE_BEFORE_HOMING.
I think this parameter should be global.
Auto evelling should start by  : Ensuring tha that level is higher than this value, and if not raise to this axis. At the  hight there should be no constraint to move the nozzle.

After home X and Y.  And finally homing Z.
An improvement to that would be to ensure X is greater  than X_MIN_POS while homing Y, and vice vesa.
So the nozzle would never go in the corner.

Roxy-3DPrintBoard, 2015-06-25:
> But that is not enough. On auto levelling there is a Z_RAISE_BEFORE_HOMING.

It may be time to consider the alternate approach where  we have a set of #define's that say what the actual height is supposed to be at different points.   Something like:

#define Z_HEIGHT_BEFORE_PROBING ????
#define Z_HEIGHT_AFTER_EACH_PROBE ????
#define Z_HEIGHT_WHEN_DONE_PROBING ????

This approach would get rid of the errors that are causing air printing and driving the nozzle into the bed.

AnHardt, 2015-06-25:
>I am installing a sensor to implement auto leveling, and during one test the probe hang the screw of the bed. So I moved the end stop to avoid that but by the same I reduce the suface on witch I can print.

Your slicer knows where not to print?

>Clearing the endstops is the minimum to do but starting with Z. X and Y should not move as long as we are not certain the nozzle does not touch the bed.
But that is not enough. On auto levelling there is a Z_RAISE_BEFORE_HOMING.
I think this parameter should be global.

As said before, that is not a good idea. Before homing we do not know where we are. We are in danger to crash into the upper boundary. Only when touching the z-end-stop we know we are to deep.

>define a X_MIN_POS_FOR_HOMING and Y_MIN_POS_FOR_HOMING.
ensure X is greater than X_MIN_POS_FOR_HOMING while homing Y, and vice vesa. So the nozzle would never go in an area around the corner where there may be conflics with others mecanical parts.

Please try to remember what homing is for - to find out where the nozzle is. As long we do not know where the nozzle is, we can not automatically avoid forbidden areas. We need a second positioning system - __You__.

When homing manually, just raise your nozzle enough to not hit your bed-mounting. (Don't start your car and kick the foot throttle without looking left an right). When homing within a g-code program and you are sure you will never be close to the top boundary, just raise the nozzle a bit (but you can never be sure, without a look at your printer).

AnHardt, 2015-06-25:
Or use a z-max-endstop.

brainscan, 2015-06-25:
I thought it was just me thinking this.

plgarcia, 2015-06-25:
If you want, I take this and propose a update of the software I will be able to test with normal Z endstop and with capacitice sensor and uto leveling with mu Pusa I3.
And I would write a bit of documentation to document this feature.
How can we synchronize ? Just cloning ?

plgarcia, 2015-06-25:
@AnHardt 
I understand all this and the aim to auto home.
I use auto home for 2 reasons. When switching the machine on, and when I want to set the position of the endstop. When we start the machine we do not know where we are but we know after. We can know if an autohome has already been done and then if the position is quite relyable of not.
And therefore we can use 2 différnet strategies depending on the case we are. 

Anyway this strategy must ensure as much as possible not to brake either the bed, one of the nuzzle, or a sensor.

It is easy when we  have only 1 nozzle, and no sensor, jest set the X and Y  endstop at the right position. It is more complicated when many nozzles and a sensor, because in that case either you reduce  the surface you are able to print on on you accept nozzles and sensors  have to go outside the bed. And then there  is a risk to damage the printer  if you go up or down. But we are  more often closer to the bed than to the upper limit.

A way to avoid to reach the upper limit woud be to  reduce automaticcaly the upper limit for the only few milimeters we may raise  the Z axis when homing. This does not cover the case where the user by whatever mean  has raised theover the limits or the printer hasb een stopped while homing as it was  already over the limits. I think this is not a real risk.

The simpler would be not to have these screws over the bed  to have no specific constraints. I will change that but that is not as simple as ta say it. and if somebody has good ideas many poeple would appreciate.

We know all the constraints

thinkyhead, 2015-06-26:
On cartesian printers with a real Z-min endstop the solution is actually pretty simple. Make Z home first, then raise the Z axis, and then home XY.

jbrazio, 2016-03-07:
@thinkyhead, @plgarcia  Didn't #3001 and #3102 kinda fixed this feature request by first raising Z by `Z_RAISE_BEFORE_HOMING`, then homing XY and only then probe Z home ?

The proposed idea of reducing `Z_MAX_POS` temporarily in order not to ever reach the limit is impossible to implement other than with a physical z-max endstop as the start Z position is unknown, we could be already on the limit when the G28 is issued.

Blue-Marlin, 2016-03-07:
@jbrazio 
Yes they did. The fault in 3001 was only rising too often.
My #3107 additionally does not rise z when we know we are high enough (moved z up before) and works even if AUTO_BED_LEVELING_FEATURE is disabled.

jbrazio, 2016-03-10:
@thinkyhead I suggest to close this FR.

closed 2016-03-10 by thinkyhead

plgarcia, 2016-03-10:
I changed the software, and I do not encounter problems. With the changes I made there is no risk the printer raise too much. I managed to use the full size of the bed even if many heads, even with the probe. All heads can print on the full surface.
I hope I find time this week-end to make a video to show what I did.
Regards
Pascal Garcia

jbrazio, 2016-03-11:
We are looking forward for it.
Cheers.