MarlinFirmware/Marlin #2316: Invalid suffix "p" on integer constant

ratulo, 2015-06-19:
Hello,
long time ago I was iploading marlin to my prusa. Today I want to make som adjustment for bed leveling and I found this problem:
Marlin_main.cpp:2809:14: erroer: invalid suffix "p" on integer constant
Marlin_main.cpp:2821:47: erroer: invalid suffix "p" on integer constant
Can you help me to solve this problem please?
Thanks
All the best

AnHardt, 2015-06-19:
Wold you please be so kind to tell us about your version, or branch of Marlin? Presumably there is something wrong with your Configuration.h. Could we take a look at it?

ratulo, 2015-06-19:
Thanks for your help. Few seconds ago I found problem:
#define Z_PROBE_OFFSET_RANGE_MAX -5p
its work now

closed 2015-06-19 by ratulo