MarlinFirmware/Marlin #2319: Full graphic LCD - Not showing X, Y or Z location during print.

mr-sneezy, 2015-06-20:
Hi All,
Not sure if this is a 'me' bug or a firmware bug. 
Since updating to 1.0.3 Dev a few weeks back I notice that the display no longer shows where the print head currently is. In particular it was good to see how far up the build plate the Z axis was during the print.
I've checked during a few prints and the X, Y & Z mm display bar on the graphics LCD just stay at ---.-- all print long now...

AnHardt, 2015-06-20:
The display is showing ---.-- now when the printer does not know its position.
Especially before homing, or when the motors where shut down.
Since #2216  also DISABLE_AXIS is respected in conjunction with DEFAULT_STEPPER_DEACTIVE_TIME instead of shutting down __all__ steppers.
Should work now (since 2015/06/10).

mr-sneezy, 2015-06-20:
Thanks for the reply, and I must admit that I can't understand the issue #2216 after reading all of it, I will trust that you think it's related to what I see above :)

One thing, to make sure I explained it well enough, the graphic LCD doesn't show real mm position even while printing, not just before homing etc. 
For sure it makes sense that it would show ---.-- before homing has been done.
With 1.0.2 Release I had no issue.
Thanks.

AnHardt, 2015-06-20:
Let me try to expain.
* When the printer is switched on it does not know where it is - displays ---.
* After the printer is homed an axis the printer knows where it is - displays coordinates for that axis.
* When an axis is not active its motor is switched off - depending on DISABLE_AXIS. When a motor is switched off it will skip to the next full step - so its position is unknown again - consequently the display shows --- again.
* When DEFAULT_STEPPER_DEACTIVE_TIME is over (starting with the last move), before #2216 all steppers where switched off, regardless of the DISABLE_AXIS settings. This can happen, for example, during heating up the bed (M190).
The --- was implemented after the release of the `Release`.

Nandox7, 2015-06-20:
I got recently a Full Graphics display and I've noticed the same.
I understand that the printer needs to perform the homing before it can "locate" itself.

But for example this photo was taken during a printing job and it still doesn't display any coordinates.
https://goo.gl/photos/QPqwy6jtsMMi6raA6

I can't recall but I think even before with 1.0.3 dev I wasn't getting the coordinates even with the UltraLCD.

mr-sneezy, 2015-06-20:
Yep, that photo is what I get now too during a print, and that's not what it used to be before my firmware update.

AnHart thanks for dumbing it down for me.
I did fully understand that the display would read ---.-- when the steppers were either off or at a unknown location, that is quite obvious, but I'm ONLY talking about during an actual running print job here, not during the heating or before homing...

What I'm confused with now is, is this a real code change for a reason and will stay this way now, or is it a bug and will go back to the way it used to be in 1.0.2 ?

AnHardt, 2015-06-20:
@Nandox7 
--- was introduced with
https://github.com/MarlinFirmware/Marlin/commit/1a4a09c26f6ed206970668911e816c5539d37954 
on Mai 14, 2015
1.0.2 was released Mar 04, 2015.
`Development` was the default download (branch) until around Mai 20, 2014.

AnHardt, 2015-06-20:
@mr-sneezy 
So check your settings for DISABLE_AXIS.
The change was intentionally.

ntoff, 2015-06-20:
>When an axis is not active its motor is switched off - depending on DISABLE_AXIS. When a motor is switched off it will skip to the next full step - so its position is unknown again - consequently the display shows --- again.

stupidest

idea

ever.

eboston, 2015-06-20:
>stupidest
>
>idea
>
>ever.

Which part are you saying is the stupid idea?  If it is about skipping to the next full step when switched off, that is because of physics, not because of design.  It takes current to hold the motor between full steps.

mr-sneezy, 2015-06-21:
Are you really saying that you turn off the Z axis (lets just discuss only Z here at the moment) during printing between print layers, and that the head position is therefore unknown in Z ?

(Also, are we talking about microstepping here verses full steps, because I was not aware my Melzi board did any microstepping, maybe my ignorance showing).

Nandox7, 2015-06-21:
The current structure is a bit messy, what should be the branch to checkout in order to test this?
I'm running the code from tag 1.1 at the moment.

Nandox7, 2015-06-22:
Btw I noticed that it got even more weird halfway during a print it started showing jus the values for the Y axis.

Wackerbarth, 2015-06-22:
The tag 1.1 is a “beginning of development” tag. — You want to use the “dev" or “Development” branch to get the latest development code. (They are the same except that the dev version has the commits rebased to provide a cleaner history for the next release.)

mr-sneezy, 2015-06-22:
If it were to come to a vote, I'd prefer to have X,Y & Z position shown during prints and have that as the default behaviour like before.
 I spent much time making sure I set the motor current limiting properly and have the right amount of heatsinking to handle the motor holding on all the time. Fixing hardware issues in software could be called lazy...
My 2 cents, and thanks for efforts with the firmware, on the whole it's a good bit of code.

AnHardt, 2015-06-22:
@mr-sneezy 
So don't change the default value for:

>#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false

in Configuration.h
and use the ' Development' branch after June 11, 2015,
or set DEFAULT_STEPPER_DEACTIVE_TIME to 0 in Configuration_adv.h

Nandox7, 2015-06-22:
@Wackerbarth thank you!
I'll merge my local repo with it and give it a try.

thinkyhead, 2015-06-23:
> Fixing hardware issues in software could be called lazy

We try to assiduously avoid that. The display of "---" for unknown coordinate is meant to inform the user that the printer doesn't know where it is, for whatever reason. It wasn't _precisely_ designed to happen when a motor turns off during a print, but I support it nonetheless, because users may not realize that turning off the motors during printing is a cause of printing problems (which is why `axis_known_position[AXIS]` gets cleared whenever an axis motor is disabled.) Meanwhile, most feedback we get is to get rid of the useless display of X and Y during printing.

mr-sneezy, 2015-06-23:
Yes agree that X & Y are not useful during print for me either. Z definately was though, as I could keep an eye out for a particular layer about to start, then watch it for issues. I did this often while dialing in the Prusa i3. Perhaps the ---.-- could show 'MotOff' when the Z motors are disabled (the Prusa i3 has two).
Also can layer height be indicated from the G-code line currently being printed instead ?

Nandox7, 2015-06-23:
Agreed, I don't care much for the X & Y but I definitely use the Z a lot for different purposes like checking printing progress.

In the mean time I loaded the latest 'Development' code and it seems to be working better, I can see the values showing up after the G28 is done.

AnHardt, 2015-06-23:
@mr-sneezy 

>Also can layer height be indicated from the G-code line currently being printed instead ?

No. We could calculate where z is supposed to be - not where we are.

mr-sneezy, 2015-06-23:
If we can't have actual Z position then a calculation from the G-code would be good enough for me.

mr-sneezy, 2015-06-23:
I've downloaded the current Dev branch tonight and installed it on the Prusa i3.
Yes X, Y and Z are updating values during a print. 
Like spoken about above, X & Y don't have much use, but Z does for me.

About the actual display, I note that the xxx.xx values now show xxx with no decimal when the value is a whole number. i.e. with no trailing .00
Not sure if that was the coders intention....

thinkyhead, 2015-06-24:
> I note that the xxx.xx values now show xxx with no decimal when the value is a whole number

Yes, the change that was made to improve the readability of the display.

> Downloaded the current Dev branch . Yes X, Y and Z are updating values during a print.

Excellent! Closing the issue.

closed 2015-06-24 by thinkyhead