MarlinFirmware/Marlin #2320: Z Probing with inductive Sensor?

stevie72, 2015-06-21:
Hello, 

is there a way to run a inductive autolevel sensor on mendel90?
there is a holder for that at https://www.youmagine.com//designs/mendel90-z-probe-bracket
but nowhere is described how to setup the firmware for this sensor http://www.ebay.de/itm/390985599535?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

Greethings, Stefan

brainscan, 2015-06-21:
It's actually easier than with a servo. There are some YouTube videos (Thomas Sanladerer's channel) can explain better than I can.

fmalpartida, 2015-06-21:
If you are looking for a tutorial you can also tap on my channel. But it's in Spanish.

Tommy-LSA, 2015-06-22:
Maybe you can learn from comments here:
http://www.thingiverse.com/thing:447719/#comments

stevie72, 2015-06-22:
If someone who has a mendel90 with autobed leveling and a inductive Sensor would share the configuration.h file that would help understanding 

My email: stevie72@gmx.de
Greethings, Stefan

brainscan, 2015-06-22:
Hopefully someone will post this for you but you might have to wait a while. Setup should be similar for most printers so might be better to ask specific questions about what you need to know. I've always managed to get mine going and I've only used info made for other printers and adjusted accordingly.

thinkyhead, 2015-06-23:
> If someone who has a mendel90 with autobed leveling and a inductive Sensor would share the configuration.h file that would help understanding

Definitely request at the RepRap forums in the Mendel topic.

Tommy-LSA, 2015-06-23:
This Thing is made for K8200 but the Config should be nearly the same. Check the comments Thread of this thing: http://www.thingiverse.com/thing:447719

stevie72, 2015-06-23:
@ brainscan 

You are right. there is a video from Thomas Salander.... but this won´t work with my mendel90.

The only command that is working is the G28 Command. G29 will not work .. the Printbed is running out of range backwards an a reset only stops it..

Anyone an Idea where to look for an solution for that?

@ fmalpartida

I don´t find the documents - do you have a link?

Greethings Stefan

fmalpartida, 2015-06-23:
Try this lot:
wiring - http://blog.electrofunltd.com/2014/11/esta-semana-repasamos-como-conectar-un.html
setup - http://blog.electrofunltd.com/2014/11/configuracion-del-sensor-inductivo.html

In spanish, though.

thinkyhead, 2015-06-24:
@fmalpartida There's so much superior documentation in Spanish, especially in the RepRap wiki's "Clone Wars" section. I had started to translate a bunch of Clone Wars pages into English using automated translation software, but I don't speak Spanish so it was very slow going. Still, I would love to see all that stuff translated into English.

fmalpartida, 2015-06-24:
@thinkyhead if you post a link to what you currently have I'll try to find some time to move this on a bit too.

stevie72, 2015-06-24:
I´ve solved the problem wiith the help of a real nice friend!
Thanx svanteg !
Your configuration was the final goal !

Thanx so much !

Stefan

closed 2015-06-24 by stevie72