MarlinFirmware/Marlin #2321: Z probe does not compile with RedRapDiscount lcd

MysticDarkLord, 2015-06-22:
I cannot get the file to compile properly when enabling the Z Probe routine in Configuration.h when I have my LCD enabled (RepRap Full Graphics display). 
Internal out of range error messages come up from UltraLcd file and U8glib.
Can anyone help with this issue or suggest a solution to try.

Nandox7, 2015-06-22:
I managed to have it compile using the Arduino version 1.6.x.

MysticDarkLord, 2015-06-22:
I have been using Arduino 0023 as this was the only arduino i could get to compile the Marlin files when I first started with my printer. Do you know where I can get information giving credible instructions in changing the things I need to update to 1.6 particularly in reference to Arduino.h as I cant get anything to compile with 1.6.2

thinkyhead, 2015-06-23:
#1741 #1156

MysticDarkLord, 2015-06-23:
Nandox7 - Cheers great advice, in the end this is what I needed to do and when I deleted files out of the library and caused the IDE to look else where for them I had success.

closed 2015-06-23 by MysticDarkLord