MarlinFirmware/Marlin #2323: Thermal runaway problem

Drakelive, 2015-06-22:
Hi

When I try to set a temperature of the extruder (after 10 seconds) I get this error message:

Setting hotend temperature to 185.000000 degrees Celsius.
Error:Heating failed, system stopped! Heater_ID: 0
[ERROR] Error:Heating failed, system stopped! Heater_ID: 0
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

I use a controller for thermocouples  MAX6675

Regard
Drk

AnHardt, 2015-06-22:
We had a lot of changes in this area of the code since 1.0.2.
If you allready are on the Development branch please play with the parameters WATCH_TEMP_PERIOD and WATCH_TEMP_INCREASE in Configuration_arv.h. If you can't find reasonable values you can deactivate the test by commenting THERMAL_PROTECTION_HOTENDS in Configuration.h.

Drakelive, 2015-06-22:
Hi AnHardt

I read how THERMAL_PROTECTION_HOTENDS and I think I understand how to working.

I think the problem is that the extruder is heated very slowly. My Ramps 1.4 is powered by a power supply is 30A. This problem may depend on the PID variables? In this case what should I change?

Drk

Drakelive, 2015-06-22:
Hi

I just saw that in Configuration.h There are three preset profiles the PID:

1) Utimaker (Active Default)
2) MakerGear
3) Mendel Parts V9 on 12V

My printer is a custom made by me and as extruder has a E3D, which has a metallic cartridge Heater.
The third profile seems the most suitable one .....

In general which of the 3 parameters of the PID change improves the speed of temperature rise?

Drake

grob6000, 2015-06-23:
From a cold start, only the heater power matters, as the heater will be 'full on' until it reaches the PID control region. Your E3D has enough power to show an obvious rise in temperature when working.

Follow @AnHardt 's advice first by switching off thermal protection, and checking the heater otherwise works - I have a feeling the heater is not working, and the thermal error is valid (it's supposed to detect a switched-on device not responding).

* Check the heater resistance
* Check the heater works when DC power is supplied directly
* Check with an external temperature probe (e.g. a multimeter + thermocouple) that the displayed temp is correct

AnHardt, 2015-06-23:
@Drakelive 
If you want to find good PID values for __your__ heater, take a look at M303.
Until the temperature is close to the target the heater if full-on. PID is only working close to the target temperature. 
30A are ok to heat a E3D pretty fast. Possible errors: Fuses to hot? 24V heater cartridge on 12V system? Bad thermal contact of the thermometer? ...

Drakelive, 2015-06-25:
Ok, I solved the problem, Cartridge Heater was getting broken.
After replacement, the temperature rises quickly.

Thank you.

closed 2015-06-25 by Drakelive