MarlinFirmware/Marlin #2326: Z probe ignore y axis

SquirrelCZE, 2015-06-23:
Hi folks, trying to setup printer with actual git version

-> setuped three-points z probing and it kind of ignores Y coordinates... while Y axis works fine when I move it manually, on Z-Probe, always probes on Y=0 -> any tips?

http://git.veverak.org/printer.git/blob/master/Marlin/Configuration.h
http://git.veverak.org/printer.git/blob/master/Marlin/Configuration_adv.h

atfield90, 2015-06-23:
try disabling max_endstop, worked for me

thinkyhead, 2015-06-24:
@SquirrelCZE It sounds like @atfield90 has the right idea. Make sure you have the following line un-commented if you only have endstops at the zero ends of your axes, and none at the max-coordinates ends:

    #define DISABLE_MAX_ENDSTOPS

Hope that's all you need!

SquirrelCZE, 2015-06-24:
Nope, didn't worked

thinkyhead, 2015-06-28:
@SquirrelCZE What does `M119` tell you about your endstops as you press and un-press them?

closed 2015-11-03 by AnHardt