MarlinFirmware/Marlin #2328: Feedrate Very Slow between Z Probs for Auto Bed Levelling

nickpda, 2015-06-24:
Configuration.h 
#define AUTO_BED_LEVELING_GRID
#define XY_TRAVEL_SPEED 8000
G28 Movement OK as usual
G29 Movement very slow for the 4 points LEFT,RIGHT,FRONT & BACK_PROBE_BED_POSITION
After G29 Movement resume to normal feedrate.
Anybody encounter this?
Thanks

CarpetHead, 2015-06-26:
I have also encountered this, I only get it when I slice with simplify3d. With slicer it is fine. If I use the start of my start gcode from slicer in place of the gcode from simplify3d, it fixes the problem. Here is the relevant gcode:

Works Perfectly:

M107
M190 S50 ; set bed temperature
M104 S190 ; set temperature
G28 ; home all axes
G29 ; auto bed level

M109 S190 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0

VERY slow:

G90
M82
M140 S50
M190 S50
M104 S180 T0
M109 S180 T0
G28
G29
G92 E0

Hope this helps

thinkyhead, 2015-06-28:
@CarpetHead Are you able to determine the minimum difference in GCode that leads to the issue? For example, do these two pared-down snippets produce different results…?
```
M107 ; fan off immediately
M190 S50 ; set bed temperature and wait (compare to M140)
M104 S190 ; set hot-end temperature, but don't wait
G28 ; home all axes
G29 ; auto bed level
```
```
G90 ; absolute coordinates
M82 ; E absolute
M190 S50 ; set bed temperature again, wait
M109 S180 T0 ; set hotend 0 temperature again, wait
G28 ; home all axes
G29 ; auto bed level
```

CarpetHead, 2015-06-30:
I don't have much more time to test but it seemed that removing both the M140 and M104 commands fixed the issue in the slow one

thinkyhead, 2015-07-04:
> removing both the M140 and M104 commands fixed the issue

Well that's a helpful lead, anyway. We'll sort it out soon, I'm sure!

boelle, 2015-08-03:
moved to: https://github.com/MarlinFirmware/MarlinDev/issues/31

closed 2015-08-03 by boelle