MarlinFirmware/Marlin #2329: power on behaviour

plgarcia, 2015-06-24:
I am adding the power on feature to my printer.
I used the menu power off, and this went well, it swiched off the 24v power supply. The power supply of the arduino is still on.
Then I try the preheat pla menu, and the power is not swiched on and the arduino tries heating hotend and bed.

I beleive that when a action that needs power (printing, heating, ...) either that command shoul not be taken into account, or the power should be switched on (if the function is activated of course.

I beleive also that power off should not be immediatly taken into account if EXTRUDER_AUTO_FAN is activated, but deffered as long as the temp needs the fan on. It shoud of course stop heating bed and nozzles, if we want the have a chance the hotend to cool down. The power off should be cancelled if an action occurs before the hotend is cold.

thinkyhead, 2015-06-26:
If the printer can tell when the power is on or off, this would be a good thing. One could argue that if the power is off and you try to do any stepper movements, the power should be turned on so the motors are powered. In that case all movement commands to stepper motors would have to test the power state.

But then there is the question: How long does it take for the power to reach the electronics and for the chips on the board to be ready to process signals? Likewise, when you turn off the power how long does it take for the capacitors to discharge?

So clearly, when the power is off, then you not only have to enable power but you must wait for a proper amount of time before trying to do anything with some parts of the electronics.

Automatically powering up probably works ok for heaters and fans, we just have to be careful about doing it for other purposes. As the stepper movement example shows, the check for the power state adds overhead that might be detrimental to performance in some places.

It seems like a good idea to have power off wait for the extruder-auto-fan, but there ought to be some visual indication that it's in a powering-down state, such as blinking the power LED or the LCD screen.

ntoff, 2015-06-26:
>In that case all movement commands to stepper motors would have to test the power state.

I would just use the menu to test power availability. 

power supply type and whether it's powered on or off at boot tells the printer the state of the PSU when first powered on, and obviously it would know the power is off if you turn it off via gcode / the menu.

As soon as you press the encoder it triggers a watch timer, if you press the encoder again within that period to enter a sub menu it turns the power on. This would probably weed out any accidental presses as you bump the knob.

plgarcia, 2015-06-26:
Aggreed with the contraints given by thinkyhead.
@ntoff 
Before defining how,  we should define what.

The fact is that the actual situation where action that nedds power are commanded without the power is is not correct.

I change a bit my mind :
As there are M80 and M81 commands to manage power is gcode noting should be automatic.

1 - I think we should have the indication of power on , power off and shutting down available on main screen (A character for example arrow up, down or horizontal or something else no matter). This is for all printer already existing. An optional feature can be added to have an extra led and button to manage power.
2 - The menus should be reorganized to isolate in submenus the functions that should not be available without power (most prepare options), and these actions should be hidden when power is off or shutting down.
3 - either abort printing or wait for manual power on with a message on the screen, when an action is done in printing that needs power (moves all gcode,  heating ...) and power is not on. Abort is probably the simpler and report the error.
4 - Check M80 code waits enough to ensure the power is ready for use to continue.
5 - desable menu power off is an action needing power is ongoing. (auto levelling, printing, under control by usb by rapetier for example, ...)

We have a connex subjet that is, when power is swiched off or when the steppers are disabled, we may have lost the 0 position and ther should be a full homing done.

Is that functionnal description a good starting point for that.

ntoff, 2015-06-26:
>As there are M80 and M81 commands to manage power is gcode noting should be automatic. Even more as when power of we may have lost the 0 how can it be automtic.

My startup gcode turns the printer on, heats the hot end and bed and then homes the printer. This is the gcode that's added to every sliced object. I can literally hit "print" and walk away knowing the automated process will just work.

Why complicate everything? If the printer is off you'll know it, the fans won't be on. Move the power on/off menu item to the main menu (which I've done for my own copy of Marlin) and it's right there staring you in the face when you open the menu, it says "Turn printer on".

thinkyhead, 2015-06-27:
> turns the printer on, heats the hot end and bed and then homes the printer

Homing is a continuous process of movement until the endstop is hit, so it it pretty well guaranteed to work, even immediately after a power-up. However, the following code would be problematic:
```
M81    ; Turn off high voltage
G4 S4  ; Wait 4 seconds for power to drain
G1 X50 ; A move that needs to turn on the power first
```
The movement cannot be guaranteed to work without an initial delay, because some of the first stepper signals may be sent to the board before it has been fully powered. It may work fine on "warm" electronics. But starting up "cold" with all the capacitors discharged may add a small (~10ms) delay.

> Move the power on/off menu item to the main menu (which I've done for my own copy of Marlin) and it's right there staring you in the face

That's a good idea for a separate feature request.

grob6000, 2015-06-28:
The thoughtful usage @plgarcia describes is good - it would be a smarter, more convenient interface for him, but others might want slightly different operation.

I can see about 4 different feature requests that would help him achieve his goal, but at the same time allow others (as seen above) to take different approaches:

(1) Adding a configurable blocking delay to M80 (to wait for power on - this would save all the "G4"s in the scripts above, and I see no reason to do anything after M80 before the power is ready).
(2) Customising function of LCD commands - e.g. by having LCD commands point to gcode scripts on the SD card - could allow everyone to specify exactly the behaviour they want (re implementation - if script/SD not found/emabled just do the default). The LCD/SD is a host in itself, so giving it a bit of intelligence (and behaviour that's not identical to other firmwares) is no big deal. The easier to use and/or more flexible the interface, the more people will like it and use it I think.
(3) Adding a generic "wait for pin state" command, (in the style of Marlin M42 for controlling them), so our scripts could do things like checking that auto-fan was off (just by pointing to the auto-fan output pin), or the PSU wait could monitor an ATX supply's PWR_OK signal. Note that something like smoothieware has the ability to do this with their generic "switch" feature.
(4) Have M80 (configurably) block until a pin state (e.g. for ATX PWR_OK). While cool, and ATX supplies are common and all have this signal, whether this is any more useful than a delay (and thus worth the extra I/O channel) is of course debatable.

Maybe separate and discuss?

closed 2016-05-04 by jbrazio