MarlinFirmware/Marlin #2334: SAV MK1... almost there... and then not

boelle, 2015-06-26:
So I got an SAV MK1 fmalpartida... 

and we have gotten it really close to working... without auto bedlevel everything works

with auto bed level enabled i can also check that the z probe works and the servo works... but then i cant turn on the extruder fan from pronterface or by sending m106 and m107

i use a copy of marlin that is not more than 2 weeks old as it was the best working copy... https://github.com/fmalpartida/Marlin/tree/SAV-MkI-config/Marlin

setup info is here: http://reprap.org/wiki/SAV_MKI

scroll down about half way to you see: Setting-up the environment

pins file: http://pastebin.com/ByVb29Lv

configuration.h: http://pastebin.com/QpQHGXHG

configuration_adv.h: http://pastebin.com/tEkBivhF

anyone that can give a clue on what is goin on?

boelle, 2015-06-28:
another issue i have:

as i have a mendel90 i only have a Z max switch, there is no Z min. So when i home Z it moves up and when it touches the switch it knows how far down it is.

with autolevel enabled and the probe touching the bed i would still expect G28 Z to make it move up and touch the switch. but it does not... it moves down

i tried DISABLE_Z_PROBE_ENDSTOP but then it cant compile as it complains about the missing probe

SanityCheck.h:120: error: #error You must have a Z_MIN or Z_PROBE endstop to enable Z_PROBE_REPEATABILITY_TEST.
SanityCheck.h:135: error: #error You must set Z_PROBE_PIN to a valid pin if you enable Z_PROBE_ENDSTOP.

boelle, 2015-06-28:
How M48 behaves with above config

http://youtu.be/YhOfq3u8Afc

youtube uploading and procession the file

boelle, 2015-06-28:
@thinkyhead first video up.... not sure if it makes any good to do a G29, but will try it later today

boelle, 2015-06-28:
and this is how G29 behaves: https://www.youtube.com/watch?v=8DvzcqGlsL4

boelle, 2015-06-28:
Just upgraded to development and it behaves the same... will try and fiddle with the pins file but i dont expect it to be any difference

boelle, 2015-06-28:
also i just remembered that the servo needs 2 angle 0 commands to get to angle 0

boelle, 2015-06-28:
when i try to make my pins file look like this for the section about the endstops:

//#define X_STOP_PIN         13
#define X_MIN_PIN          13
//#define Y_STOP_PIN         14
#define Y_MIN_PIN          14
//#define Z_STOP_PIN         15
#define Z_MAX_PIN          15

#define Z_PROBE_PIN        36

i get this error: 

Marlin_main.cpp:277: error: 'X_MAX_PIN' was not declared in this scope
Marlin_main.cpp:277: error: 'Y_MAX_PIN' was not declared in this scope
Marlin_main.cpp:277: error: 'Z_MIN_PIN' was not declared in this scope
Marlin_main.cpp: In function 'void homeaxis(AxisEnum)':
Marlin_main.cpp:1634: error: 'X_MAX_PIN' was not declared in this scope
Marlin_main.cpp:1634: error: 'Y_MAX_PIN' was not declared in this scope
Marlin_main.cpp:1634: error: 'Z_MIN_PIN' was not declared in this scope

boelle, 2015-06-29:
@MarlinFirmware/collaborators @MarlinFirmware/testers 

anyone that could pitch in on what is wrong here?

MagoKimbra, 2015-06-29:
Because sensitive_pins use axes_MIN_PIN and axes_MAX_PIN..
If you use axes_STOP_PIN the routine in pins.h define axis_MIN and MAX:
#ifdef X_STOP_PIN
  #if X_HOME_DIR < 0
    #define X_MIN_PIN X_STOP_PIN
    #define X_MAX_PIN -1
  #else
    #define X_MIN_PIN -1
    #define X_MAX_PIN X_STOP_PIN
  #endif
#endif

But you define only X_MIN_PIN you must define X_MAX_PIN -1

Sorry for my bad english.... :(

Wackerbarth, 2015-06-29:
The problem is that X_MAX_PIN is defined as a SENSITIVE_PINS even though it doesn't exist in your configuration.

boelle, 2015-06-29:
oki.. thanks for the pointers

what about the homing stuff?

MagoKimbra, 2015-06-29:
If you put command M119 Z_PROBE is triggered or not??

boelle, 2015-06-29:
all switches including the probe worked right last time i checked... but will flash config and try again..

but i think the "issue" might be that my printer is set to move up when i home... because my Z switch is top mounted

time will tell

MagoKimbra, 2015-06-29:
A ok.. Yes only in DELTA home is MAX and probe is MIN...
In cartesian and CORE if you have Z_HOME_DIR 1 the direction in G28 Z0 is to max...

boelle, 2015-06-29:
yep and that is wrong in my case at least, the mendel90 printer has the switch at the top

but i corrected the pins and flashed... everything works as expected.. i can home all 3 axes... turn heaters on and off... turn fan on and off

then i enable autolevel and i click home Z in pronterface...

now for each time i click home Z in pronterface it moves down

X and Y switches report as triggered since they are homed

Z max reports open as it has now moved down a few times

Since my Probe is retracted it reports triggered as it should due to the design

http://forums.reprap.org/file.php?292,file=32162,filename=block.jpg
http://forums.reprap.org/file.php?292,file=32164,filename=Extruder2.jpg 

when the probe is not in use it pulls up the red part  in the first pic, one downside is that this will report the probe as triggered

when using the proble the servo just lowers the red part and the probe is no longer triggered, only  when the Z goes down far enough to push up the red part

this is the first issue i have

will test how M48 and G29 behaves

boelle, 2015-06-29:
M48 reports:

SENDING:M48
M48 Z-Probe Repeatability test
Mean: -12.000000
Standard Deviation: 0.000000

does that sound right?

boelle, 2015-06-29:
running it a 2nd time... will do a video at 3rd test

boelle, 2015-06-29:
https://youtu.be/uH2XG-paK7w

does it look right?

boelle, 2015-06-29:
i just find my numbers to good to be true

is there a way i can make it write "touch" on serial output so i know its has touched

boelle, 2015-06-29:
G29 behaves like this:

https://youtu.be/saqQ_p-Ytso

so what i see it tries to probe at the 3 points most far away from the cam

what i belive i have defined in config is 1 point close to the cam and 2 most far away...

soon from above it should have been like this:

*-----------------
|                     |
|                     |
|                     *
|                     |
|                     |
*-----------------

MagoKimbra, 2015-06-29:
Because the Axis X is more hight, test this G28 then G29...

boelle, 2015-06-29:
i do home X and Y first... marlin tells me to do that :-D

and i have 0,0 in the center

boelle, 2015-06-29:
my probe points:

      #define ABL_PROBE_PT_1_X 84
      #define ABL_PROBE_PT_1_Y 0
      #define ABL_PROBE_PT_2_X 86
      #define ABL_PROBE_PT_2_Y 90
      #define ABL_PROBE_PT_3_X 86
      #define ABL_PROBE_PT_3_Y -90

boelle, 2015-06-29:
uppps

i forgot to put - in front of 84

boelle, 2015-06-29:
so its now probing the right places... but it does not touch the bed

boelle, 2015-06-29:
i think i have figured it....

it must be the way my probe works compared to what marlin expects

MagoKimbra, 2015-06-29:
No, the problem is that if X is a height too high G29 behaves so. You must first do the zero for all three axes so only G28, no G28 X0 Y0 and NO G28 X0 Y0 Z0, only G28 then G29...

Roxy-3DPrintBoard, 2015-06-29:
> No, the problem is that if X is a height too high G29 behaves so. You must first do the zero for all three axes so only G28, no G28 X0 Y0 and NO G28 X0 Y0 Z0, only G28 then G29...

M48 (potentially) moves all 3 axis.  A G28 should be done first so Marlin feels comfortable moving the nozzle around.

It would be interesting to see something like:  M48 L 5    where it will move the X & Y also.

> M48 reports:
>SENDING:M48
M48 Z-Probe Repeatability test
Mean: -12.000000
Standard Deviation: 0.000000

>does that sound right?

No.   It would be great if that was true.  But most likely you will end up with a Standard Deviation of something like .001 give or take.

boelle, 2015-06-29:
yep.. i'm just measuring my switch... think its faulty

boelle, 2015-06-29:
the switch was faulty...

now with a working one when i issue M48 P10 it only samples 1 time...

boelle, 2015-06-29:
SENDING:M48 P10
M48 Z-Probe Repeatability test
Mean: -0.648828
Standard Deviation: 0.000000
echo:endstops hit:  Z:-0.65 Z_PROBE:-0.65

boelle, 2015-06-29:
>>>M48 P4 X0 Y0 V2  L2
SENDING:M48 P4 X0 Y0 V2  L2
M48 Z-Probe Repeatability test
1 of 4   z: -0.446875
2 of 4   z: -0.446875
3 of 4   z: -0.307031
4 of 4   z: -0.419531
Mean: -0.405078
Standard Deviation: 0.057698
echo:endstops hit:  Z:-0.42 Z_PROBE:-0.42

but this one moved outside the bed

Roxy-3DPrintBoard, 2015-06-29:
Yes...   You need to be a little ways away from the edge of the bed if you specify some number of legs with the L parameter.   What L does is move straight away from the point where the measurement is supposed to be taken, and then sort of orbits that point.   On the very last leg of travel it heads back to where the point is supposed to be taken.

So...   M48 L 7 X 100 Y 100 will do something reasonable and let you see how it works.

boelle, 2015-06-29:
getting better now... 

are there ways to spec more raise between probes?

boelle, 2015-06-29:
#define Z_RAISE_BETWEEN_PROBINGS  

right`?

boelle, 2015-06-29:
nah that was not it

So it does not raise Z petween probes when doing M48

hmm

Roxy-3DPrintBoard, 2015-06-29:
There is no way to specify more of a raise between points with a M48 parameter.  It uses what ever the firmware is configured at.   So Yes, you need to configure all the #define Z_RAISE_xxx_xxxx items.

boelle, 2015-06-29:
#define Z_RAISE_BEFORE_PROBING 15  
#define Z_RAISE_BETWEEN_PROBINGS 15  
#define Z_RAISE_AFTER_PROBING 15

but it does not raise

Roxy-3DPrintBoard, 2015-06-29:
Hmmmm....   M48 uses run_z_probe()  which is the same routine that G29 uses.   I will have to look at the current code to see where (and what) the raises are.

boelle, 2015-06-29:
maybe and i'm only guessing wild here... but when my probe is retracted.. ie the servo lifts the rod that touches the bed it also makes the switch go triggered....

Roxy-3DPrintBoard, 2015-06-29:
You could try a M401 (engage Z Probe) and then try a M48 to see if the behavior changes for the first point.

boelle, 2015-06-29:
checked with G29..

it skips the first point
then it lowers and touches the next point

but it does not raise before it goes to the next point

boelle, 2015-06-29:
just for the fun of it

in pronterface it tried to move Z up 10 by the buttons

it made X move 10 mm away from the switch also

boelle, 2015-06-29:
homing of X and Y made that go away... just odd

Roxy-3DPrintBoard, 2015-06-29:
That doesn't sound right.   It should just refuse to move anything that is not homed.

boelle, 2015-06-29:
M48 seems to perform better with M401 first...

will test a few times

boelle, 2015-06-29:
SENDING:M48 P4 X0 Y0 V2 L1
M48 Z-Probe Repeatability test
1 of 4   z: 96.064842
2 of 4   z: 96.114845
3 of 4   z: 96.114845
4 of 4   z: 96.367965
Mean: 96.165618
Standard Deviation: 0.118592
echo:endstops hit:  Z:96.37 Z_PROBE:96.37

Roxy-3DPrintBoard, 2015-06-29:
Well....  The reason to try it with a M401 is because that eliminates the engage_probe() from the equation.   But that really should not be necessary.   Those numbers are way too unrepeatable.   Something is causing bad triggers.

boelle, 2015-06-29:
yep agreed.... but it is also moving over the edge of the glass

will try and run the test a few more times

And rememember i use X0 and Y0 as that is my center of bed

Roxy-3DPrintBoard, 2015-06-29:
Oh!  OK on the (0,0) being the center.

boelle, 2015-06-29:
SENDING:M48 P4 X0 Y0 V2 L1
M48 Z-Probe Repeatability test
1 of 4   z: 96.235153
2 of 4   z: 96.235153
3 of 4   z: 96.235153
4 of 4   z: 96.520309
Mean: 96.306442
Standard Deviation: 0.123476
echo:endstops hit:  Z:96.52 Z_PROBE:96.52

boelle, 2015-06-29:
hmmm it does not raise between 3 of the probes... but it did for the last one as the move for it was over the edge...

boelle, 2015-06-29:
http://forums.reprap.org/file.php?292,file=32162,filename=block.jpg
http://forums.reprap.org/file.php?292,file=32164,filename=Extruder2.jpg

pics of how my probe works

boelle, 2015-06-29:
so it will need some raise between points or the switch will stay triggered.... and stowing the servo will not work as that will cause a false trigger

Roxy-3DPrintBoard, 2015-06-29:
Does not raise???   Do you mean the probe does not dis-engage?   Or do you mean the Z Axis does not raise?

boelle, 2015-06-29:
Z does not raise

Roxy-3DPrintBoard, 2015-06-29:
What about in the case of a G29?   Does that raise?

boelle, 2015-06-29:
let me check... i think it does but want to be sure

boelle, 2015-06-29:
nope... did not raise between probes

Roxy-3DPrintBoard, 2015-06-29:
OK...  I think the thing to do is get G29 working...  And the M48 should work also at that point.  M48 pretty much uses the same library of routines that G29 uses.

[607 more replies]

closed 2015-11-24 by AnHardt