MarlinFirmware/Marlin #2335: Marlin Google Hangout • Saturday 27 June 2015 at 21:00pm UTC • Topics TBD • Be there!

boelle, 2015-06-26:
```
Name             Zone               online weekdays      online weekend
@thinkyhead      UTC-7
@Wackerbarth     UTC-5
@Jnesselr        UTC-5
@clefranc        UTC-4
@alexborro       UTC-3
@fmalpartida     UTC+1
@AnHardt         UTC+1
@Wurstnase       UTC+1
@CONSULitAS      UTC+1
@paclema         UTC+1
@boelle          UTC+1              13:00 - 20:00        ~8:00 - 22:00
@emartinez167    UTC+8
@ntoff           UTC+10
```
The above UTC+ values change at summer/winter time by one hour

"Official" Clock :  http://www.timeanddate.com/time/aboututc.html

<a href="https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia" target="_blank">Hangout</a>

link to googledoc with topics:

<a href="https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing" target="_blank">Topics</a>

boelle, 2015-06-26:
No topics yet.... if you want anything added go to very end of googledoc linked above and add comments

you should not need an account for that....

and please keep posts to be about the hangout only... will zapp post that is not about the hangout

thinkyhead, 2015-06-26:
Posting some topic ideas…

AnHardt, 2015-06-27:
* Chances of fixing G29 in a reasonable time.
* Making current `Development` an `RC` regardless of the errors with G29, to avoid blocking innovations.
* Separate branches for releases.
* Discuss the currently open PRs - issues.

fmalpartida, 2015-06-27:
I'm going to skip tonight's handout. A small compromise with some friends.

boelle, 2015-06-27:
np... at least we know :-D

fmalpartida, 2015-06-27:
Can we take a look at updating the integration branch with the development contents.

boelle, 2015-06-27:
it will be on topics list

@thinkyhead will you join us tonight?

Wackerbarth, 2015-07-11:
Suppose they held a meeting and nobody showed up.

@thinkyhead @Jnesselr @clefranc @alexborro
@fmalpartida @AnHardt @Wurstnase @CONSULitAS
@paclema @boelle @emartinez167 @ntoff

Jnesselr, 2015-07-11:
It would seem to be because the original post didn't actually tag us, so I was not aware this meeting was going on, and did not schedule time to have it.

fmalpartida, 2015-07-11:
Yep, missed it altogether! 

How about if we reschedule form next week?
Tagging does indeed help.

boelle, 2015-07-11:
I have allready said that i will not be along for the next month or so...

closed 2015-07-11 by Wackerbarth

thinkyhead, 2015-07-13:
Next meeting in 2 weeks? Consider this advance notice, and set those smartphone alarms…

Jnesselr, 2015-07-13:
I believe the other issue said it was the 18th.

Wackerbarth, 2015-07-13:
Please refer to issue #2424 for discussion about the next hangout.