MarlinFirmware/Marlin #2342: Improvement - Add deceleration option to movements

luisonoff, 2015-06-27:
Hello, I had an idea that could help improve printing quality. Not sure if it is already implemented or has been tried before, or if it can be done at all.

Rigth now when you make a movement (printing or travelling) from point A to point B, you only specify speed and distance, and firmware handles acceleration. I propose to add an optional deceleration, so  it reaches the final point at speed 0, and accelerate again in next movement.

Of course this will slow printing, that's why it should be optional or activable somehow, but I believe it would improve prints and specially dimensional accuracy.

What do you think?

thinkyhead, 2015-06-28:
For print quality, smooth movement (where the motor tries to match the speed of the following movement) is far superior to stoppage or dropping the speed to 0. In fact, stoppage like that will only increase blobs and spots where the nozzle heats the previous layer, and it will prevent the printer from almost ever reaching the asked-for movement speed.

luisonoff, 2015-07-03:
Thank thinkyhead for your comment. I sure did not think about oozing or extra heating.

Still, I think this method could improve dimensional accuracy in printers with heavy printing heads, like many extruders, that have lots of inertia. And this would be of course and optional behaviour.

I was thinking of a case scenario of printing a 90º corner. Print head is moving at 100% in one axis, and suddenly (infinite deceleration?) has to stop in that axis and move in the other axis. I believe many printers can't do this reliably, because of inertia, and they make 90º corners deformed.

Oozing should not be a "big" problem if you monitor speed and adjust flow accordingly while accelerating and decelerating. Also extra heating could be avoided if you set a high acceleration and deceleration.

Of course, it will slow prints and maybe it never reaches desired printing speed.

I am not familiar with Marlin code, so I have no idea how hard this could be, and I understand it is very unpopular request, but if somebody decides to try it I would be glad to beta test it.

Maybe a third opinion? Anyone?

Wackerbarth, 2015-07-03:
I believe that you are asking for something that is already a part of Marlin.

If you look, I think that you will find that we already have a limit on per-axis acceleration. This also applies to de-acceleration. Why should there be different values?

 As you describe a corner, the planner will slow things down as it approaches the corner.

If you have a heavy print head, you will need a lower maximum for this limit.

The problem is that, effectively, the extruder has additional "mass" and there is a time delay in its response. Hence, you get better volumetric accuracy if you keep the extruder moving at a constant speed and feed rate. (At the expense of rounding corners)

So, it becomes a tradeoff.  I use different limits for various objects depending on the aspects of the particular object that are most important to me.

luisonoff, 2015-07-03:
Thanks for the answer.

I didn't know this limit was also being used for deceleration, that is smart. I see no reason to have different limits.

You said that reaching a corner print head slows down, to what speed? Is this adjustable?

From the rest of your answer I understand that rigth now there is no implemented way of adjusting flow in the middle of a movement to take into account acceleration and deceleration, is this like this? So this explains why it is much better to try to keep a constant speed, right?

Wackerbarth, 2015-07-03:
"right now there is no implemented way of adjusting flow" -- That is not completely true. We do adjust the FEED rate proportionally with the change in the axis speeds. However, because of inherent mechanical characteristics, that is not the same as adjusting the EXTRUSION rate (In the short term).

As you may note in another thread, there is a discussion about alternate algorithms that have been tried in other firmware series.  These revised algorithms are all an attempt to improve the match between the EXTRUSION rate and the velocity of the nozzle.

It is certainly possible that we will be trying some of these ideas in a future version of Marlin.

thinkyhead, 2015-07-04:
> You said that reaching a corner print head slows down, to what speed?

First, you should think of each axis as having independent movement and independent inertia….

So let's say it is a square corner aligned with the bed. You will get X accelerating from 0 to its max speed (or as high as it can reach based on its max acceleration), then X decelerates all the way down to 0 before the Y line is started.

Next, imagine going from a straight line on the X into a diagonal line in the same direction on X. The planner can look ahead and decide that X doesn't need to decelerate all the way down to 0, but it still needs to decelerate close to 0 because Y is starting from a resting position.

Finally, imagine going from a diagonal line in XY into another diagonal line in XY. Looking ahead, the planner can check the needed acceleration / deceleration of both axes and the slowest one will take precedence. The axes will only accelerate / decelerate to their new target speeds.

There are also Jerk values to take into account, which allow sudden speed changes under a small threshold.

closed 2015-11-03 by AnHardt