MarlinFirmware/Marlin #2344: How do I fix my fork?

thinkyhead, 2015-06-28:
Since __Development__ was replaced with the contents of __dev__, when I make a PR from my fork's my __Development__ branch it has lots of extra items. How can I fix up my fork so that it has the same __Development__ branch as __MarlinFirmware__ now has?

Wackerbarth, 2015-06-28:
To some extent it depends on how your Development differed from the upstream/Development before.

First you should locally rebase each feature branch of your pending work to your Development head.
This may require you to resolve some conflicts (which already was true before the replacement).

Then move your local Development to the new MarlinFirmware version of it.

Finally rebase each of the branches to this new base. (This operation should not require any intervention)

Wackerbarth, 2015-06-28:
@thinkyhead -- Since you are using Mac OS X, I suggest that you consider GitX http://rowanj.github.io/gitx/
The graphical interface helps you visualize the branching structure.

As a workflow, I would suggest that you rebase on rather than performing your often merges of the MarlinFirmware:Development branch.

For your convenience, I have cleaned up your commits leading to the creation of your ENABLE/DISABLE macro stuff. You might add my repo as a remote to yours and adopt (NOT MERGE) my version of redo_overrides_1 as a replacement for your own.
You should also adopt the MarlinFirmware:Development branch instead of your own that was merged.

gralco, 2015-07-01:
Here's how I did it:
```
git remote add upstream https://github.com/MarlinFirmware/Marlin.git
git fetch upstream
git rebase upstream/Development
git mergetool                   **only necessary if patching says it failed, if it did review and remember where the conflicts are**
git rebase --continue           **only necessary if patching says it failed**
git push -f origin Branch-Title
```

Then if you saw conflicts search for there occurrences:
```
<<<<<<< HEAD
upstream code here
=======
your conflicting code here
>>>>>>> YOUR BRANCH NAME HERE
```

Fix the conflicts by editing/selecting the code you'd like in your PR, make sure to remove the conflict markers when you're done.

Then per usual do:
```
git commit -a
git push
```

thinkyhead, 2015-07-04:
I would like to reset my "Development" branch to match upstream/Development and drop all my commits in that branch, because I only ever made changes to keep the branch identical to upstream/Development. I don't care about preserving its commit history. If I can drop my copy of Development altogether and just base my features on upstream/Development then I would prefer to do that, then I don't have to keep updating my copy of the Development branch.

Once I have a clean Development branch of one sort or another, then I will rebase my other branches starting from upstream/Development (or my clean Development branch copy) if possible.

AnHardt, 2015-07-04:
@thinkyhead 
Maybe there is a more convenient way to do it.
For me worked hard resetting the local Development branch to a point before dev was forked.
Than pulling upstream/Development.
Than i made new feature-branches by cherry-picking the old ones.

The key to update the origin/Development was to make an other origin/branch the default, than un- and re-publishing and making origin/Development default again.

thinkyhead, 2015-07-04:
@AnHardt Actually I ended up just deleting it and recreating my `origin/Development` from `upstream/Development`.

It seems most convenient to start a new feature with `git checkout upstream/Development -b my_new_feature` rather than having to always update my `origin/Development` branch first and using that as the basis.

The only thing I want in my local `Development` branch is the latest `Development` code from `upstream/Marlin` anyway. I don't want to modify it, just keep it in sync.

closed 2015-10-12 by thinkyhead