MarlinFirmware/Marlin #2353: marlin development report with features Requests

taalabash, 2015-06-29:
i tried the marlin firmware from the development branch , it has very great modifications and it's much better than the released marlin especially the g29 and g30 routines because the printer knows if the allen key probe is deployed or retracted and that made the printer smarter .

but i think that the z feedrate while probing is slow so i saw the g30 routine in marlin_main.h to find that the feed rate in the run_z_probe() function is homing feed rate/4 i change it a little bit to make it feedrate = Z_TRAVEL_PROBE_SPEED ; after defining  Z_TRAVEL_PROBE_SPEED in the configuration .h so i can change it easier .

also in the configuration.h in the allen key deploy or retract moves the feedrate was
#define Z_PROBE_ALLEN_KEY_STOW_1_FEEDRATE HOMING_FEEDRATE_X  
and that gave a compiling error so i made it 
#define Z_PROBE_ALLEN_KEY_STOW_1_FEEDRATE homing_feedrate[Z_AXIS] 
and that fixed the problem

i have a feature request : 
maybe you can add a user defined thermistor table ( i modified the number 1 thermistor table to make may thermistor works because it is not one of the predefined thermistors )

and also maybe you can add a user defined UI lcd.. or should i only define lcd pins in pins.h for my controller ( i am using 99_pins.h with a home made mega shield )??

thank you so much for your amazing work .. you are making our printer smarter

taalabash, 2015-07-01:
Hello again 
I am still using the development version
I was testing the movements of my delta printer with marlin development version by giving my printer a gcode generated by an inkscape gcode generator plugin 
The gcode include many g03, g02,  and g4 p0 

The problem is that while the printer moves by the code it stops and starts to move down in a diagonal line and a very jerky moves until it hits the bed

thinkyhead, 2015-07-04:
Good observations and feedback! We need to know the defaults that make the most sense. Changing the `Z_PROBE_ALLEN_KEY_STOW_1_FEEDRATE` to `homing_feedrate[Z_AXIS]` should work fine.

> maybe you can add a user-defined thermistor table

I can see some advantage to that, but the way you did it is also fine. What was the make/model of the thermistor, if I may ask?

> maybe you can add a user defined UI lcd.. or should i only define lcd pins in pins.h for my controller

You can start with whichever `pins_THE_BOARD.h` file best matches your hardware, make a copy, and add a custom define to your `boards.h` file for it. Basically, implement it just as if you were adding support for a whole new board to Marlin, then keep a copy of all your modified files in a safe place. BOARD_99 has been hanging around for legacy reasons, but it might be good to set it up with more documentation and make it the starting point for custom hardware.

taalabash, 2015-07-04:
Actually i dont know the make of the thermistor i use but i got its table using another high accurate thermocouple...  I think there is only a few number of people doing this way so it's not a big problem we can modify one of the existed tables

closed 2015-07-28 by taalabash