MarlinFirmware/Marlin #2357: Z home offset not working (M206)

Lukelectro, 2015-06-29:
Similair to #1936.

My mendel90 homes upward to top endswitch at approximately 190mm above the bed (measured from nozle to bed.

I replaced parts of the bed, now it is 190.something. (190.2 to 190.5 I think)

Instead of changing configuration.h and reuploading firmware I thought about using M206 to set a home offset for Z. This does not work.

So, I got the newest "dev" branch (https://github.com/Lukelectro/Marlin/tree/Development)
It still does not work.

Here is what I did and what I expected:

Homed Z (to top): G28 Z
After home, M114 says Z is at 190. That is correct.
Go to Z0: G1 Z0
Report: M114 confirms Z0. Head is above the bed by approx. 0.2mm
M206 Z0.5 (set home offset)
G28 Z to home Z again. M114 now reports 190.5 after home so that should be right, thats what I expected. 
G1 Z1 to stay clear for sure, then down in small steps using the buttons in the gui.
Some 0.2mm to 0.5mm above the bed the head won't go down any further. That's not what I expected.
M114 reports Z at 0.5mm. So there should be room to go down further... Lets be bold: G1 Z0. Printer does not move. That's strange.

Does anybody else have a printer homing to top and the same issue?

(I will try to fix this one too)

Lukelectro, 2015-06-29:
That pull request includes way too many commits because I'm still a bit of a newb to github. Please tell me what to to better next time (Preferably in a positive manner)

Kind of cool that it picked up on the commit automatically :)

thinkyhead, 2015-07-03:
Hmm, that is interesting that Z won't move when it has 0.5 as the position. But maybe the raw stepper position is being used somehow….

I saw a commit recently related to this... did you fix it already?

Lukelectro, 2015-07-04:
Yes, but I made some newbe errors in the pull request.

now others have also changed the same part of the code, so I'd have to test this before making a new PR. I made a new branch but had merge conflicts, so now it's maybe even messy-er then before :( and the commit that should fix this is now named "merge conflict"...

sarf2k4, 2015-09-05:
Hopefully this issue will be solved soon. I use prusa i3, my endstop is at the bottom and setting z offset to 0.3 would mean that after homed the nozzle is far from bed, but issuing command to z0 is not possible.

The only thing can go is to have the physical nozzle below the bed level, set the offset to negative, issuing z0 command will move the z axis upwards

Lukelectro, 2015-09-05:
Have you tried the newest development version yet?

I was working on this, but others have changed that part of the code as 
well so I'd have to test again to see if it's not already fixed before I 
start breaking things :)

Once I find some time somewhere I'll look into this again.

dposea, 2015-09-06:
In the release branch Z is constrained to 0.5 in the definition of the configuration  function in ultralcd,cpp, 

in the development version this is fixed:
MENU_ITEM_EDIT(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);

minimum is set in Configuration.h . There some other issues caused by using the same variable zprobe_zoffest for a couple of different things, including the  z offset value. I'm still trying to figure out a good solution, if it's still an issue. I can not get the Dev firmware to run this week, it tries to sink the hotend into the bed.

sarf2k4, 2015-09-07:
Hi, I tried with the latest dev, downloaded yesterday, it is still the same, having z home offset at 0.5 will not make the z axis go lower than 0.5

I think this has to do with the marlin's travel limit such as what's registered after z home marlin takes that position as z=0.00mm, but what's displayed when requesting m114 is from the home offset, thus marlin limits the movement for positive home offset values

I'm sorry, I don't know about marlin codes and how it works, this is just an assumptions

thinkyhead, 2015-09-07:
The way things are currently set up, the `Z_MIN_POS` is based on treating the homed Z position as 0. So if you have `Z_MIN_POS` set to 0 the printer won't move lower than your "home position" no matter what offset you applied. Thus you can't move lower than 0.5mm – your homed Z position. To get around this, set `Z_MIN_POS` to `-0.5`. Then you should be allowed to move 0.5mm lower than the home position.

sarf2k4, 2015-09-07:
But this is somewhat inefficient though, to keep flashing the firmware on each changes. My work around is that luckily my head is outside of print bed and I put it lower than print bed, use negative z home offset

Maybe you can implement some workaround

if (z-home-offset >0)
z min pos = z min - z offset

or adds g92 z home offset value after homing the z axis?

just a suggestion :)

thinkyhead, 2015-09-07:
Sorry it's implemented this way at present. The `Z_MIN_POS` and so on are meant to be set just once according to the physical limitations of the machine (based on fixed endstop positions). If your Z axis is 190.5 units in length, ideally you should just set `Z_MIN_POS` to `0` and your `Z_MAX_POS` to `190.5` and then you shouldn't need any `M206` `home_offset`.

closed 2015-11-03 by AnHardt