MarlinFirmware/Marlin #2362: Z_PROBE vs. Z_MIN

Wurstnase, 2015-07-01:
Hi all,

i have no idea why we have a Z_PROBE_PIN and a Z_MIN_PIN. I guess this will make some confusion.

I run autobedleveling with my Z_MIN-Endstop. Anyone how can tell me the advantage of this?

linchunguang, 2015-07-01:
Motor self-locking function

Wurstnase, 2015-07-01:
Can you explain?

Roxy-3DPrintBoard, 2015-07-01:
> i have no idea why we have a Z_PROBE_PIN and a Z_MIN_PIN. 

Some people home their Z to the Max position and also have a Z Probe.   There are advantages to this.  If you print things that fill the entire Z build envelope, you don't have to raise the nozzle prior to engaging the Z probe.  You have plenty of room.   If you just use the Z Probe to home, you have to give up what ever your Z probe raises prior to engagement in the Z direction.

Wurstnase, 2015-07-01:
I understand to have a probe and a z max endstop. But not a probe and a z min endstop.

Wackerbarth, 2015-07-01:
I think that the intention was to allow either a probe or a Z_MIN end-stop rather than having both at the same time.

The distinction has to do with how they are used in various places within the code.  For example, the end-stop would be available during operation, but the probe would not. The different names are used to control the (convoluted) code of how the sensor is used under various conditions.

closed 2015-07-08 by Wurstnase