MarlinFirmware/Marlin #2363: Info AutoLeveling & Probe

Drakelive, 2015-07-01:
Good Morning

In recent days there is great interest on topics such as: AutoLeveling, Z-Probe and Z-Offset
In parallel I am configuring my printer to work with AutoLeveling
I use an inductive proximity sensor SN04-N  :  
http://www.amazon.com/4-8-30V-3-wire-Inductive-Proximity-Sensor/dp/B00979RA1M/ref=pd_sim_hi_6?ie=UTF8&refRID=0ECWKWZR0EXKXV5DXEAQ

This sensor is used as the Z-EndStop (connected to Z MIN) and for the AutoLeveling. 
This is the photo of the carriage of the X: 
https://goo.gl/photos/WYKoWK4uh94DmqgG9 

With this solution, I can do the Home axis (G28) and the AutoLeveling of the plane (G29) 
Probably I am wrong but I would like to hear your advice. 
 
To manage a proper Homing axes, especially Z, I had to move the X-Endstop so that the sensor SN04-N was 
above the printing plane: https://goo.gl/photos/sj9sYUs6dWHqhRM7A
 
In this case I can manage Homing all the axes properly but unfortunately lose 33.5 mm printing area along the X axis.
The ideal case would be to use the entire area of the plane as in this photo: 
https://goo.gl/photos/ZPsiDHQokRYCEWECA

Unfortunately in this way the sensor SN04-N comes out from the plane, and it is not possible to manage a correct reading of Homing for the Z axis

Is there a trick software to handle this?
We can think of a new schedule management in marlin?

Regard
Drake

AnHardt, 2015-07-01:
Have a look at `Z_SAFE_HOMING`.

MoonshineSG, 2015-07-02:
What you need are the  settings that describe the offset of the probe in relation to the nozzle.  In the case of below settings, even if you probe at X0 Y0, the firmware knows that your probe is at X25,Y29 away and will actually move to those coordinates (nozzle will be at X25,Y29, probe at X0,Y0) 
The Z setting is to tell how much higher the probe is vs the nozzle. 

Configuration.h around line 398 (as of today source code):
```
// these are the offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER -25
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -29
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.35
```

Drakelive, 2015-07-02:
Hi

Z_SAFE_HOMING was useful to run the Probe Z-axis in the center of the
plane, Thanks @AnHardt

@Ovidiu Hossu

I know these variables and have been configured according to my proximity
sensor
Now G28 (Home) and G29 (Auto-leveling) working but needs to further
adjustment.

The variables X_PROBE_OFFSET_FROM_EXTRUDER, Y_PROBE_OFFSET_FROM_EXTRUDER,
Z_PROBE_OFFSET_FROM_EXTRUDER were configured according how it was mounted
proximity sensor SN04-N.

Now when I run the command G28 the extruder nozzle is spaced about 0.6mm (I
can not measure it with precision )
from the plane. How can I change this distance?

In this position, the command M114 says that the Z axis is positioned at
the value of Z_PROBE_OFFSET_FROM_EXTRUDER.
The correct value of this variavile should allow you to place the extruder
in contact with the plane, sending the command "G0 Z0" "
But why then "G28 Z" does not move the extruder Z0?

Drake

MoonshineSG, 2015-07-02:
Not sure I fully understand... Is your Z offset value correct? 

There's also a M851 that is worth a look..

Drakelive, 2015-07-02:
the variable Z_PROBE_OFFSET_FROM_EXTRUDER defines the distance between the
sensor Auto-alignment and the tip of the extruder. This value shall be
accurate. Once configured Z0 will be the point of contact (physically)
between plane and extruder

In Configuration.h there is a variable to define an offset between the tip of
the extruder and the plane.

Why?

Wurstnase, 2015-07-02:
Homing is for the printer to know where it is, not to drive to 0-0-0.

closed 2015-09-25 by AnHardt