MarlinFirmware/Marlin #2364: Feature Request - LCD displays current print speed in mm/S ?

mr-sneezy, 2015-07-02:
There has been discussion on another issue here about the value of the X, Y & Z print position ifno bar on the Full Graphics LCD displays. This is related to that discussion, in that the relatively useless (while printing) X & Y position fields could be better used (Z position is actually useful).

I have a suggestion that the current printing speed in mm/S might more useful compared to the aforementioned X & Y numbers.
I'm not sure it's even possible to get this info realtime from the current G-code line being run or the calculations done during printing by the Marlin firmware, if not then the request is not possible.
If it is, then the display bar that usually has X, Y & Z data might have...

Speed: 30 mm/S   Z: 000.0

Perhaps more useful, certainly more interesting ??

AnHardt, 2015-07-02:
For performance reasons the display is updated only ~ once a second.
As "print speed" is a fast changing value, this is as useless as the x,y-coordinates.

AnHardt, 2015-07-02:
Thinking about it, we could talk about the "given-print-speed" from the G-codes. Displaying the "actual-print-speed" (during acceleration - deacceleration) makes no sense.

mr-sneezy, 2015-07-02:
That would be fine. My thought is that during long perimeters and during infills etc where speed doesn't change for many seconds the info might be good to see. I guess I'm thinking of it as confirmation of what you expected to happen, or a value you will change when you see it not doing the best job at that speed.

thinkyhead, 2015-07-04:
There has also been some discussion of adding a second "debug info" display, accessible with a double-click or click-and-hold, which could display other useful info, including Z, layer number, total filament used, etc.

atunguyd, 2015-07-09:
Must say I agree with Sneezy on the XY co ordinates being pointless. How about something that shows layer number vs total when doing a print from SD card? It would require that the gcode file is parsed to determine maximum layer but I see that as having much more value then the current XY location.

thinkyhead, 2015-07-09:
> layer number vs total

Layer height isn't "known" by the firmware, but I supose it could be gleaned by looking at the average size of several Z moves.

atunguyd, 2015-07-10:
Scott, 

I did not mean layer height but total print height. At the start of the print the firmware could skip to the end of the file and tag backwards till it finds the last Z command. It then uses this as the total height and then could display the z like "9.3/36mm".

daid, 2015-07-10:
> Displaying the "actual-print-speed" (during acceleration - deacceleration) makes no sense.
Actually, I think it does. Displaying the average speed would be quite interesting. So we can check the claims of "I'm printing at 150mm/s!" While in reality acceleration is keeping you back.

Another interesting value to keep track of and display is the average extrusion in mm^3/sec. As the more I learn about materials and machines, this is actually one of the most important values to keep track off.

thinkyhead, 2015-07-10:
> tag backwards till it finds the last Z command

My Ending GCode moves Z up by 10mm before moving Y to the back of the bed. That would also throw off the total height, but maybe it can look for the last Z move before an E+ move.

mr-sneezy, 2015-07-10:
All good ideas :)
I'm happy to see desired or average speed and Z height :)
The speed would need to be averaged over a few seconds so it made sense per layer (assuming some layers are small).

closed 2016-05-04 by jbrazio