MarlinFirmware/Marlin #2365: Servo should be turned off when performing z axis move during bed autoleveling (G29)

foulowl, 2015-07-03:
Hi folks,

Currently the block diagram for bed autoleveling is as follows: (for each of the four points)
Lower arm -> lower z axis until switch hit -> record data -> raise z axis -> raise arm

The downside of this is that my servo arm skips around roughly +-5 degrees during any axis moves. Discussion on #reprap IRC channel states this servo skipping is a known issue.

A better procedure would be:
Turn servo on -> lower arm -> turn servo off -> lower z axis until switch hit -> record data -> raise z axis -> turn servo on -> raise arm -> turn servo off

Thanks!

(As an item of note, I have a RAMPS 1.4 board.)

Roxy-3DPrintBoard, 2015-07-03:
You must be running a very old version of Marlin.   There are options to control the engagement and retraction of the probe now.   And the default is to leave it engaged during the G29.

foulowl, 2015-07-03:
Marlin version was checked out three days ago.

Let me clarify: I'm not interested in controlling the engagement and retraction of the probe. I want G29 to do exactly what it does now, except that during z axis moves I want the _supply voltage turned off_ to the servo pins. The reason for this is that there appears to be a hardware bug in the RAMPS1.4 board that causes the servo arm to randomly jump around during any axis moves.

Wackerbarth, 2015-07-03:
"during z axis moves I want the supply voltage turned off to the servo pins" -- During ALL z-axis moves? Or only some of them? I cannot imagine the hardware being very happy if this supply is being switched off and on very often.

If you just want it done associated with probing, then what we need to do is implement the switch in G-code and make the probe action something that can be customized. (We have discussed doing this for all of the probe actions, unifying deltas, sleds, etc.)

foulowl, 2015-07-03:
To be more specific: "during all z axis moves while G29 is executing I want the supply voltage turned off to the servo pins".

foulowl, 2015-07-05:
Just to confirm: I built an external 5V power supply for my 9g bed leveling servo, but it had no effect on this issue. So this isn't an issue of the servo 5V supply being underpowered, as suggested on the reprap forums. 

I like the suggestion that was made on IRC #reprap, which is to turn off the servo during the z axis moves, then turn it on again after the move has completed.

An item of note: this is a crucial bug for me, as with the random jiggling on my servo I can't get a proper bed autolevel to complete. It will randomly work sometimes, it just depends on how much the servo arm has jiggled around. 

If I post a bounty on this issue, would anyone like to work with me on getting it resolved?

Thank you!!

Ziggy2013, 2015-07-05:
There is already a parameter in configuration.h to turn the servo off between moves. By default this parameter is commented out.

#define PROBE_SERVO_DEACTIVATION_DELAY 300

Uncomment this parameter and the servo should stay stable after moves. This parameter also introduces a delay after the servo write to allow the servo to settle.

foulowl, 2015-07-05:
That works perfectly! Thank you for the help, sorry I missed that.

closed 2015-07-05 by foulowl