MarlinFirmware/Marlin #2366: Q: Probing range

AnHardt, 2015-07-03:
At the moment i try to test `AUTO_BED_LEVELING_GRID` with more than 3 points.
With this:https://github.com/AnHardt/Marlin_O/pull/7/files configuration is get an error:

    SanityCheck.h:195:10: error: #error The Y axis probing range is too small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS.

Why do we make this test at all? Why do we limit the distance between the probings? Why is the distance between nozzle and the probe a criteria?

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/SanityCheck.h#L173 - ...

thinkyhead, 2015-07-04:
> Why do we limit the distance between the probings?

Possibly it was intended to limit the number of probe points in a small space more than the distance between them, _per se_. This test was carried over from the older code.

> Why is the distance between nozzle and the probe a criteria?

I'm not sure why the distance between the nozzle and probe is used as the minimum distance between probes. It's only that way now because the original implementation used it that way. This might make more sense…

      #define PROBE_SIZE_X (MINIMUM_PROBE_DISTANCE * (AUTO_BED_LEVELING_GRID_POINTS-1))
      #define PROBE_SIZE_Y (MINIMUM_PROBE_DISTANCE * (AUTO_BED_LEVELING_GRID_POINTS-1))

AnHardt, 2015-07-06:
Traced back the origin of the code to: https://github.com/MarlinFirmware/Marlin/commit/3afe66bb0c59f94d551b0ed10dda27e3d8683ec5
It is a fix for #1248.
The problem in #1248 was that the probe-rectangle did not fit the area the probe can reach. The fix worked, but is a bit over-restrictive. Correcting the rectangle sufficiently at one side would have been enough.
Since we now have a proper test for the probe-rectangle (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/SanityCheck.h#L154 - L172)
we can drop: 
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/SanityCheck.h#L173 - L202 what is the shifted and mutated original fix.
Tested for cartesian printers, with probe_offsets up to +-50mm and up to 8x8 points.

thinkyhead, 2015-07-07:
Sounds good to me! The check to see if points can fit is definitely getting too nitpicky. Sometimes it might be fun to probe every 2mm.

closed 2015-07-11 by AnHardt