MarlinFirmware/Marlin #2369: Full Graphics LCD no longer displays Z position (dev Branch)

WheresWaldo, 2015-07-03:
I am not sure when this happened, it is working in 1.0.0 and 1.0.2. The Full Graphics LCD in 1.0.3 dev behaves in a way that does not display the Z height. As expected when starting up the display for X, Y and Z all show ----.-. When you issue a <b>G28</b> command to home all axis the display will show first the X axis of 000.0 when home, then the Y axis of 000.0 when home. Upon moving to the middle of the build plate to home the Z axis, the LCD correctly displays X and Y locations, after homing the LCD panel still displays ----.- for the Z location. Any subsequent command such as <b>G29</b> or any other command that requires moving along the Z axis completes as expected but the display still shows ----.- for Z height.

I am not sure if this is a consequence of fixes to <b>ABL</b> for <b>Z_OFFSET</b>.

ntoff, 2015-07-03:
I'm pretty sure it's probably because they've now made it so that if the steppers are deactivated, it now shows ---.-- because the position is now somehow unknown due to it being able to jump to the next full step.

Try turning off the options to disable the steppers when not in use and reflash.

AnHardt, 2015-07-03:
@ntoff 

> I'm pretty sure it's probably because they've now made it so that if the steppers are deactivated

The opposite is the case.
Yes, we deactivate the display now for an axis, where we do not know the exact coordinates.
Yes, we do not now the exact position, when the steppers where switched off.
No, we did not changed the default behavior for switching off the motors. By default they are NOT switched off.

    #define DISABLE_X false
    #define DISABLE_Y false
    #define DISABLE_Z false

    // @section extruder

    #define DISABLE_E false

This change, to show --- when the motors where disabled, enabled us to find a bug with  M84 (DEFAULT_STEPPER_DEACTIVE_TIME) where all steppers where shut down instead of respecting the DISABLE_? settings.

@WheresWaldo 
Please check your DISABLE_? settings.

ntoff, 2015-07-03:
>The opposite is the case.

Opposite? You just repeated what I said, how's that opposite?

AnHardt, 2015-07-03:
@ntoff 
The steppers are NOT deactivated! Furthermore they are not deactivated by error anymore.

AnHardt, 2015-07-03:
@ntoff 
Sorry overlooked the 'if'.

WheresWaldo, 2015-07-04:
I am going to comment and close, that is exactly what it was #define DISABLE_Z

closed 2015-07-04 by WheresWaldo