MarlinFirmware/Marlin #2371: arcs in delta printers

taalabash, 2015-07-03:
Does Marlin firmware support arc or G02 and G03 for delta printers ??

I am giving my delta printer a gcode with many g03 and g02 codes  but the printer starts to do strange move with vibration whenever it reach a gcode line that contains g02 or g03 until it hit the bed if i dont shut it off

thinkyhead, 2015-07-04:
`gcode_G2_G3` calls `plan_arc` which then calls `plan_buffer_line` to move the axes directly. At no point does it do any delta calculation. And it doesn't call `prepare_move` which is the delta-compatible way to add a move to the planner. So it appears that `gcode_G2_G3` is not compatible with non-cartesian machines.

taalabash, 2015-07-04:
Actually i prefer marlin over ather firmwares but it is a negative point if we can't perform arcs in deltas 
I know you are doing your best for users and thank you for that but  is there any chance to make it work?

AnHardt, 2015-07-04:
@taalabash 

>but is there any chance to make it work?

Yes, but it will take a while. #2377 only deactivates, the not working proper, G2/3 for deltas.

MagoKimbra, 2015-07-07:
I always ask apology for the intrusion ...
But do not just put:

clamp_to_software_endstops(arc_target);
    #ifdef DELTA
      calculate_delta(arc_target);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], arc_target[E_AXIS], feed_rate, active_extruder, active_driver);
    #else
      plan_buffer_line(arc_target[X_AXIS], arc_target[Y_AXIS], arc_target[Z_AXIS], arc_target[E_AXIS], feed_rate, active_extruder, active_driver);
    #endif
  }

  // Ensure last segment arrives at target location.
  #ifdef DELTA
    calculate_delta(target);
    plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], target[E_AXIS], feed_rate, active_extruder, active_driver);
  #else
    plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], feed_rate, active_extruder, active_driver);
  #endif

  // As far as the parser is concerned, the position is now == target. In reality the
  // motion control system might still be processing the action and the real tool position
  // in any intermediate location.
  set_current_to_destination();
}
I tried now on my Delta and makes circular movements ...
But it's something that I've never used it and I do not know if there is actually something wrong ...

thinkyhead, 2015-07-09:
@MagoKimbra That seems like a good idea, but the arc segments need to be further segmented so delta moves won't do weird dips. So this code might be better:
```
    clamp_to_software_endstops(arc_target);

    #ifdef DELTA
      feedrate = feed_rate;
      prepare_move_delta();
    #else
      plan_buffer_line(arc_target[X_AXIS], arc_target[Y_AXIS], arc_target[Z_AXIS], arc_target[E_AXIS], feed_rate, active_extruder);
    #endif
  }

  // Ensure last segment arrives at target location.
  #ifdef DELTA
    feedrate = feed_rate;
    memcpy(destination, target, sizeof(destination));
    prepare_move_delta();
  #else
    plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], feed_rate, active_extruder);
  #endif
```

thinkyhead, 2015-07-09:
#2405 follows that idea.

MagoKimbra, 2015-07-09:
Yes... Good job @thinkyhead !!!

taalabash, 2015-07-09:
@thinkyhead should we (users) try the marlin firmware from your delta_arc_support branch and give feedback or not yet ??

thinkyhead, 2015-07-09:
> should we (users) try the marlin firmware from your delta_arc_support branch and give feedback or not yet?

Sure! I haven't actually published that branch so let me do that first…

taalabash, 2015-07-10:
Hello again ...
i was trying the delta_arc_support branch but things doesn't went well because the printer stops when it reach a g2/3 code for few seconds then starts moving again with very slow steps (about 2 moves by second) and it randomly stops for many seconds while performing arcs 
it even stops randomly without arcs sometimes

i tried it many times without printing just to test its movements but i didn't knew if it is performing arcs in right coordinates or no because as i said it was very slow with many stops while moving

thank you again and i hope i can help more...

thinkyhead, 2015-07-11:
@taalabash I'm not surprised that it doesn't work very well. We might have to try some other strategy, such as allowing the arcs to be done without sub-dividing the arc segments, or only sub-dividing arc segments over a certain length.

JohnsonLv, 2015-07-11:
@thinkyhead 
 I met the same problem , I use delta type machine for laser engraving .when met G02/03 ,the machine stop and move slowly

thinkyhead, 2015-07-15:
I've updated the code to use the following method instead of subdividing segments (which already default to 1mm in length, small enough to make any "delta dipping" negligible):

    calculate_delta(target);
    #ifdef ENABLE_AUTO_BED_LEVELING
      adjust_delta(target);
    #endif
    plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], target[E_AXIS], feed_rate, active_extruder);

Please give it another test and see if it works properly.

taalabash, 2015-07-15:
@thinkyhead i tested the new code you added and the problem still exists as before (many stops with slow moves) but i tried some things that works

i made some changes in the marlin_main and motion_control from rich cattel  marlin version as  @MagoKimbra  said in his first post in this issue and tried it out to find that it works 
the current marlin version is very different from the old one so i don't know what to modify currently 

hope i am helping here

thinkyhead, 2015-07-15:
@taalabash Well, the changes I just posted are essentially the same as @MagoKimbra 's initial suggestion. The only difference is that I've added `adjust_delta(target)` if there is bed leveling enabled. As far as I know that line is needed. So, perhaps it is something about the latest Development code which is causing it to stutter.

Q: Are you using a graphical display? Does the Development branch in any other way (apart from arcs) perform differently than the current Release branch?

taalabash, 2015-07-17:
@thinkyhead i think you are right.. There something else that is effecting the moves because the printer sometimes stops randomly even without using arcs and it stops everytime i use arcs

Right now I can't test what you have done recently but i will test it in about 10 days because i am away from home

taalabash, 2015-07-27:
@thinkyhead @MagoKimbra  I tested the last version of marlin from development branch to find that the firmware now can make arcs move with delta printer 
the problem fixed so thank you so much for your work and for making marlin firmware better

closed 2015-07-27 by taalabash