MarlinFirmware/Marlin #2376: Development fails Repetier initialization in "ping-pong" mode

lrpirlet, 2015-07-04:
I downloaded the Development version yesterday 3-Jul in order to test the patch over the G29…
I always use Repetier-Host V1.0.6 in “ping-pong” mode where the host software waits the acknowledge before issuing the next command…
The downloaded one just fails. Repetier stops with
```
09:31:33.607 : N10 M111 S6 *118
09:31:33.607 : echo:ok
```
It looks like repetier is waiting for the ack… The GUI announce 2 commands pending… I waited about 2 minutes without any progress...

Note that it used to work with older version of the Marlin firmware…

a working one (development 2-6-2015)…
```
09:24:00.927 : start
09:24:00.927 : o: Free Memory: 2930  PlannerBufferBytes: 12Last Updated: 2015-06-00 12:00 | Author: LRP Tuned 7.05.15
09:24:00.927 : Compiled: Jul  3 2015
09:24:00.927 : echo: Free Memory: 2930  PlannerBufferBytes: 1232
09:24:00.927 : echo:Hardcoded Default Settings Loaded
09:24:00.927 : echo:Steps per unit:
09:24:00.927 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
09:24:00.927 : echo:Maximum feedrates (mm/s):
09:24:00.927 : echo:  M203 X200.00 Y200.00 Z1.0*?45.00
09:24:00.927 : echo:Maximum Acceleration (mm/s2):
09:24:00.927 : echo:  M201 X1000 Y1000 Z50 E5000
09:24:00.927 : echo:Accelerations: P=printing, R=retract and T=travel
09:24:00.927 : echo:  M204 P1500.00 R2000.00 T1500.00
09:24:00.927 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms)X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
09:24:00.927 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
09:24:00.927 : echo:Home offset (mm):
09:24:00.927 : echo:  M206 X0.00 Y0.00 Z0.00
09:24:00.927 : echo:Material heatup parameters:
09:24:00.927 : echo:  M145 M0 H180 B70 F0
09:24:00.927 : echo:  M145 M1 H240 B110 F0
09:24:00.927 : echo:PID settings:
09:24:00.927 : echo:  M301 P22.41 I1.56 D80.41 C1.00
09:24:00.927 : echo:Filament settings: Disabled
09:24:00.927 : echo:Z-Probe Offset (mm):
09:24:00.927 : echo:  M851 Z-2.33
09:24:01.473 : N1 M110 *2
09:24:01.956 : N1 M110 *2
09:24:04.842 : echo:SD init fail
09:24:04.842 : echo:Unknown command: ""
09:24:04.842 : ok
09:24:04.842 : N2 M115 *4
09:24:04.920 : ok
09:24:04.936 : FIRMWARE_NAME:Marlin 1.1.0 dev SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
09:24:04.936 : ok
09:24:04.936 : N4 M111 S6 *67
09:24:04.951 : N5 T0 *31
09:24:04.951 : ok
09:24:04.951 : N6 M20 *55
09:24:04.967 : echo:Active Extruder: 0
09:24:04.967 : ok
09:24:04.967 : Begin file list
09:24:04.967 : N7 M80 *60
09:24:04.967 : End file list
09:24:04.967 : ok
09:24:04.967 : ok
09:24:04.967 : N8 M220 S100 *73
09:24:04.967 : N9 M221 S100 *73
09:24:04.967 : ok
09:24:04.967 : N10 M111 S6 *118
09:24:04.967 : ok
09:24:04.967 : ok
09:24:04.967 : N11 T0 *42
09:24:04.983 : echo:Active Extruder: 0
09:24:04.983 : ok
09:24:33.673 : N26 G28 *7
09:25:20.099 : ok
```

	The Bad One

```
09:31:29.535 : start
09:31:29.535 : echo: External Reset
09:31:29.535 : Marlin 1.0.3 dev
09:31:29.535 : echo: Last Updated: 2015-06-00 12:00 | Author: LRP Tuned 7.05.15
09:31:29.535 : Compiled: Jul  4 2015
09:31:29.551 : echo: Free Memory: 2926  PlannerBufferBytes: 1232
09:31:29.551 : echo:Hardcoded Default Settings Loaded
09:31:29.551 : echo:Steps per unit:
09:31:29.551 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
09:31:29.551 : echo:Maximum feedrates (mm/s):
09:31:29.551 : echo:  M203 X200.00 Y200.00 Z1.00 E45.00
09:31:29.551 : echo:Maximum Acceleration (mm/s2):
09:31:29.551 : echo:  M201 X1000 Y1000 Z50 E5000
09:31:29.551 : echo:Accelerations: P=printing, R=retract and T=travel
09:31:29.567 : echo:  M204 P1500.00 R2000.00 T1500.00
09:31:29.567 : echo:Advanced variablesS=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
09:31:29.567 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
09:31:29.567 : echo:Home offset (mm):
09:31:29.567 : echo:  M206 X0.00 Y0.00 Z0.00
09:31:29.567 : echo:Material heatup parameters:
09:31:29.567 : echo:  M145 M0 H180 B70 F0
09:31:29.567 : echo:  M145 M1 H240 B110 F0
09:31:29.582 : echo:PID settings:
09:31:29.582 : echo:  M301 P22.41 I1.56 D80.41 C1.00
09:31:29.582 : echo:Filament settings: Disabled
09:31:29.582 : echo:Z-Probe Offset (mm):
09:31:29.582 : echo:  M851 Z-2.33
09:31:30.019 : N1 M110 *2
09:31:30.581 : N1 M110 *2
09:31:33.482 : echo:SD init fail
09:31:33.482 : echo:Unknown command: ""
09:31:33.482 : ok
09:31:33.482 : N2 M115 *4
09:31:33.576 : ok
09:31:33.576 : FIRMWARE_NAME:Marlin 1.0.3 dev SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
09:31:33.576 : ok
09:31:33.576 : N4 M111 S6 *67
09:31:33.576 : echo:ok
09:31:33.576 : N5 T0 *31
09:31:33.591 : echo:Active Extruder: 0
09:31:33.591 : ok
09:31:33.591 : N6 M20 *55
09:31:33.591 : Begin file list
09:31:33.591 : End file list
09:31:33.591 : ok
09:31:33.591 : N7 M80 *60
09:31:33.591 : ok
09:31:33.591 : N8 M220 S100 *73
09:31:33.591 : ok
09:31:33.591 : N9 M221 S100 *73
09:31:33.607 : ok
09:31:33.607 : N10 M111 S6 *118
09:31:33.607 : echo:ok
```

lrpirlet, 2015-07-06:
Repaired this bad behavior by changing the M111 code from 
```C++
inline void gcode_M111() {
  marlin_debug_flags = code_seen('S') ? code_value_short() : DEBUG_INFO|DEBUG_COMMUNICATION;
  
  SERIAL_ECHO_START; 
  if (marlin_debug_flags & DEBUG_ECHO) SERIAL_ECHOLNPGM(MSG_DEBUG_ECHO);
  // FOR MOMENT NOT ACTIVE
  //if (marlin_debug_flags & DEBUG_INFO) SERIAL_ECHOLNPGM(MSG_DEBUG_INFO);
  //if (marlin_debug_flags & DEBUG_ERRORS) SERIAL_ECHOLNPGM(MSG_DEBUG_ERRORS);
  if (marlin_debug_flags & DEBUG_DRYRUN) {
    SERIAL_ECHOLNPGM(MSG_DEBUG_DRYRUN);
    setTargetBed(0);
    for (int8_t cur_hotend = 0; cur_hotend < EXTRUDERS; ++cur_hotend) {
      setTargetHotend(0, cur_hotend);
    }
  }
}

```
to
```C++
inline void gcode_M111() {
  marlin_debug_flags = code_seen('S') ? code_value_short() : DEBUG_INFO|DEBUG_COMMUNICATION;
  
  SERIAL_ECHO_START; 
  if (marlin_debug_flags & DEBUG_ECHO) SERIAL_ECHOLNPGM(MSG_DEBUG_ECHO);
  // FOR MOMENT NOT ACTIVE
  //if (marlin_debug_flags & DEBUG_INFO) SERIAL_ECHOLNPGM(MSG_DEBUG_INFO);
  //if (marlin_debug_flags & DEBUG_ERRORS) SERIAL_ECHOLNPGM(MSG_DEBUG_ERRORS);
  if (marlin_debug_flags & DEBUG_DRYRUN) {
    SERIAL_ECHOLNPGM(MSG_DEBUG_DRYRUN);
    setTargetBed(0);
    for (int8_t cur_hotend = 0; cur_hotend < EXTRUDERS; ++cur_hotend) {
      setTargetHotend(0, cur_hotend);
    }
  }
  SERIAL_EOL;	//added By lrp to avoid Hanging the M111
}
```
Sorry, no idea why it works , I just tried that to "close" the line so that the ack is not taken a an itm to print... Someone knowing C++ should have a look...

thinkyhead, 2015-07-07:
Ok, #2394 includes this adjustment, which might just be a kludge, but we'll see. I'm not sure where the "echo:ok" is coming from. I can't find any "ok" with `SERIAL_ECHO_START` in front of it, and I can't find any place where `SERIAL_ECHO_START` is being called without something following it.

thinkyhead, 2015-07-07:
Actually, before we think about merging #2394, see if this also fixes the issue for you (somehow):
```
inline void gcode_M111() {
  marlin_debug_flags = code_seen('S') ? code_value_short() : DEBUG_INFO|DEBUG_COMMUNICATION;
  
  if (marlin_debug_flags & DEBUG_ECHO) {
    SERIAL_ECHO_START;
    SERIAL_ECHOLNPGM(MSG_DEBUG_ECHO);
  }
  // FOR MOMENT NOT ACTIVE
  //if (marlin_debug_flags & DEBUG_INFO) SERIAL_ECHOLNPGM(MSG_DEBUG_INFO);
  //if (marlin_debug_flags & DEBUG_ERRORS) SERIAL_ECHOLNPGM(MSG_DEBUG_ERRORS);
  if (marlin_debug_flags & DEBUG_DRYRUN) {
    SERIAL_ECHO_START;
    SERIAL_ECHOLNPGM(MSG_DEBUG_DRYRUN);
    disable_all_heaters();
  }
}
```

lrpirlet, 2015-07-07:
@thinkyhead, many thanks, The code above, that made its way in the development, completely solve the issue... I do not know where this "ok" did come from, except maybe if the acknoledge was mistakenly used as an argument (?continuation?) to the SERIAL_ECHO_START command...

closed 2015-07-07 by lrpirlet

thinkyhead, 2015-07-08:
It was the extra `SERIAL_ECHO_START`. If you didn't set either `DEBUG_ECHO` or `DEBUG_DRYRUN` then it would only print `echo:` with no newline, swallowing the next `ok`.

OutsourcedGuru, 2019-09-05:
Sounds like you might have this exact problem with M501 as well.  [Example problem](https://community.octoprint.org/t/serial-connection-commands-bufferend-until-fake-ack/11839) as seen in OctoPrint.