MarlinFirmware/Marlin #2380: M99 Free Memory Watcher

Roxy-3DPrintBoard, 2015-07-04:
@Wackerbarth

OK, I have reliable numbers.   Using the BeckDac Marlin fork at https://github.com/beckdac/Marlin there are 4390 bytes of free RAM for the heap and stack to share after reset with the firmware ready to accept commands.

After a G29 n 3   (This fork uses the original parameter name to select the number of points) there are 3767 bytes of free RAM that never have been used.

After a G29 n 5 there are 3255 bytes of free RAM that never have been used.

These numbers have a high level of confidence.  Built into the code is also the ability to look for data corruption within the free memory block (due to bad point dereferencing).

I think what this means is there is enough room to statically hard allocate G29's arrays if people want to go that way.  Even with a 5 x 5 grid being probed, there is more than 3 KB of untouched memory.  Personally, I don't think it would be a wise use of the memory, but there is plenty of memory there to do it.

@Wackerbarth, Are there any features or commands you would like added?   Right now the code is structured as a M99 command.    

M99 I    Initializes the free area so it can easily be watched for changes.
M99 D   Dumps the free area in a nice format that allows any errant data corruption to be spotted.
M99 F   Displays the Free Memory size of the block.   It also looks for, detects and reports any data corruption within the block.
M99 C x  Corrupts x locations within the current free memory block to help validate the M99 F command and M99 D command are working correctly.

The code can be structured as a stand alone file that is controlled by a single #define in Configuration.h to turn it on.  With it turned off, it would take no space.  The D and C commands can be disabled to further shrink the foot print in Flash memory.  A  case 99: m99_code(): break; with #ifdef's around it was added to the Marlin_main.cpp code that processes M commands.

Originally I was not thinking it had a place in the code base.  But because of your comments I did the code very clean and made sure it could be disabled (or enabled) with a single #define.

Wackerbarth, 2015-07-04:
@Roxy-3DPrintBoard — Excellent start. A well described test.

Which processor are you using?

However, you have looked at only part of the situation.
You should be able to report (separately) the maximum amount of heap used and the maximum stack depth.

You also need to stress the stack end by running some complicated print sequences (G-codes that use a lot of stack)

Now, if you will post your code, I won’t need to reinvent the wheel.

Roxy-3DPrintBoard, 2015-07-05:
I have a PrintRboard.  I think it uses an AT90USB1286.  

The problem with trying to report the exact amount of heap and stack used is the code has no way of knowing where __brkval and the Stack Point started.   But what it can do non-invasively is watch the remaining free memory from when it gets initialized.  In reality, there is very little information already on the stack or heap when Marlin starts processing commands.   The amount is not very significant but the point I'm making is this code will never know EXACTLY how much that is.

With that said....   Review the following cut and paste from PronterFace.   The M99 command gets initialized and reports where it found __brkval and the Stack Pointer.     A G29 n3 is issued and at the worst case point in that command's execution 3771 bytes were found to be free at 0x0EB7.

Then a G29 n5 was issued and Marlins worst case free memory dropped to 3259 bytes free at 0x10B7.    And then a M99 D command was given.    You can see how deep the heap activity got from the top and you can see how wound up the stack got from the bottom.   

And then just to show how things get displayed a M99 C 2   was issued to corrupt the free memory.   And then the free space was dumped with another M99 D command.   If an unexpected value is found (non-0xE5) it will print a ? at that position in the right column.   That makes it is pretty easy to find anything that has tampered with stuff in the middle of the free memory block.  And of course, the M99 F will report that the free space has been corrupted:

Connecting...
Printer is now online.
>>>M99 I
SENDING:M99 I
Initializing free memory block.
__brkval : 3421
Stack Pointer : 7883
4390 bytes of memory initialized.
Done.
>>>G28
SENDING:G28
>>>G29 n 3 V 4
SENDING:G29 n 3 V 4
Roxy's Enhanced G29 Auto_Bed_Leveling Code V1.81:
Full support at http://3dprintboard.com
Bed x: 20.00 y: 19.00 z: 4.69
Bed x: 104.00 y: 19.00 z: 4.61
..
Bed x: 188.00 y: 179.00 z: 4.28
Eqn coefficients: a: -0.00 b: -0.00 d: 4.68
Mean of sampled points: 4.386938
Correct +.14 with one clockwise turn.

Bed Height Topography:
 --0.23926 +0.22640 +0.30743
 --0.22273 +0.12057 +0.11627
 --0.10466 --0.03223 --0.17180
planeNormal x: 0.00 y: 0.00 z: 1.00

Bed Level Correction Matrix:
0.999999 0.000000 -0.001624
-0.000002 0.999999 -0.001257
0.001624 0.001257 0.999998
echo:endstops hit:  Z:4.28
>>>M99 F
SENDING:M99 F
Found 3771 bytes free at 0x0EB7
Done.
>>>G29 n 5 V 4
SENDING:G29 n 5 V 4
Roxy's Enhanced G29 Auto_Bed_Leveling Code V1.81:
Full support at http://3dprintboard.com
Bed x: 20.00 y: 19.00 z: 3.50
...
Bed x: 188.00 y: 179.00 z: 3.18
Eqn coefficients: a: -0.00 b: -0.00 d: 3.48
Mean of sampled points: 3.261902
Correct +.14 with one clockwise turn.

Bed Height Topography:
 --0.31047 --0.02506 +0.12278 +0.16445 +0.23721
...
 --0.07764 --0.02076 --0.01513 --0.08822 --0.15834
planeNormal x: 0.00 y: 0.00 z: 1.00

Bed Level Correction Matrix:
0.999999 0.000000 -0.001404
-0.000001 1.000000 -0.000760
0.001404 0.000760 0.999999
echo:endstops hit:  Z:3.18
[... 65 lines cut ...]
18B0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
18C0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
18D0:E5 E5 E5 02 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |   ?
18E0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
18F0:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
...
1D30:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
1D40:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
1D50:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
1D60:E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 |
1D70:E5 E5 E5 00 00 00 00 00 00 05 A2 00 00 01 80 00 |   ?????????????
1D80:00 00 00 00 51 A6 8C A6 7C 95 12 80 D9 95 3F 00 |????????????????
1D90:00 22 43 CD 1D CC 1D 3F 95 D9 80 FA 80 1D BA D7 |????????????????
....
1EA0:BB 40 1E CA 0E 54 42 E4 00 F1 1A C2 4C F2 34 00 |????????????????
1EB0:E0 1A 56 28 D1 00 00 87 F1 00 56 04 0B 1E 00 00 |????????????????
1EC0:20 00 CF F5 40 9F D5 9F AD 22 03 80 21 79 E1 1E |????????????????
Done.
>>>M99 F
SENDING:M99 F
Found 612 bytes free at 0x10B7
Found 1461 bytes free at 0x131D
Found 1182 bytes free at 0x18D4
Memory Corruption detected in free memory area.
Done.

Wackerbarth, 2015-07-05:
> 
> I have a PrintRboard. I think it uses an AT90USB1286.
IOW, 8k of RAM. If there are any Marlin implementations on smaller processors, then this version of G29 cannot be used since there would not be enough RAM available.
> The problem with trying to report the exact amount of heap and stack used is the code has no way of knowing where __brkval and the Stack Point started.
If you care about where __brkval really starts, you can save that value as the firmware starts up.
As for the stack depth, again you can get an “initial value” at startup. This will include some overhead associated with the initial frame stack. However, that frame is never released and can be considered in the same context as static variables. What is important (of interest) is the maximum amount of additional stack space that gets consumed while the printer is operational.

Note that this may well be in some part of the code other than G29.

If the maximum stack depth occurs during routines called while executing the G29 command, then (from the perspective of RAM consumption) it will make little difference whether the temporary matrix storage comes from static allocation, allocation on the heap, or allocation on the stack. However, if the maximum stack depth occurs in some other portion of the code, then static allocation prevents dual usage of that memory and should be avoided.
> But what it can do non-invasively is watch —
> 
> <redacted>

I’m not doubting the results that you have obtained.
However, I AM looking for tests to determine which part of the code requires the maximum stack (unknown at this time) and which part requires the maximum heap (almost certainly G29).

I also renew my request that you share the code that you have written.

Roxy-3DPrintBoard, 2015-07-05:
| IOW, 8k of RAM. If there are any Marlin implementations on smaller processors, then this version of G29 cannot be used since there would not be enough RAM available.

Ah...  No!  Not true!   Remember The Enhanced G29 lets the user specify how many grid points to probe.  People with smaller processors just can't specify so many points they run out of memory.

| If you care about where __brkval really starts, you can save that value as the firmware starts up.
| As for the stack depth, again you can get an “initial value” at startup. This will include some overhead associated with the initial frame stack. However, that frame is never released and can be considered in the same context as static variables. 

Right.  I understand what you are saying.  With deeper changes to the code base we could save the exact starting point of each.  But my point is it doesn't really matter.   What ever is necessary to get the printer up and running and accepting commands is what it is.  And it isn't likely to change much.  This code is very benign.  It doesn't need much except access to the current __brkval value.  

So right now this code uses the state of the machine when it is ready to start processing commands as it's starting point.  

|What is important (of interest) is the maximum amount of additional stack space that gets consumed while the printer is operational.

I've been thinking about this.  There are several ways to do this.   But probably the value in doing this is knowing exactly what Gxx or Mxx command is causing the memory hit.   It can be done manually by looking at the top of the stack using a M99 D command and from that looking at the return addresses to see who sponsored the calls that wound up the stack.  Another way would be what I'm looking at right now and seeing if I can hook the timer ISR with a light weight process that watches for the max stack and heap usage and records what command is being processed at the instant it finds a new max.   The good news is there is lots of memory in the middle of the free memory pool to save logs!  :)

| Note that this may well be in some part of the code other than G29.

Understood!  But almost for sure, G29 is the big heap pig.

| If the maximum stack depth occurs during routines called while executing the G29 command, then (from the perspective of RAM consumption) it will make little difference whether the temporary matrix storage comes from static allocation, allocation on the heap, or allocation on the stack. However, if the maximum stack depth occurs in some other portion of the code, then static allocation prevents dual usage of that memory and should be avoided.

Yes...  I haven't done it yet.  But the thing to do would be do a G29 and then a M 99 D to dump the free space.   Then start a print and at the end do another M99 D to see if things changed.   If the free memory is the same size and in the same place, nothing else ate into the free memory pool more than the G29 command.   (I haven't actually used this code to explore the system much yet)

| I’m not doubting the results that you have obtained.  However, I AM looking for tests to determine which part of the code requires the maximum stack (unknown at this time) and which part requires the maximum heap (almost certainly G29).

Understood!

| I also renew my request that you share the code that you have written.

The code is one new file that gets added to the code directory of the project.   And then only a couple of other changes have to be made.  In Marlin_Main.cpp you need to add a declaration up at the top somewhere saying:

#ifdef M99_FREE_MEMORY_WATCHER
void m99_code();
#endif

and then I put the actual call to it just above the M48 case: statement:

#ifdef M99_FREE_MEMORY_WATCHER
case 99:  m99_code();
    break;
#endif
// M48 Z-Probe repeatability measurement function

And then in Configuration.h you need 1 line to control the option:

//
// M99 Free Memory Watcher
//
#define M99_FREE_MEMORY_WATCHER		// uncomment to add the M99 Free Memory Watcher for debug purpose

Unfortunately GitHub doesn't allow files to be attached to posts.  And I don't have this hooked into a fork that you can pull from.   I can email the file to you if you want.  Or I could post it over at 3DPrintBoard.  Its up to you.  But once you have it, you can just wire up the three things I mentioned above and it should go first time.   You are a collaborator.  Can you add this to the Development Branch once you verify it?

scotty1024, 2015-07-05:
I took the time to think about a reasonable upper bounds check for the P argument to G29. I've added a separate commit to my PR for the qr_solve slimming update I produced.

Chris works with the sanguino based Boards (so he might be able to further comment). The '644 AVR has 4KB of RAM but with 64KB of FLASH I don't realistically foresee anyone flipping options and sweating FLASH bytes out of code to make bed leveling fit. Most '644 users, as Bo can attest, quickly realize the '644 is too small so they upgrade to the '128. In addition to a smokin' large 12KB of RAM it has the barely large enough 128KB of FLASH to fit your average set of options along with ABL. As I mentioned before, the official Printrbot firmware fork of Marlin uses 99.1% of its available FLASH.

I suspect squeezing 4KB of FLASH out of qr_solve will be welcome to them as it would allow them to grid map less expensive beds in larger printers instead of needing to use MIC6 beds due to only being able to do 3 point ABL due to FLASH running out.

On the topic of more accurate monitoring of real time high water marks for stack and heap...

I can go find malloc()'s sbrk(). My suggestion would be to invest 8 bytes to install high water mark recorders in malloc()'s sbrk() and the timer IRQ handlers.

M99 could then report, optionally reset, these marks.

Roxy-3DPrintBoard, 2015-07-05:
> I took the time to think about a reasonable upper bounds check for the P argument to G29. I've added a separate commit to my PR for the qr_solve slimming update I produced.

Do you know what the side effects of that qr_solve() bug are yet?  I am interested but I haven't dug in on that yet.

>  In addition to a smokin' large 12KB of RAM it has the barely large enough 128KB of FLASH to fit your average set of options along with ABL. As I mentioned before, the official Printrbot firmware fork of Marlin uses 99.1% of its available FLASH.

That's interesting...  I don't have an LCD Panel but I have most everything else turned on.  I only use about 85KB of flash.

> I suspect squeezing 4KB of FLASH out of qr_solve will be welcome to them as it would allow them to grid map less expensive beds in larger printers instead of needing to use MIC6 beds due to only being able to do 3 point ABL due to FLASH running out.

Well...  We can save 10KB of code as soon as the bug fix phase is over by using the iterative LSF code.   It is pretty small, but it takes 20 seconds to crunch all the numbers on a many point grid.

scotty1024, 2015-07-05:
Almost completely forgot...

I nominate Marlin's code base as re-affirming Wirth's Law. :-)

Wirth's law: 

"Software is getting slower more rapidly than hardware becomes faster."

scotty1024, 2015-07-05:
Found a URL that I thought contained an insightful discussion on Arduino's and their love/hate relationship with malloc()/free().

Thought it might help put some emotional cotton around Marlin's recent disturbance in The Force.

http://www.reddit.com/r/arduino/comments/101jb1/til_arduino_does_not_like_malloc_and_free/

Stumbled into it looking for the easy sbrk() answer prior to getting out the pliers and propane torch.

scotty1024, 2015-07-05:
Here we go! The definitive document on malloc in AVR-Libc. Complete with color diagrams, tuneables and wisdom of the lib-AVR creators.

http://www.nongnu.org/avr-libc/user-manual/malloc.html

scotty1024, 2015-07-05:
So who wants to play with __malloc_margin first?

😀

Roxy-3DPrintBoard, 2015-07-05:
I can confirm the very first picture in that document is the way memory is organized in Marlin.  The other pictures do not reflect what we have.   

And...  Just for giggles, I made the following change to the  M99 code:

#ifdef M99_FREE_MEMORY_CORRUPTOR		
	if ( code_seen('C') ) {
		int x;			// x gets the # of locations to corrupt within the memory pool
		x = code_value();
char test[x];
  		SERIAL_ECHOLNPGM("Corrupting free memory block.\n");   
  		ptr = (unsigned char *) __brkval;
  		SERIAL_ECHOPAIR("\n__brkval : ",(long) ptr );
  		ptr += 8;

In the Arduino environment, we can allocate variable size arrays on the stack at execution time.   If you load the M99 command you can play with that.    Do a M99 I to initialize and then a M99 D to dump the memory.  Everything looks normal.   Then do a M99 C 3 and you will see new stack frames offset 300 bytes higher on the stack.

My vote would be to allow the dynamic allocation based on the number of grid points desired but move them to the stack.

scotty1024, 2015-07-05:
You will note the lack of null checking in the original qr_solve. The code was written for Big Iron where page 0 in each virtual process space is left unmapped so naughty kids that don't null check their malloc' get a free SEGV error.

No free SEGV's on the Arduino's. So another not so obvious potential head gouges your expensive MIC6 bed issue. If library code, or Marlin code, doesn't check each and every malloc call to be non-null things can get stomped on.

What *I* learned today: Don't malloc like this on Arduino's!

  x = ( double * ) malloc ( n * sizeof ( double ) );
  jpvt = ( int * ) malloc ( n * sizeof ( int ) );
  qraux = ( double * ) malloc ( n * sizeof ( double ) );
  r = ( double * ) malloc ( m * sizeof ( double ) );
  itask = 1;

  ind = dqrls ( a_qr, lda, m, n, tol, &kr, b, x, r, jpvt, qraux, itask );

  free ( a_qr );
  free ( jpvt );
  free ( qraux ); 
  free ( r );

This little chunk of code was a little seam of knowledge that just keeps going and going like the Energizer Malloc.

Roxy-3DPrintBoard, 2015-07-05:
Every one of those malloc()'s should check for NULL.   At least the AVR free() routine doesn't try to give back a NULL pointer:

void free(void *p)
{
	struct __freelist *fp1, *fp2, *fpnew;
	char *cp1, *cp2, *cpnew;

	/* ISO C says free(NULL) must be a no-op */
	if (p == 0)
		return;

scotty1024, 2015-07-05:
They're all on the stack in Marlin_main and the new qr_resolv I pushed creates *none* of the big ones inside its operational path ways. It creates a couple int[3] and a double[3].

After reading the document I believe the engineering approach to this would seem to involve adding a high water detector to the timer IRQ handlers to help determine how much slack we need to tune into __malloc_margin.

As well as inspect every single call to malloc to ensure its results are being null checked.

scotty1024, 2015-07-05:
I'd like to think the AVR could have stumbled its way down past null'd malloc()'s to those free()'s but some how I think the wheels would have come off before getting there. 😛

Roxy-3DPrintBoard, 2015-07-05:
> They're all on the stack in Marlin_main and the new qr_resolv I pushed creates *none* of the big ones inside its operational path ways. It creates a couple int[3] and a double[3].

Are you using variable dimensioned arrays?   Because then people with smaller setups can use the exact same code and only be limited by the fact they can't set high counts for the P parameter.   If the arrays are not dimensioned using the P x parameter, I would like to encourage you to try that.  It will probably be trivial to change the code to do that.

scotty1024, 2015-07-05:
Huh folks?

I think those were the only malloc's in the Marlin code! My dev branch, which has my qr_solve in it , turns up zippity do nothing when grep'ing the source files for malloc.

scotty1024, 2015-07-05:
They are indeed, and I had already added a new commit to my qr_resolve pull request that caps P.

Roxy-3DPrintBoard, 2015-07-05:
> It creates a couple int[3] and a double[3].

My thinking is as long as you are warming over qr_solve() to shrink it, wouldn't it make sense to have it output its results to some arrays defined outside of malloc() ?    qr_solve() is only run once and certainly not concurrently.  Why not just go all the way and put the results in a known place?

> I think those were the only malloc's in the Marlin code! My dev branch, which has my qr_solve in it , turns up zippity do nothing when grep'ing the source files for malloc.

From that other thread where I was diagnosed as retarded, I posted this:

>Incidentally, malloc() is used 8 times in three different files. It is used in new.cpp, qr_solve.cpp and WString.h. If somebody feels strongly on this topic (which I do not), it should not be very difficult to eliminate the usage of malloc() and free().

But those are from a search of the BeckDac fork that I'm using.  It is possible they have been pulled in the main branch.

scotty1024, 2015-07-05:
It does. Check out the 

double plane_equation_coefficients[3];

Just above the call to qr_solve() in Marlin_main.cpp after pulling in my pull request.

The two big arrays, technically known as A and B to qr_resolve() 😀, are heavily re-folded spindled, pivoted and just plain zero'd out in places by the time the call is complete.

I was planning how to have the compiler pre-fill and pre-factor A until Richard tasered me and introduced me to G29 P. 😜

Every time I compile it with my code not checked out, then with my code checked out, I'm getting about 4KB of FLASH squeezed out of the Marlin image.

Roxy-3DPrintBoard, 2015-07-05:
Do you want to play with the Iterative Least Squares Fit code?  It saves 10KB of flash space.   It pretty much drops in place and replaces the qr_solve.cpp file.   It might not hurt to add the Configuration.h #define option to pick which solution the user wants and then put #ifdef's around the code to pick which gets compiled it. You are already doing pull requests for your other qr_solve() changes.   This would get everything bundled up and working together.

Wackerbarth, 2015-07-05:
Yes, those were the only ones. But getting rid of them doesn’t “solve” the problem of running out of available RAM.
If you allocate them on the stack, then you run the same risk of having the stack collide with the other data (or heap) that uses the RAM.

> On Jul 5, 2015, at 5:21 PM, scotty1024 <notifications@github.com> wrote:
> 
> Huh folks?
> 
> I think those were the only malloc's in the Marlin code! My dev branch, which has my qr_solve in it , turns up zippity do nothing when grep'ing the source files for malloc.

Roxy-3DPrintBoard, 2015-07-05:
Wackerbarth, can you get the M99 merged into the Development Branch for me?

Roxy-3DPrintBoard, 2015-07-05:
> Yes, those were the only ones. But getting rid of them doesn’t “solve” the problem of running out of available RAM.  If you allocate them on the stack, then you run the same risk of having the stack collide with the other data (or heap) that uses the RAM.

Yes, but the religious arguments about malloc() and free() go away!   :)

scotty1024, 2015-07-06:
Roxy, don't you have your own local git clone of Marlin? You can pull gists into it yourself if you do.

Roxy-3DPrintBoard, 2015-07-06:
> Roxy, don't you have your own local git clone of Marlin? You can pull gists into it yourself if you do.

Yes, but everything I have is sick.  The forks are sick.  And my GitHub clients are all goofed up.  Right now I need people to merge stuff for me.   :(

Wackerbarth, 2015-07-06:
@Roxy-3DPrintBoard -- Not until I get the source file. I cannot log in to that 3dPrinter forum to download the file.   It's not very large. Just email it to me. (See my Github account for an address)

Roxy-3DPrintBoard, 2015-07-06:
OK.  It will be on its way shortly.

Wackerbarth, 2015-07-06:
@Roxy-3DPrintBoard  --

OK, I created a branch on Wackerbarth/Marlin:MemoryWatcher

If you and @scotty1024 will take a look, assuming that everything is OK, we can create a PR for it.

Roxy-3DPrintBoard, 2015-07-06:
> If you and @scotty1024 will take a look, assuming that everything is OK, we can create a PR for it.

I have it downloaded.   But I can't look at it until the morning.  First thing, I'll build it, load it, and make sure everything behaves as expected.

Thanks for your help!

Roxy-3DPrintBoard, 2015-07-06:
@Wackerbarth : You have mail.

closed 2015-09-25 by AnHardt