MarlinFirmware/Marlin #2383: PID Antiwindup protection

zeryabmoussaoui, 2015-07-04:
There is no antiwindup protection in the  PID controler to prevent from overshoot happening in a large change of setpoint.

One solution could be to set pid_reset[e] to true , if the error is seen null during a certain time.

thinkyhead, 2015-07-07:
@AnHardt This sounds like your cup of tea…

zeryabmoussaoui, 2015-07-07:
References :

http://www.isep.pw.edu.pl/ZakladNapedu/lab-ane/anti-windup.pdf 
http://www.controlguru.com/2008/021008.html
http://www.calerga.com/contrib/1/PIDWindup.pdf

AnHardt, 2015-07-08:
As far i understand the problem, and i read the above 3 links, the code for 'limiting the i-therm by back-integration when the output is saturated, to prevent windup' is in Marlin since the end of 2014 (#1249)

Made some tests today. With this code the overshoot is much less than without. So i think it does it's job. We already have a working `windup protection`. 

@zeryabmoussaoui 
If you have further improvements in this area of the code, please explain them, or even better - make a PR.

zeryabmoussaoui, 2015-07-08:
@AnHardt 
Thanks for your consideration.
Actually , i found out that i'm based on a old release of Marlin , i patched with the #1249 , it worked fine.
It's time to do some merge.

closed 2015-07-08 by AnHardt

wangkun321, 2021-09-25:
> References :
> 
> http://www.isep.pw.edu.pl/ZakladNapedu/lab-ane/anti-windup.pdf
> http://www.controlguru.com/2008/021008.html
> http://www.calerga.com/contrib/1/PIDWindup.pdf

hi, Could you send the link pdfs to me? The link is Invalidation。 Thank you~