MarlinFirmware/Marlin #2384: Acceleration not contiguous during arc

garrinfelber, 2015-07-05:
When I have acceleration enabled (low values, like 200mm/s²), acceleration seems to be ignored for arcs. This is OK except when going from a small linear segment to a big arc, there is a big jerk/lurch in the motion.

I have been able to get around this by adjusting the XYjerk value down, but it incurs an overall speed reduction. I usually use XYjerk=20 for parts with no or few arcs, and 5 or 10 when I notice the jerky motion.

thinkyhead, 2015-07-07:
This is because arcs are implemented as several small straight line moves, so the acceleration curve stuff probably isn't being interpolated over the whole arc, but rather just along individual axes, and only for the course of each straight move. At the moment there's no facility within Marlin to perform arcs as a special planner move. The Marlin planner only knows about straight lines.

thinkyhead, 2016-04-25:
We just put out `1.1.0-RC6` so if you have a chance to test it out, I'd like to know whether it works any better. We have an acceleration fix (a very minor change) that is on hold. I'd like to test the fix using `G2` / `G3` arcs and see if it has any effect.

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-04 by jbrazio

garrinfelber, 2016-09-24:
Yes! So sorry for the long delay, my printer was down. The arc movement is
now normal, it doesn't go crazy anymore on big arcs. Thanks!