MarlinFirmware/Marlin #2385: PT100 type 404 (from pp3dp Up printer) how to connect

tom1687, 2015-07-05:
Hello,

Im busy with mine 3d printer (pp3dp up) to change the board to a rambo 1.3.

now i want the printhead to be original but i have a problem with the PT100 type 404 (https://en.wikipedia.org/wiki/Resistance_thermometer) how can i connect it to the marlin? i tryed all the codes but marlin keeps saying temperature to high and goes in stop modes

i have the temperature table in the link only i don't know how i can put that in marlin

thanks tom

jgrjgr, 2015-07-06:
you have the generate a inverted temperature table, or better yet just replace the thermistor they used

tom1687, 2015-07-06:
How do I create a inverted temp table? 
The thing is that I want to be original so with the pt100 and not with a thermistor

AnHardt, 2015-07-06:
@tom1687 
At first be sure to define only the new sensor. Else the MAX- or MIN-TEMPERROR may come from an other sensor. On the serial console you get the number of the sensor with the error.
Than be sure you have the right pin. Setting a jumper, or not,  should change the error from MAX to MIN, or MIN to MAX.
If your sensor is a naked PT100, with no additional board or resistors, table 147 should be very close to what you want.
If the temperatures are still far away from your expectations, check if you really have a working PT100. Measure the resistance of your sensor. (0° ~ 100 Ohm,  25° ~ 110 Ohm)

Wurstnase, 2015-07-07:
Did you connect the PT100 directly on the Rambo? If yes, this will not work. You need some extra electronics to measure that low resistance.

tom1687, 2015-07-07:
Thanks wurstnase, 

I tryed directly and with a exthermocoupleboard but that is also not working. 
Can you tell me with board I need? Or can I make something myself?

Wurstnase, 2015-07-07:
It's not so easy. Just start with Google "pt100 circuit". I have no idea how to start. I don't spend much time with this.

closed 2015-09-25 by AnHardt