MarlinFirmware/Marlin #2386: Do I need special permission to submit a PR?

emartinez167, 2015-07-05:
I have the required changes to add the RigidBot settings to the firmware, but not sure how to proceed, as I tried to issue a PR from a graphical interface (SourceTree) in OSX, but got an error.

Wackerbarth, 2015-07-05:
What are the nature of the changes? If it is nothing more than customization of the Configuration files, if you don't have a fork of the repository, posting that file(s) in a public place and providing a reference should be adequate.

If it involves any changes to the executable code itself, you should make a github fork of Marlin and create a feature branch based on our Development branch. Push your changes to your github repository and then use the github web interface to submit a Pull request.

emartinez167, 2015-07-05:
It includes a new pins_RIGIDBOT.h file, additional lines to pins.h, boards.h and some required code additions in other files because of some glitches in the Rigidbot board design. I guess I will have to create a local branch, and follow your indications.

Regards,

Ernesto

Wackerbarth, 2015-07-05:
Please do.  That will permit us to review, and subsequently include, your proposed changes without danger of missing some aspect of the submission.  After the issue is closed, you should delete the branch from your repository and utilize our incorporated version in its place.

emartinez167, 2015-07-05:
I branched out, added ny changes, created a PR, and got this error:

remote: Permission to MarlinFirmware/Marlin.git denied to emartinez167.

-- 

"This e-mail and any attachments thereto, are intended only for use by the
addressee(s) named herein and may contain confidential or privileged
information. If you are not the intended recipient of this e-mail, you are
hereby notified that any dissemination, distribution or copying which
amounts to misappropriation of this e-mail and any attachments thereto, is
strictly prohibited. If you have received this e-mail in error, please
immediately notify me and permanently delete the original and any copy of
any e-mail and any printout thereof."
P Please consider the environment before printing this email

Wackerbarth, 2015-07-05:
You cannot push to OUR repository.  You need to create a fork of the repository in YOUR OWN github account and push it there. Then submit your pull request referencing that copy of the repository.

closed 2015-09-25 by AnHardt