MarlinFirmware/Marlin #2388: Thermal runaway bug?

TickTock-, 2015-07-06:
In temperature.cpp, line 1048, it appears thermal runaway only triggers when the temperature falls below target-hysteresis.  If temperature is above this, the code keeps updating *timer.  I would think we would want it to trigger when temperature exceeds target, too.

I started looking into this because I kept getting false triggers when I changed the target (plan to add a revert to state1 anytime the target changes).  If indeed this is a bug as I think, I would like to also make it revert to state1 each time the target temperature changes.

TickTock-, 2015-07-06:
I was thinking something like this:

#ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
void thermal_runaway_protection(int *state, unsigned long *timer, float temperature, float target_temperature, int heater_id, int period_seconds, int hysteresis_degc)
{
/*
      SERIAL_ECHO_START;
      SERIAL_ECHO("Thermal Thermal Runaway Running. Heater ID:");
      SERIAL_ECHO(heater_id);
      SERIAL_ECHO(" ;  State:");
      SERIAL_ECHO(*state);
      SERIAL_ECHO(" ;  Timer:");
      SERIAL_ECHO(*timer);
      SERIAL_ECHO(" ;  Temperature:");
      SERIAL_ECHO(temperature);
      SERIAL_ECHO(" ;  Target Temp:");
      SERIAL_ECHO(target_temperature);
      SERIAL_ECHOLN("");    
*/
  static float lastTarget;
  if (target_temperature > lastTarget){
      *state = 1;
      lastTarget = target_temperature;
  }
  else if (target_temperature < lastTarget){
      *state = 2;
      lastTarget = target_temperature;
  }
  
  if ((target_temperature == 0) || thermal_runaway)
  {
    *state = 0;
    *timer = 0;
    return;
  }
  switch (*state)
  {
    case 0: // "Heater Inactive" state
      if (target_temperature > 0) *state = 1;
      break;
    case 1: // Initial positive acquisition state
      if (temperature >= target_temperature) *state = 3;
      break;
    case 2: // Initial negative acquisition state
          if (temperature <= target_temperature) *state = 3;
      break;
    case 3: // "Temperature Stable" state
      if ((temperature >= (target_temperature - hysteresis_degc)) && (temperature <= (target_temperature + hysteresis_degc)))
      {
        *timer = millis();
      } 
      else if ( (millis() - *timer) > period_seconds*1000)
      {
        SERIAL_ERROR_START;
        SERIAL_ERRORLNPGM("Thermal Runaway, system stopped! Heater_ID: ");
        SERIAL_ERRORLN((int)heater_id);
        LCD_ALERTMESSAGEPGM("THERMAL RUNAWAY");
        thermal_runaway = true;
        while(1)
        {
          disable_heater();
          disable_x();
          disable_y();
          disable_z();
          disable_e0();
          disable_e1();
          disable_e2();
          manage_heater();
          lcd_update();
        }
      }
      break;
  }
}
#endif

scotty1024, 2015-07-06:
Absolutely correct. 

But think about this failure mode for a second. The hardware has failed in such a way that yanking the pin up or down no longer turns off the flow of power. Since the pin is ineffective having the code yank the pin on/off has no effect ergo what are you going to do about it?

If it were me I'd think about running around yelling fire right? So beep the buzzer if equipped? If the option is set saying there is a pin that can shutdown the power supply? Maybe you wait a bit more, hopefully able to beep, then yank that pin to shutdown the heater block that way.

Have fun!

AnHardt, 2015-07-06:
We have changed a lot in this area since `Release`.
Please recheck with `Development`-code.

TickTock-, 2015-07-06:
Good points.  I guess this is really a feature request and the original is working as intended.  I'll take a look at the new code.

closed 2015-07-06 by TickTock-