MarlinFirmware/Marlin #2392: Why are steppers always active?

ruggb, 2015-07-06:
ver 1.0.2-1 does not activate steppers until there is a movement command and deactivates them after about 60sec of inactivity.
ver 1.0.3 dev does not activate steppers until there is a movement command but NEVER seems to deactivate them unless u reset the bd, do an emergency stop, or disconnect Repetier-Host.

WHY NOT??

AnHardt, 2015-07-06:
You have to allow the deactivation now.

>#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false

Ziggy2013, 2015-07-06:
If the steppers are disabled during microstepping (ie power turned of) they will move to their closest detent position if there is not enough friction to hold them in place. Disabled steppers can cause inaccurate X, Y or Z microstep positioning.

ruggb, 2015-07-06:
Thk you for the explanation.
Where is the idle time set. Obviously, if the printer is not printing there is no need to hold any position, right?

ruggb, 2015-07-06:
Can the Z axis idle time be set separately?

thinkyhead, 2015-07-07:
There are two places in `manage_inactivity` where axes could be disabled.

- At the top of `manage_inactivity` it waits for `stepper_inactive_time` (as set by `M18` / `M84`) and then disables any steppers flagged by `DISABLE_[XYZ]` boolean setting.

- At the end of `manage_inactivity` it calls `check_axes_activity` - part of `planner.cpp`. This function seems to do the following:
 - If any `blocks` are queued and those blocks include any `steps` then axes with steps are flagged "active."
 - Any axis that isn't "active" is disabled immediately (?) if its `DISABLE_[XYZ]` flag is `true`.

Currently the idle time for all axes is a single setting. `M84 S2` will set the idle time to 2 seconds. Marlin has extended `M84` with `XYZE` flags, but they are just indicators that the axis should be disabled, and they don't take parameters. I'm not sure if any host or slicer software uses that Marlin-specific extension, so perhaps the `XYZ` parameters could be changed to allow setting separate timeouts for each axis.

Ziggy2013, 2015-07-07:
Disabling X, Y, Z or E steppers during printing is bad practice and will cause position errors..  Steppers only should be disabled when the printer is idle - so there doesn't seem much point in having separate timeouts for each stepper.

ruggb, 2015-07-07:
Since there r separate flags for each stepper my thought was that the disable was keyed by that stepper being idle not all steppers being idle. Other text is written implying it is also based on individual stepper activity. But if it is all steppers idle, separate timeouts doesn't matter. I was thinking a print could be on one layer longer than 60 sec causing Z to time out.

It begs the question, why r there separate flags? I suppose that one might not want to disable Z, for instance, if a platform is loose enough to fall on its own weight.

So this >> #define DEFAULT_STEPPER_DEACTIVE_TIME 60 << is in seconds????, I assume, and setting disables to true will disable those steppers 60 sec after a print stops, IE, there r no more commands in the buffer. And M84 overrides the 60 setting.

Is that correct?

So is it a bug in 1.0.2-1 that the steppers disable even though the disables are all FALSE?

maukcc, 2015-07-08:
we are using 1.0.3 and "#define DISABLE_Z true" does not work.
So the stepper is only released when a M84 command is given.
This used to work in 1.0.2 (unless layer time was below "x" sec.)
Can this be because of auto/manual bed levelling.

AnHardt, 2015-07-08:
@maukcc 
>Can this be because of auto/manual bed levelling.

Yes. Bed leveling works because the z-steppers adjust the height all the time. They are always in motion. So the deactivation delay is constantly reset.

AnHardt, 2015-07-08:
@ruggb 
Additionally there is check_axes_activity() in the planner.cpp.

ruggb, 2015-07-08:
I am trying to stitch all the above comments together and make some sense out of them but I am not having any luck. Some of them appear to conflict with others and themselves.

Apparently, #define DISABLE_X, et al, does not work in 1.0.2 because I never chgd them to true and it always timed out in 60 sec. But it does not time out in 1.0.3 either way, so there is a bug in both versions

In @thinkyhead's comment, the first bullet seems to say that axis activity is determined by ANY axis being active. If none are active then any axis with disable true is turned off.
the second bullet seems to say that is it a per axis activity and if there is no activity scheduled on an axis it is disabled.
The bullets can't both be true.

@maukee's comment says there is also a bug in 1.0.3.

@AdHardt's comment is vague because it isn't clear whether that "deactivation delay" is one for all or for Z only.

Is it possible to get a clear definition of how it works so it can be tested to verify if it needs fixing???

maukcc, 2015-07-08:
ver 1.0.2-1 does not activate steppers until there is a movement command and deactivates them after about 60sec of inactivity. 
Or give M84 command
Or they are disabled immediately after each movement of the appropriate axis if the "#define DISABLE_Z" (or X or Y) is set to true.
And they will be enabled about 2 sec. before they start moving because of look ahead.
I have never tested this on X, Y and E because of before mentioned reason of position error.

ver 1.0.3 dev does not activate steppers until there is a movement command but NEVER seems to deactivate them unless u reset the bd, do an emergency stop, or disconnect Repetier-Host. 
Or give M84 command.
This bug or feature in "#define DISABLE_Z true" is probably because of bedlevelling but is not gone when bedlevelling is turned off.

ruggb, 2015-07-08:
This I see:
ver 1.0.2-1 does not activate steppers until there is a movement command and deactivates them after about 60sec of inactivity.
Or give M84 command

Never looked for this:
Or they are disabled immediately after each movement of the appropriate axis if the "#define DISABLE_Z" (or X or Y) is set to true.

This I also see:
ver 1.0.3 dev does not activate steppers until there is a movement command but NEVER seems to deactivate them unless u reset the bd, do an emergency stop, or disconnect Repetier-Host.
Or give M84 command.

Only MS would call this a "feature", actually they would call it "design intent"
This bug or feature in "#define DISABLE_Z true" is probably because of bedlevelling but is not gone when bedlevelling is turned off.

I vote for bug!

IMHO, it should not deactivate anything if there is movement on any axis. Once all movement has stopped for the specified period, then disable axis flagged as true. Or more simply, if it is printing all are active. when it finishes it deactivates.

thinkyhead, 2015-07-09:
If someone feels like producing a table showing the variations in behavior with these flags individually enabled/disabled, both with and without bed leveling, that would give some sense of where axes are failing to be disabled when they should be. The code behavior I described is from a simple read-through of the code, so I might be missing some nuances according to my anomalous description of `check_axes_activity`. Obviously axes aren't _always_ being disabled whenever there are no blocks in the planner buffer.

c-born, 2015-07-27:
I just came across this bug. Steppers were hot even when the machine hadn't been running for hours, they were never disabled. After reading this issue I set the DISABLE_[XYZ] flags to true, but now the steppers were disabled immediately from `check_axes_activity`.
My quick-fix was to comment out the call to  `check_axes_activity` at the bottom of `manage_inactivity` until this is sorted out.

In my case I'm happy for the steppers to be disabled a minute or so after the printer stops, so they aren't running hot and chewing up power unnecessarily.
Does anyone know of any reason you would want to disable the steppers immediately, from within `check_axes_activity`? Any particular hardware configuration that may require this?
If not, why don't we just remove all the stepper-disabling code from that function?

thinkyhead, 2015-07-27:
> anyone know of any reason you would want to disable the steppers immediately?

That part of `check_axes_activity` has been confusing me also. We could dig through the history and see when it was added, and perhaps why. That might make it clearer, or not. Anyway, if the behavior seems altogether wrong (as it does to me) then we might just go ahead and remove it. It seems like the code to check inactivity at the top of `manage_inactivity` already does what is needed.

AnHardt, 2015-10-05:
Fix is near. See #2661

closed 2015-10-05 by AnHardt