MarlinFirmware/Marlin #2393: Lowercase G-Code support?

thinkyhead, 2015-07-07:
A small number of G-Code commands will accept lowercase letters as G-Code parameters. Should we make it so that all G-Code can be lowercase, or can we simply remove these instances where lowercase letters are checked for?

grob6000, 2015-07-07:
Some thoughts from a curious user / engineer...

As I understand it:

(1) The original g-code standard (though we don't adhere to it strictly for many reasons already) is pretty clear about everything needing to be upper case.
(2) Hosts are reliable in sending only uppercase commands
(3) The uppercase requirement is well understood by users at this point.

It follows from (2) that case-insensitivity (i.e. p <--> P) would have no impact on host-compatibility, and while technically may fulfil the robustness principle (https://en.wikipedia.org/wiki/Robustness_principle) as far as Marlin is concerned, in the long run I think it could encourage users (and their resulting published material on the web) to be lazy with their case, which would then not be compatible with other stricter firmware.

Case-sensitivity is thus good, but we should continue to only accept uppercase letters, as lowercase is not necessary. If a need is identified later, at least they're available for use as unique parameters.

Thus, I would support your latter option of removing all checks for lower-case letters from g-codes. Unless someone can find a counterexample to refute (2) above.

Roxy-3DPrintBoard, 2015-07-07:
> (2) Hosts are reliable in sending only uppercase commands
> (3) The uppercase requirement is well understood by users at this point.

Bully for the hosts.   But from day one I patched my Marlin_main.cpp with:

  if(code_seen('G') || code_seen('g') )
  {
    switch((int)code_value()

Just because it is less work to type commands and send them to my printer.   How about we treat upper and lower case as equivalent?

thinkyhead, 2015-07-07:
If we're going to support both, then I think we should just patch `code_seen` and drop the double calls to it.

thinkyhead, 2015-07-07:
> Case-sensitivity is thus good, but we should continue to only accept uppercase letters, as lowercase is not necessary.

I agree, and the point of the standard was specifically to allow modest embedded processors to be able to parse it without all the added overhead of checking for every parameter twice. The two PRs referenced above this comment are (1) a definite merge to get rid of the current set of "special cases", and (2) the code that would be required for lowercase GCode support.

closed 2015-10-12 by thinkyhead

AnHardt, 2015-11-16:
@thinkyhead 
The current brittle construct
```
float code_value() {
  float ret;
  char* e = strchr(seen_pointer, 'E');
  if (e) {
    *e = 0;
    ret = strtod(seen_pointer + 1, NULL);
    *e = 'E';
  }
  else
    ret = strtod(seen_pointer + 1, NULL);
  return ret;
}
```
is used to prevent reading a 'float with exponent' when the following parameter is for the Extruder and is not separated by a space.
This fails miserably when the float is written with an 'E' instead of an 'e' - the exponent then is omitted - the read number potentially some magnitudes to large or small. 
Writing a float with 'E' is perfectly allowed for strtod().
I could not find any reference about, if 'floats with exponent' are allowed in g-code or not.

Wackerbarth, 2015-11-16:
Originally, there was no concept of "E" notation for floating values.
Allowing them makes the parser more complex.
Why is there a NEED to allow the notation? Can't you express any needed values without using the notation?

AnHardt, 2015-11-16:
No, i don't see the need for the floating point notation with exponent. I just saw a potential misinterpretation. If we would think about allowing lowercase g-codes/parameters this is a further place for changes, not handled in this (closed) PR. It just cached my eyes while looking after the space-separator problem.
If we say: "Floats with exponents are not allowed." everything is OK. If we can parse them correctly with a lowercase 'e', then that's a hidden feature.

Wackerbarth, 2015-11-16:
I suspect that "hidden features" are not a good idea.  When you have one, somebody is going to start using it and then, if the behavior changes, they are going to complain because their "feature" no longer works.

I think that it is far better to either have a feature, well documented, or not have it at all.

Roxy-3DPrintBoard, 2015-11-16:
I don't have strong feelings on this one, but given the fact we don't have
a need for exponents, it seems like it is asking for trouble.   We just had
capital N's show up as a problem with M117.

@AnHardt There is another place where spaces cause problems.   Can you add
this patch to the code_value() patch up above?

~~~~
bool code_has_value() {
  int i = 1;
  char c = seen_pointer[i];

  while (c == ' ')        // skip past any white space between GCode and
number
     c = seen_pointer[++i];

  if (c == '-' || c == '+') c = seen_pointer[++i];
  if (c == '.') c = seen_pointer[++i];
  return (c >= '0' && c <= '9');
}

~~~~

AnHardt, 2015-11-16:
@Wackerbarth 
Do you have a 'nice' alternative to also checking for 'e' and parsing upper and lower case wrong, adding a new variable to remember if we replaced 'e' or 'E'. Throwing an error is no option because it could be the __E__ xtruder parameter.