MarlinFirmware/Marlin #2397: change homing movement order

macsimski, 2015-07-08:
doing some mill type work with a sharpie, i noticed that the marlin 1.0.0 firmwar hase some issues with the order of things in homing.
currently homing behaves like:
move x 10
home x
move y 10
home y
move z10
home z

this could be inproved by nesting the movements, saving some hotends and beds

move Z 10 (lift hotend/drill off the bed before x/y movement)
move X 10
Move Y 10
home Y
home X
Home Z

thinkyhead, 2015-07-09:
I have been thinking of something similar because almost no one wants to home XY while Z is touching the bed. However, instead of moving all axes away from the endstops before homing —which will lead to grinding in many cases— we should first home Z, then move Z away from the bed (if it's a min endstop), and then home XY last. In general a change in behavior like this has to be done as a new feature, disabled by default, so users see consistent behavior on the next update.

macsimski, 2015-07-09:
as a stopgap i use the following gcode
G0 Z5
G28XY
G28Z
this seems to work. what do you mean by grinding? I do see this as a bug, so why disable this behaviour?

Wurstnase, 2015-07-09:
What is when you bed is on the maximum position? Your printer don't know the position until it's homed. In that case you can't move down.

thinkyhead, 2015-07-09:
> what do you mean by grinding?

If you have your axis at the right end (for example) and try to move right ("move X 10") you will get a loud grinding sound as the motor torque fails to move the axis and the motor slips out of its magnetic hold.

macsimski, 2015-07-10:
Wurstnase: if the bed is not on the home position (min stop not engaged) there should be no movement away from the min position. if you rely on a probe as a min sensor, maybe a max sensor is handy. I agree that with all types of delta printers around, things get complicated, and i can see a option that looks for a delta flag somewhere to choose the order of things as the order used now ruins a lot of beds and nozzles as well on xyz printers without a heated bed

c-born, 2015-07-28:
@thinkyhead With my setup in the XY home position the print head is located off to the side of the bed, and in the Z endstop trigger height the head is below the level of the bed surface. This means you have to home X and Y before Z, or you will drive the head into the print bed.
I think this is a fairly common configuration.

On a related matter, there seems to be a problem with the current build sometimes disregarding the Z minimum endstop on a home operation. It works ok when homing each axis individually, but several times now when given a home command it has continued driving Z homewards, right through the opto-endstop with enough force to break it in half.
I have a z-probe in series with an opto Z-min endstop as a backup, and in testing it appears the Z-probe is not being deployed as it should be, and the "emergency" Z-min is also being ignored.

thinkyhead, 2015-07-29:
@c-born It appears that the current conditions surrounding the call `deploy_z_probe` in `homeaxis` are not correct, so it's just not getting called. The number of probe- and endstop-related settings is starting to become a bit messy. I hope to sort out the correct conditions asap. Can you tell me what you have set for  `DEACTIVATE_SERVOS_AFTER_MOVE` and `SERVO_DEACTIVATION_DELAY`?

c-born, 2015-07-29:
@thinkyhead 
`#define DEACTIVATE_SERVOS_AFTER_MOVE`
`#define SERVO_DEACTIVATION_DELAY 310`

thinkyhead, 2015-07-29:
Thanks! And I assume your `Z_HOME_DIR` is `-1`… Hmm, so far the current Development code looks like it should at least try to call `deploy_z_probe`. But perhaps that is not the case. I have updated my test branch, if you would like to see some extra logging of the issue: https://github.com/thinkyhead/Marlin/tree/debug_G29

thinkyhead, 2015-10-12:
The latest RC (RC2) addresses some of this. If the main issue remains, please comment.

c-born, 2015-10-12:
I updated on Saturday and have run a few test pieces, so far without any problems, thanks

closed 2016-02-22 by thinkyhead