MarlinFirmware/Marlin #2399: Print pauses periodically for a couple seconds

ruggb, 2015-07-08:
I have been using 1.0.2 without issue for a while. I installed 1.0.3 to test QH and started a print. I noticed that periodically (did not measure period) the print would stop for about 2-3 seconds then restart. Not too good for quality, so I aborted it and reloaded 1.0.2- It is again printing without issue.

So why is 1.0.3 pausing? All my mods in both versions r the same (I think) except of course for QH and those required to go to 1.0.3

I am running Repetier-Host 1.5.0 on one computer and uploaded 1.0.2 via another computer - so nothing changed on the printing computer.

thinkyhead, 2015-07-09:
@ruggb Hard to know. If you can do any sleuthing, please help to narrow down the cause. Perhaps you can find some pattern, such as always happening with certain parts of a test model…

ruggb, 2015-07-09:
I didn't even shut down the printing computer, so nothing changed there.
I switched the cable and loaded 1.0.2 and switched the cable back and it is printing the same file without any pauses.

It appears like the printer is accepting a buffer full then waiting till the buffer is empty to let it fill up again and waiting till it is full to continue printing. Both ver r set to 250000 and nothing changed on the printing computer.

When it stopped it was not at the same location in the print.
OK, I'll go back to 1.0.3 and see if it does it on another print - or something.
Just thought u might clue me in on what to look for.

thinkyhead, 2015-10-12:
Are we seeing this with the latest RC (RC2)?

ruggb, 2015-10-12:
I have not seen this in RC2. So I guess things r good.
thx

closed 2015-10-12 by ruggb