MarlinFirmware/Marlin #2400: Negative X value reported after G29

foulowl, 2015-07-08:
Hi folks,

After running the following:
G28 X0 Y0
G29

then doing M114, my printer reports X has a negative value, usually between -30 and -40.

Terminal output looks like:
Recv: echo:endstops hit:  X:-33.90 Z:5.68

I have also tried doing:
G28 X0 Y0
M114
and it correctly reports x=0 and y=0, so the bug must lie within the code for G29.

This has resulted in printer crashes (!!) since it thinks the origin is 30-40 units away from the endstop, my xmax is effectively 30-40 units greater! 

Thanks!

AnHardt, 2015-07-08:
@foulowl 
Until now we only know, that you have a servo z-probe and you are not homing z.
Maybe you have noise on your x-endstop.
Please tell us more about your machine and your configuration.

foulowl, 2015-07-08:
I realized the issue here. I had my bed probing test points set to 0, 0 for the left and front edges. I was under the impression that these coords were the coords of the hotend, not the coords of the probe.

The fix was to add the X and Y offsets to the probing test points.

Apologies for the false alarm.

closed 2015-07-08 by foulowl