MarlinFirmware/Marlin #2404: feedmultiply can not change from LCD panel with Delta configuration.

GabeHC, 2015-07-09:
Board BOARD_RAMPS_13_EFF
Version 1.0.2 Delta configuration
LCD REPRAP_DISCOUNT_SMART_CONTROLLER
Symptom:  Feedrate stay fix at 100%. can not be change by encoder, M220 working, only happened with Delta configuration.
still not finding anything wrong in ultalcd.cpp code yet. am I looking at the right place?

thinkyhead, 2015-07-09:
That's probably fixed in the current Development HEAD. Are you able to try that at this time?

GabeHC, 2015-07-09:
Confirm fixed V1.0.3
Like to know how to fix it on V1.0.2

thinkyhead, 2015-07-09:
Do you have the line `#define ULTIPANEL_FEEDMULTIPLY` in your `Configuration_adv.h` file? Evidently it was left out of one or more of the example configurations, including the delta config.

GabeHC, 2015-07-10:
Thanks! It works now on V1.0.2

closed 2015-07-10 by AnHardt