MarlinFirmware/Marlin #2407: Improving idle()

Wurstnase, 2015-07-09:
Hey all,

just start to think loud:

```
inline
void ms20_idle() {
	lcd_buttons_update();
}

inline
void ms100_idle() {
	manage_heater();
	manage_inactivity();
}

inline
void ms1000_idle() {
	lcd_update();
}

millis_t idle_millis = 0;
uint8_t counter_ms100 = 0;
uint8_t counter_ms1000 = 0;

void idle() {
  millis_t new_millis = millis();
  if (new_millis >= idle_millis + 20) {
	ms20_idle();

    counter_ms100++;
	if (counter_ms100 >= 5) {
	  ms100_idle();
	  counter_ms100 = 0;

	  counter_ms1000++;
	  if (counter_ms1000 >= 10) {
		ms1000_idle();
		counter_ms1000 = 0;
	  }

	idle_millis = new_millis;
	}
  }

  some_fast_update_things();
}
```

thinkyhead, 2015-07-09:
Here's something bot faster and more memory-saving:
```
void idle() {
  millis_t ms = millis();
  if ((ms % 20) == 0) {
    ms20_idle();

    if ((ms % 100) == 0) {
      ms100_idle();

      if ((ms % 1000) == 0) ms1000_idle();
    }
  }

  some_fast_update_things();
}
```

epatel, 2015-07-09:
@thinkyhead That doesn't look safe as if `some_fast_update_things()` would take longer than 1 ms there would be a risk that one could miss some steps, which the @Wurstnase code actually tries to handle.

thinkyhead, 2015-07-10:
@epatel Truly! It can still reduce the number of variables required, and they can be static locals if they're not used anywhere else. Plus, it's nicer to use a "next_ms" value rather than a "last_ms" type value. So…
```
millis_t next_idle_ms;

void setup() {
  next_idle_ms = millis() + 20;
}

void idle() {
  static uint8_t counter_ms = 0;
  millis_t ms = millis();
  if (ms >= next_idle_ms) {
    next_idle_ms = ms + 20;
    counter_ms++;
    ms20_idle();
    if (!(counter_ms % 5)) {
      ms100_idle();
      if (!(counter_ms % 10)) ms1000_idle();
    }
  }
  some_fast_update_things();
}
```

thinkyhead, 2015-07-10:
Of course `ms20_idle` is only guaranteed to execute _at least_ 20ms after the last one. So, if for some reason `ms20_idle` were only able to execute every 30ms, then `ms100_idle` would only be called every 150ms, and `ms1000_idle` would only be called every 1500ms.

Wurstnase, 2015-07-10:
The ground-idea comes from Teacup. I know, that this is not perfect and need some discussion. Also the update_button and update_lcd don't work actually this way.

https://github.com/Wurstnase/Teacup_Firmware/blob/master/clock.c

thinkyhead, 2016-04-25:
We continue to refine, and this will shake out in the short term, I am sure.

closed 2016-04-25 by thinkyhead