MarlinFirmware/Marlin #2408: Delta auto-leveling issues on development 7/9

skhameneh, 2015-07-09:
Last night I configured development for my delta hoping to fix some auto-leveling issues with prior versions of Marlin.  Everything is looking promising and I like the latest code refactors in the dev branch, but I'm having an entire new set of issues.

1) G29 won't do anything unless I enable enable Z_PROBE_ALLEN_KEY, unfortunately my z_min is attached directly to the print head (no retractable probe).  G29 results in the print head coming close to the bed and "Error:Z-Probe failed to retract!".

2) G28 works flawlessly, much better than prior versions.  Running G29 without homing results in an error as expected.  Running G29 after G28 causes all axis to reverse before moving the probe down, causing the belts to grind.  Is this a config issue?

I appreciate any help and would be more than happy to do some code edits (and testing).

Thanks,

Shawn

thinkyhead, 2015-07-09:
@skhameneh Sorry for your current `G29` troubles. It sounds like Marlin might not have the best support for non-servo and non-allen-key probes on Delta machines right now. Also, Z isn't _supposed_ to raise up in cases where Z is homed to a Z max endstop, but it might be trying to do that anyway due to `Z_RAISE_BEFORE_HOMING`, so you might try setting that to 0. Meanwhile, I have a debug branch that I will be patching up with the latest code, and as soon as I have that ready you can grab it and give it a test run. All it does differently is print lots of position information to the serial console, but from that we should be able to see where it's getting confused.

I'll let you know when my debug branch is patched up – should be pretty soon!

thinkyhead, 2015-07-10:
Ok, this can be used to get more debug output…
https://github.com/thinkyhead/Marlin/tree/debug_G29

skhameneh, 2015-07-10:
Thanks, here's a nice long debug output.  I'll need to skim through it myself. :+1: 
```
>>>G29
SENDING:G29
gcode_G29 >>>
G29 Auto Bed Leveling
deploy_z_probe > current_position: (0.00, 0.00, 286.20)
prepare_move_raw > destination: (0.00, 0.00, 286.20)
reset_bed_level
enable_endstops(true)
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
probe_pt >>>
> ProbeAction:0
> current_position: (0.00, 0.00, 286.20)
Z Raise to z_before 15.00
> do_blocking_move_to Z: (0.00, 0.00, 15.00)
do_blocking_move_to: (0.00, 0.00, 15.00)
prepare_move_raw > destination: (0.00, 0.00, 15.00)
> do_blocking_move_to XY: (25.00, -40.00, 15.00)
do_blocking_move_to: (25.00, -40.00, 15.00)
prepare_move_raw > destination: (25.00, -40.00, 15.00)
run_z_probe (DELTA) 1
prepare_move_raw > destination: (25.00, -40.00, -10.00)
run_z_probe (DELTA) 2 > current_position: (25.00, -40.00, 14.99)
<<< probe_pt
z_before = (between) 5.00
probe_pt >>>
> ProbeAction:0
> current_position: (25.00, -40.00, 14.99)
Z Raise to z_before 19.99
> do_blocking_move_to Z: (25.00, -40.00, 19.99)
do_blocking_move_to: (25.00, -40.00, 19.99)
prepare_move_raw > destination: (25.00, -40.00, 19.99)
> do_blocking_move_to XY: (12.00, -40.00, 19.99)
do_blocking_move_to: (12.00, -40.00, 19.99)
prepare_move_raw > destination: (12.00, -40.00, 19.99)
run_z_probe (DELTA) 1
prepare_move_raw > destination: (12.00, -40.00, -10.00)
run_z_probe (DELTA) 2 > current_position: (12.00, -40.00, 19.97)
<<< probe_pt
z_before = (between) 5.00
probe_pt >>>
> ProbeAction:0
> current_position: (12.00, -40.00, 19.97)
Z Raise to z_before 24.97
> do_blocking_move_to Z: (12.00, -40.00, 24.97)
do_blocking_move_to: (12.00, -40.00, 24.97)
prepare_move_raw > destination: (12.00, -40.00, 24.97)
> do_blocking_move_to XY: (-1.00, -40.00, 24.97)
do_blocking_move_to: (-1.00, -40.00, 24.97)
prepare_move_raw > destination: (-1.00, -40.00, 24.97)
run_z_probe (DELTA) 1
prepare_move_raw > destination: (-1.00, -40.00, -10.00)
run_z_probe (DELTA) 2 > current_position: (-1.00, -40.00, 24.96)
<<< probe_pt
z_before = (between) 5.00
probe_pt >>>
> ProbeAction:0
> current_position: (-1.00, -40.00, 24.96)
[... 734 lines cut ...]
> probing complete > current_position: (12.00, 51.00, 264.36)
ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
-68.31 16.47 21.46 6.50 -8.46 -23.42 -38.39 -33.40 11.49
-28.41 -13.45 31.44 26.45 21.46 16.47 11.49 36.42 51.39
96.28 36.42 41.41 46.40 51.39 56.37 61.36 66.35 16.47
111.24 106.25 101.26 96.28 91.29 86.30 81.31 76.33 71.34
106.25 111.24 116.23 121.21 126.20 131.19 136.18 141.16 146.15
191.04 186.05 181.06 176.08 171.09 166.10 161.11 156.13 151.14
255.88 191.04 196.03 201.01 206.00 210.99 215.98 220.96 176.08
201.01 215.98 245.90 240.91 235.93 230.94 225.95 265.85 280.81
235.93 290.79 275.83 250.89 255.88 260.86 255.88 250.89 315.73
stow_z_probe > current_position: (12.00, 51.00, 264.36)
prepare_move_raw > destination: (12.00, 51.00, 314.36)
prepare_move_raw > destination: (-64.00, 56.00, 23.00)
prepare_move_raw > destination: (-64.00, 56.00, 3.00)
prepare_move_raw > destination: (-64.00, 56.00, 43.00)
prepare_move_raw > destination: (0.00, 0.00, 43.00)
Error:Z-Probe failed to retract!
[ERROR] Error:Z-Probe failed to retract!

Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
[ERROR] Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)

<<< gcode_G29
```

skhameneh, 2015-11-05:
Why was this closed?  To my knowledge, it's not resolved.

AnHardt, 2015-11-05:
Just a test if you are still around. ;-)

Thanks for the log. Sadly i can't find the error.
Is your sensor connected/configured as a Z_MIN_endstop or a Z_MIN_probe?
Is your printer doing the moves to deploy the allen_key?
What is the output of M119 in mid air.

A link to your Configuration.h & Configuration_adv.h may be helpful.

AnHardt, 2015-11-05:
Maybe related to https://github.com/MarlinFirmware/Marlin/issues/2722

max-232, 2015-11-05:
a possible dirty solution? https://github.com/MarlinFirmware/Marlin/issues/2722#issuecomment-154162669

thinkyhead, 2016-03-11:
We should be at a much better place with `RCBugFix` at this time. Now is the time to test it, because we're trying to get another release candidate out to the world, and it would be great to have all known issues cleared up.

closed 2016-04-25 by thinkyhead