MarlinFirmware/Marlin #2410: Can my fork compare against Development by default?

thinkyhead, 2015-07-10:
On my Marlin fork I see the following unhelpful message when viewing each branch:

__This branch is 1716 commits ahead, 19 commits behind MarlinFirmware:Release__

It would be nice if I could have my branches automatically compare against `Development` instead. This would also be helpful as the base for PRs, which also defaults to `MarlinFirmware:Release`. Is there any way to do this apart from making `Development` the default branch for __MarlinFirmware/Marlin__?

Wackerbarth, 2015-07-11:
Perhaps we need to have more than one Marlin repository. Rather than having multiple branches in a single repository, we could have a separate repository for Releases and for Development.
If both of these repositories are cloned from the same base, it will not be difficult to move commits from one to the other.

thinkyhead, 2015-07-14:
So, something like https://github.com/MarlinFirmware/MarlinDev …?

Wackerbarth, 2015-07-15:
Yes, that is what I was thinking.

thinkyhead, 2015-07-16:
Hmm, I don't know. I can see it creating further complications because it would be a whole separate repository, project, team, issue list, etc.

Wackerbarth, 2015-07-16:
I don't see it as a separate team, etc. It would still be a part of the MarlinFirmware project. The "Release" version would be a subset of the Development version and issues would, in general, apply to either.

Additionally, we would not apply any Pull Requests to the Release version. Instead, we would merge things through the Development repository and then simply git push updates into the Release repository.

thinkyhead, 2015-07-16:
"MarlinFirmware" is the user-name on this Github account. The project is "Marlin."

pdbogen, 2015-07-18:
This is just a git thing. If MarlinFirmware is your "upstream" remote (see git remotes -vv), then you just need to "git branch --set-upstream-to upstream/Development <YOURBRANCHNAME>", e.g.:
```
pdbogen@volxv:~/marlin/Marlin/Marlin$ git remote -vv
origin	git@github.com:pdbogen/Marlin.git (fetch)
origin	git@github.com:pdbogen/Marlin.git (push)
upstream	git@github.com:MarlinFirmware/Marlin.git (fetch)
upstream	git@github.com:MarlinFirmware/Marlin.git (push)
pdbogen@volxv:~/marlin/Marlin/Marlin$ git branch -vv
* Development e78466d [upstream/Development: ahead 1] Specify print format for reporting invalid tmp_extruder
  master      7de70dc [origin/master] Config.h improved to make variations in steps/mm and thermistors easier to track and understand.
  newmarlin   a7396c2 [origin/newmarlin] Variety of small fixes; major refactoring of serial command queue management in Marlin proper
pdbogen@volxv:~/marlin/Marlin/Marlin$ git branch -u upstream/Development Development
Branch Development set up to track remote branch Development from upstream.
```

Wackerbarth, 2015-07-18:
@pdbogen -- I think that @thinkyhead was referring to the generation of pull requests on gitub.

pdbogen, 2015-07-19:
@Wackerbarth Ahh, just so. After a bit of research, I can't find another solution. It does appear, though, that github's "hub" wrapper around the git command-line tool could ease this burden slightly, as it lets you specify the "pull base" and "pull head" on the command line. It only works for those that have push access to the base repo, but that's probably OK, since those are the people that would benefit most anyway.

closed 2015-10-12 by thinkyhead