MarlinFirmware/Marlin #2412: Z position incorrect after G29

avri-schneider, 2015-07-10:
After issuing G28, moving Z to position 0 gets Z at position 0.
After issuing G29, moving Z to position 0 gets Z at Z_PROBE_OFFSET_FROM_EXTRUDER.

It is expected tat moving Z to position 0 after G29, hotend nozzle would touch the bed and not be at position Z_PROBE_OFFSET_FROM_EXTRUDER above the bed.

Is changing

`z_tmp = current_position[Z_AXIS],`
to
`z_tmp = current_position[Z_AXIS]  - Z_PROBE_OFFSET_FROM_EXTRUDER.`

the proper fix? (it resolves the issue for me).

thinkyhead, 2015-07-10:
I'm not sure if that's the exact fix, but this would be more correct:

    z_tmp = current_position[Z_AXIS]  - zprobe_zoffset

AnHardt, 2015-07-10:
Some lines below current_position[Z_AXIS] is calculated. Is there a zprobe_zoffset therm in?
Tell us more abut your machine and config.

avri-schneider, 2015-07-10:
I'm using the latest development branch. Merged in Clean qr_solve update #2382. So yes, some lines bellow, there is the following line:

`current_position[Z_AXIS] = z_tmp - real_z - zprobe_zoffset; // The difference is added to current position and sent to planner.`

Yet, symptoms are as described.

My machine is a Makerfarm Prusa I3 8" with Auto Bed Levelling mod.

The following is my Configuration.h file:
https://gist.github.com/avri-schneider/bd3a30d61f7a9c7ef0e4

AnHardt, 2015-07-10:
Could you make a G29 V4 for us? I have the the new qr_solve() from #2382 under suspicion.
Please post similar data as i did in #2382.

thinkyhead, 2015-07-11:
This line is currently under suspicion also:

    current_position[Z_AXIS] = z_tmp - real_z - zprobe_zoffset;

You can look at it as:

    current_position[Z_AXIS] = -zprobe_zoffset + (z_tmp - real_z);

The `zprobe_zoffset` addend was just recently added, around the same time as the update to `qr_solve`. I don't think it's correct, because (1) there is already a correction to XYZ in `set_bed_level_equation_lsq` and (2) it assumes the probe is contacting the bed, but the nozzle moves up before stowing the probe on many systems.

magillus, 2015-07-11:
I noticed some differences too, what would be best solution for 1.0.2 version - which line to change?

avri-schneider, 2015-07-11:
@AnHardt: https://gist.github.com/avri-schneider/8adf5212f2b1171af929 
This is with latest development branch + the following change:
 `z_tmp = current_position[Z_AXIS] - zprobe_zoffset`

avri-schneider, 2015-07-11:
@thinkyhead:
> (2) it assumes the probe is contacting the bed, but the nozzle moves up before stowing the probe on many systems.

Just verified that this is related to _Z_RAISE_AFTER_PROBING_. I have changed mine to _60_. Even with:

`z_tmp = current_position[Z_AXIS] - zprobe_zoffset,`

Hotend is way above the bed after the issuing:

* G28
* G29
* G1 Z10 F3000

Indicating it is related to _Z_RAISE_AFTER_PROBING_ rather than _zprobe_zoffset_

Since I had _Z_RAISE_AFTER_PROBING_ set to a value just a bit larger than _zprobe_zoffset_ - I assumed _zprobe_zoffset_ was the culprit.

thinkyhead, 2015-07-11:
@magillus I'm not sure how much of the new leveling code ended up in 1.0.2, but take a look at the last part of `gcode_G29` and the last part of `set_bed_level_equation_lsq` to see how they are correcting the XYZ, and try tweaking that part to see how it affects the final result.

AnHardt, 2015-07-11:
@magillus 
Thanks for the log.
Look at the Front left corner of your bed. 

    Bed X: 50.000 Y: 50.000 Z: 12.869

looks ok, but after qr_solve it looks like: -24.50394
Even the correction-matrix has suspicious high values.

So, until it works, revert the changes from #2382.

avri-schneider, 2015-07-11:
@AnHardt 

I can confirm that there's a problem with the bed leveling, although not related to this particular issue.

After https://github.com/MarlinFirmware/Marlin/issues/2412#issuecomment-120556106

`G29 V4` still shows the `qr_solve` discrepancy you mention

But. moving `Z` to `0` gets the nozzle paper distance from bed. This is repeatable after several iterations of `G29` followed by `G1 Z0 F3000`.

However, and this probably relates to the issue you refer to (incorrect calculations in `qr_solve` #2382), moving `X` and `Y` to other corners of the bed, while leaving `Z` at position `0`, shows that although `Z` is raised/lowered - the end result is a `Z` at incorrect position, in my case either lower (nozzle crash on bed) on the back right corner, or higher (nozzle few millimeters above bed) at the front left corner.

But as mentioned, I think this is a totally different, unrelated issue. I think @thinkyhead's solution fixes this particular issue.

I will revert the changes done by #2382 and see if I can reproduce this behavior of moving `Z` to different corners, resulting in different `Z` distance from bed.

avri-schneider, 2015-07-11:
Reverted #2382 and with @thinkyhead's change:

`current_position[Z_AXIS] = z_tmp - real_z - zprobe_zoffset + Z_RAISE_AFTER_PROBING; // The difference is added to current position and sent to planner.`

 `Z` distance from bed is constant at bed corners (can't reproduce behavior I mentioned in my previous comment).

AnHardt, 2015-07-12:
> current_position[Z_AXIS] = -zprobe_zoffset + Z_RAISE_AFTER_PROBING + (z_tmp - real_z);

does not look bad, but Z_RAISE_AFTER_PROBING has to be conditional.
But what are the conditions?

AnHardt, 2015-07-12:
You can find Z_RAISE_AFTER_PROBING conditional from SERVO_ENDSTOPS, Z_PROBE_ALLEN_KEY and Z_PROBE_SLED.
As the probe for Z_PROBE_SLED and Z_PROBE_ALLEN_KEY is stowed later (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L2864)

        #ifdef SERVO_ENDSTOPS
          current_position[Z_AXIS] = z_tmp - real_z - zprobe_zoffset + Z_RAISE_AFTER_PROBING;
        #else
          current_position[Z_AXIS] = z_tmp - real_z - zprobe_zoffset; 
        #endif

could be what we are locking for.

Or should it be SERVO_LEVELING instead  of SERVO_ENDSTOPS?
Anyway PROBE_SERVO_DEACTIVATION_DELAY should be defined together with the servos, not with ABL.

avri-schneider, 2015-07-13:
This line:
`if (z_loc < Z_RAISE_BEFORE_PROBING + 5) z_loc = Z_RAISE_BEFORE_PROBING;`
Does not make sense to me.
Why lower `z_loc` if `Z_RAISE_BEFORE_PROBING + 5 > z_loc > Z_RAISE_BEFORE_PROBING`?
What is so special about the `5` constant there?

thinkyhead, 2015-07-16:
> Or should it be `SERVO_LEVELING` instead of `SERVO_ENDSTOPS`?

`SERVO_LEVELING` is just an alias for `(defined(ENABLE_AUTO_BED_LEVELING) && defined(DEACTIVATE_SERVOS_AFTER_MOVE))` so it depends (oddly) on having servos set to turn off after moving.

thinkyhead, 2015-07-16:
> What is so special about the 5 constant there?

I'm not sure what kind of situation it's meant to address. Perhaps the idea is to leave a wide margin for float precision issues. Or perhaps it's only meant to ensure that the probe is merely "close enough" to the pre-probing position before it starts probing.

I'm not sure how this affects things overall, but we should remember also that Z might be 0 because it hasn't been homed, and yet it may be very far from the bed. If Z is positioned at Z_MAX but hasn't been homed, so it's set to 0, then raising the Z axis before probing is going to lead to grinding the motors as it tries to go past the physical top (or bottom) of its travel range.

avri-schneider, 2015-07-16:
Still seems like a bug to me. How is making `z_loc` *smaller* going to help?
If user is homing all axes and `Z` is currently touching the bed - what could be the value of `z_loc` (`current_position[Z_AXIS]`)? Could this: `Z_RAISE_BEFORE_PROBING + 5 > z_loc > Z_RAISE_BEFORE_PROBING` be `true`? If so - `Z` will be lowered and crash against the bed.

thinkyhead, 2015-07-16:
@avri-schneider Well, you can use `G92` to set any coordinates you want, and see how the machine responds. My guess is, if you tell the machine Z equals any value > `Z_RAISE_BEFORE_PROBING` and less than `Z_RAISE_BEFORE_PROBING + 5`, then when you issue a `G29`, I expect Z will move down by the difference from `Z_RAISE_BEFORE_PROBING`. And I agree, that would be bad in any case where Z is set within that range, but is actually very close to the bed.

closed 2016-04-23 by jbrazio