MarlinFirmware/Marlin #2413: Auto leveling not level properly

Hat3, 2015-07-10:
Hi all

I can not understand why the auto bed leveling not level correctly and not center the object on the bed, but has an offset (I did not understand where to change it).
As you can see in the following image, the extruder in the left side is higher than that in the right side and is reflected by the fact that the extrusion of material does not adhere to the bed, and as it moves to the right the layer tapering (observing the perimeter).

The bed is uneven, the direction of the inclination goes from the left corner at the bottom right-hand corner at the top.

The print setting is 0.2mm per layer.

Image: http://www.ingress-firenze.it/img-forum/autob.jpg

How can I do to fix these two things?
Centering on the object and auto bed leveling not level?

This is my config:
// these are the offsets to the probe relative to the extruder tip (Hotend - Probe)
// X and Y offsets must be integers
#define X_PROBE_OFFSET_FROM_EXTRUDER 47  //-25 default
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0 //-29 default
#define Z_PROBE_OFFSET_FROM_EXTRUDER -3.9 //-12 default

thinkyhead, 2015-07-11:
What version of Marlin are you using?

magillus, 2015-07-11:
Hello, it could be related to my issue I noticed recently after trying to adapt autolevel, see post and comments
Normal is not in correct position on my printer when Z axis is from top to bottom - print bed lift.

https://plus.google.com/+MateuszPerlak/posts/XNpAhkCKPCE

Hat3, 2015-07-11:
@thinkyhead

Last release, can you tell me the gcode command to get the version please?

Hat3, 2015-07-11:
Result of G30 command

[code]14:30:21.667 : Bed X: 50.00 Y: 100.00 Z: 4.00
14:30:25.664 : Bed X: 50.00 Y: 100.00 Z: 4.00
14:30:29.102 : Bed X: 50.00 Y: 100.00 Z: 4.01
14:30:32.959 : Bed X: 50.00 Y: 100.00 Z: 4.01
14:30:36.355 : Bed X: 50.00 Y: 100.00 Z: 4.01
14:30:39.659 : Bed X: 50.00 Y: 100.00 Z: 4.02
14:30:42.962 : Bed X: 50.00 Y: 100.00 Z: 4.02
14:30:46.412 : Bed X: 50.00 Y: 100.00 Z: 4.01
14:30:49.684 : Bed X: 50.00 Y: 100.00 Z: 4.02
14:30:53.449 : Bed X: 50.00 Y: 100.00 Z: 4.03
14:30:56.886 : Bed X: 50.00 Y: 100.00 Z: 4.03
14:31:00.334 : Bed X: 50.00 Y: 100.00 Z: 4.03
14:31:03.524 : Bed X: 50.00 Y: 100.00 Z: 4.03
14:31:07.115 : Bed X: 50.00 Y: 100.00 Z: 4.03
14:31:10.395 : Bed X: 50.00 Y: 100.00 Z: 4.04
14:31:13.584 : Bed X: 50.00 Y: 100.00 Z: 4.03
14:31:18.104 : Bed X: 50.00 Y: 100.00 Z: 4.03
up to 10mm
14:31:34.223 : Bed X: 50.00 Y: 100.00 Z: 3.96
14:31:37.973 : Bed X: 50.00 Y: 100.00 Z: 3.97
14:31:41.908 : Bed X: 50.00 Y: 100.00 Z: 3.97
14:31:45.313 : Bed X: 50.00 Y: 100.00 Z: 3.97
14:31:49.437 : Bed X: 50.00 Y: 100.00 Z: 3.98
14:33:40.855 : Bed X: 50.00 Y: 100.00 Z: 3.98
14:33:44.328 : Bed X: 50.00 Y: 100.00 Z: 3.98
14:33:47.755 : Bed X: 50.00 Y: 100.00 Z: 3.97
14:33:51.241 : Bed X: 50.00 Y: 100.00 Z: 3.98
14:33:54.726 : Bed X: 50.00 Y: 100.00 Z: 3.98
14:33:58.398 : Bed X: 50.00 Y: 100.00 Z: 3.98
14:34:01.259 : Bed X: 50.00 Y: 100.00 Z: 3.99[/code]

thinkyhead, 2015-07-11:
@Hat3 __M115__

scotty1024, 2015-07-11:
There is an error in the qr_solve that calculates the normal plane. The 3 point leveling doesn't use qr_solve. So all those that have been saying grid based leveling has a problem: you were correct.

foulowl, 2015-07-12:
I've been running into this issue as well I believe. Also, what happens when the fourth point is not in the same plane as the first three? Does Marlin create two planes and compensate accordingly depending on which triangle the hotend is over?

Wackerbarth, 2015-07-12:
At this point in time, there is no "Marlin".  Specifically, which version are you attempting to use?
And with what configuration options?

Hat3, 2015-07-12:
Marlin version 1.0.2
Thisi is my config:
http://www.ingress-firenze.it/img-forum/Configuration.h
http://www.ingress-firenze.it/img-forum/Configuration_adv.h

foulowl, 2015-07-12:
I am also using 1.0.2. 
Config:
https://gist.github.com/foulowl/ab812144c055ef8a0844

Basically my symptoms are the same as having an unleveled bed, one corner only.

Thanks!

Hat3, 2015-07-19:
@foulowl 
G29 R5 command do not perform bed autoleveling 5 times for me.

why?

AnHardt, 2015-07-19:
@Hat3 
Because:
a.) R adjusts the ___R___ight limit of the probing grid
b.) There is no parameter to repeat the autoleveling
c.) P sets the number of points that are tested. If you test a lot of points (close together) this has about the same effect as testing the same point multiple times.

M48 has a P parameter to repeat the test.

Hat3, 2015-07-19:
@AnHardt ok thank you

thinkyhead, 2015-07-29:
A lot of adjustments to auto leveling lately. I would recommend testing, but first we need to isolate a Z servo deployment problem for `G28`. I'll let you know when that is fixed and we can proceed to get back to stress-testing auto-leveling.

otvald, 2015-07-31:
[DELTA]. I have build a k800, and trying G28, and then G29 will cause no movement. NOTE: I have a *positive* Z_PROBE_OFFSET_FROM_EXTRUDER. Besides from that, I should have a pretty OK setup. I can level manually, though! It seem like G29 makes it go in the wrong direction. It tries to go up!

Anyone else seeing this (a 4 day old dev version)?

alexborro, 2015-08-01:
@scotty1024 , what is your proposal to fix the qr_solve normal plane calculation for grid leveling?

foulowl, 2015-08-03:
Hi folks. I have noticed that every one in five prints gives me this:
http://i.imgur.com/3OreuTK.jpg

The other times, the print is perfect.

thinkyhead, 2015-08-11:
@Hat3 There have been a lot of updates recently to clean up bed leveling, especially dealing with probes themselves. I'd wait until https://github.com/MarlinFirmware/MarlinDev/issues/65 is merged, and then give the latest https://github.com/MarlinFirmware/MarlinDev code a run.

Hat3, 2015-08-12:
Ok i will waiting the release (better)

thinkyhead, 2015-08-13:
@foulowl Try using the `M48` Repeatability Test and see how accurate your probe is. If you can't get repeatable results, you may want to tighten some things.

thinkyhead, 2015-08-13:
@otvald Describe more about your probe. How is it that you have a positive `Z_PROBE_OFFSET_FROM_EXTRUDER`?

paulusjacobus, 2015-08-13:
@thinkyhead your G29_debug branch has proved it's reliability to me. I have now very consistent leveling on my Prusa i2 printer and none of my prints have failed so far (2+ weeks). Even my wife can now operate the machine with a push of the button (the G start code in CURA has the initiation string G28 and G29)

thinkyhead, 2015-08-14:
@paulusjacobus Good to hear your leveling is working reliably. It seems like the cartesians with typical Z-probes are doing well. At this point we just have to get more testing and debugging on various Deltabots before I will have full confidence in these routines.

paulusjacobus, 2015-08-14:
@thinkyhead yes the Delta ABL is the next bit that needs to be worked on and unfortunately I don't have a Delta printer to help you with testing and debugging. Let's hope we make some fast progress with Delta's.

otvald, 2015-08-16:
@thinkyhead it is a K800 delta printer form china. The level switch is connected to the probehead. When the head touches the glas (and goes down a few mm) it will activate the switch. This means that the probehead will be lower than the switch 'probe'.

An Image of the setup is shown.
[image]

I made a short clip showing the mechanics: https://youtu.be/vDqdnY4BTiQ

thinkyhead, 2015-08-27:
@otvald If the latest code is still giving you bad results, then I should update my "debug_G29" branch and add more logging around delta-specific homing & leveling so you can give it a run and we can get some detailed data. I'll work on updating that branch later today and let you know when something is ready.

Wackerbarth, 2015-08-27:
@thinkyhead @otvald -- Rather than working from Scott's "debug_G29" branch, I'm going to go on and merge PR#54 (which includes his debug code) Since there continue to be suggestions to utilize the code, we should use a version that is a compile-time option rather than a separate branch.

paulusjacobus, 2015-08-27:
@Wackerbarth That's great to hear since I have been using this branch for month without any issues and very happy with the results.

Wackerbarth, 2015-08-27:
@paulusjacobus -- Good to hear. I only wish that you had responded in a positive manner sooner. I hesitate to accept something into our main branch until we have some verification that is helping, but NOT adversely affecting others.

Wackerbarth, 2015-08-27:
By The Way, I'm going to close this issue on "Marlin" and suggest that it be reopened on  "MerlinDev" since it seems to refer to our next release, rather than an already released version.

closed 2015-08-27 by Wackerbarth

thinkyhead, 2015-08-27:
@otvald Try using the `DEBUG_LEVELING_FEATURE` [sic] feature in the latest `MarlinDev/master` and see if it provides any clues. I may need to add more delta-specific logging to catch your particular issue, but it might be obvious from the existing logging. You'll need to issue `M111 S32` to turn on the extra logging.

Wackerbarth, 2015-08-27:
@thinkyhead -- (per your "sic", is there some problem with "DEBUG_LEVELING_FEATURE" ?  I'd be happy to change the name if that is an issue.

thinkyhead, 2015-08-28:
@Wackerbarth I see you named it that way to distinguish it from the flag `DEBUG_LEVELING`. I just included [sic] in that case because I found it amusing to say "`...FEATURE` feature" but actually I kind of like having top-level features named with `FEATURE` in the name. It's certainly better than those switches that include `ENABLE` in the name.

Wackerbarth, 2015-08-28:
Yes.  My interest is that we establish a convention (and migrate toward it) which makes it clear that the user is (a) selecting some optional behavior, (2) specifying that his configuration includes some optional hardware, or (3) specifying particular characteristics of his configuration. (e.g.. the length of the "X-axis".)

Wackerbarth, 2015-08-28:
"It's certainly better than those switches that include ENABLE in the name." -- Yeah, don't complain that I also changed the enabler option for automatic bed leveling. :)