MarlinFirmware/Marlin #2418: pins_PRINTRBOARD.h needs to know if I2C SD is in use or not

shampine1, 2015-07-11:
I think it would be a good idea to modify pins_PRINTRBOARD.h. If you have SD support, you have to move the Y endstop to the "Estop" plug; the Y pins are involved in I2C SD communication.
Perhaps:

#ifdef SDSUPPORT
   #define Y_STOP_PIN         37   // {must be 37 for SD card support and plug Ystop into Estop socket}
#else 
  #define Y_STOP_PIN         8  //  {No I2C SD present, Ystop in Ystop socket is OK}
#endif

I don't know enough about github yet to do a pull request on this.  I'll try, though!

thinkyhead, 2015-07-11:
We're not seeing your pull request. You have to make it against `MarlinFirmware/Development` for it to be merge-able into this repo.

shampine1, 2015-07-12:
Thanks for the help AND advice!

Roxy-3DPrintBoard, 2015-07-12:
When you think every thing is right, I have a PrintRBoard on my printer.   I've never been able to get the printer to print from a SD Memory card.   I'll be able to test and verify everything is right!

fmalpartida, 2015-07-12:
That is very odd, with the sav works well. 
I have received a report where by with repetitier it doesn't read the card or print from it while you can see the files and print from the lcd.

shampine1, 2015-07-12:
Are you printing off the SD card socket on the board or on an LCD panel?  I've tested the panel variety, but not the board mounted one.

Roxy-3DPrintBoard, 2015-07-12:
Board mounted.   I don't have an LCD panel.

fmalpartida, 2015-07-12:
I'm printing from an on board card not on the lcd. Using the lcd just to select and print. 

The board I am using is the sav mki you can find it in the Reprap wiki. Similar to the printrboard.

AnHardt, 2015-07-12:
Is the pinning for the SD the same for 'on-board' and 'on-pannel' ?

shampine1, 2015-07-12:
I don't know.  It would make sense.  It should be easy to tell from the schematic.

shampine1, 2015-07-13:
After a quick search, it looks like the onboard SD card slot should work fine if you go into configuration.h and un-comment the line below.  YMMV; I haven't tested this, but the Printrbot folks use it.  If someone else doesn't get it tested first, I could pull off my LCD and flash it without the LCD support to check.

//#define SDSUPPORT // Enable SD Card Support in Hardware Console

thinkyhead, 2015-07-16:
> I have a PrintRBoard on my printer. I've never been able to get the printer to print from SD

With this change (soon to be merged), as I read it, you would have to move your Y endstop to another position before SD will work.

Roxy-3DPrintBoard, 2015-07-16:
> With this change (soon to be merged), as I read it, you would have to move your Y endstop to another position (if you haven't already) before SD will work.

Yes.  My Y axis has been running on the extra E driver since almost the start.   Somebody figured out the problem and suggested that but they didn't have the complete solution in place.   So, in my case, I won't even have to move any wires.   I might just get lucky and have it work!

thinkyhead, 2015-07-16:
:four_leaf_clover:

shampine1, 2015-07-22:
I just finished checking the board mounted SD card.  If you have a Panelolu2 (which implies I2C SD is ON), then BOTH the panel mounted SD slot AND the board mounted SD slot work!  I didn't try it with cards in both places, though.  I've been having problems with my cheap Indian Panelolu2 clones having crummy SD slots; authentic T2P3 boards seem to be better quality, FYI.

thinkyhead, 2015-07-22:
> didn't try it with cards in both places

Probably a good thing. That would definitely be considered _unsupported_.

closed 2016-04-23 by jbrazio